evcc-io/evcc #22208: Vehicle and Charger (i.e. Volkswagen & KEBA) Implementation lacks "welcomecharge" support in UI Configuration

SebStange, 2025-07-04:
**Is your feature request related to a problem? Please describe.**
According to the documentation, KEBA P20, P30 and P40 as well as Vehicle implementations like the one for Volkswagen support a "welcomecharge" argument, which is support via evcc.yaml but not in the UI configuration, therefor making it hard to set the argument after the fact and w/o direct access to the db/yaml.

https://docs.evcc.io/docs/features/vehicle
https://docs.evcc.io/docs/devices/chargers#keba

It is possible this applies to other chargers and/or vehicles/manufacturers as well and should be checked/added in the process.

**Describe the solution you'd like**
Please add a checkbox to support "welcomecharge" for the above mentioned (as well as other relevant) chargers/vehicles.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A (Feel free to request more context if/where needed.)

VolkerK62, 2025-07-05:
check the "advanced settings"  within the Kebe Wallbox UI configuration.
The feature is called "charge on connection"

SebStange, 2025-07-05:
Incorrect premise. Sorry.

closed 2025-07-05 by SebStange

TobiasHuber1980, 2025-07-05:
[image]

also ich finde "charge on connection" nicht (mehr) im Wallbox UI.

SebStange, 2025-07-05:
[image]

[image]

TobiasHuber1980, 2025-07-05:
🫡