evcc-io/evcc #22221: Loadpoint: reset energ metrics funktioniert noch nicht richtig

VolkerK62, 2025-07-05:
Reset der Werte hat heute Nacht funktioniert. 💪
Jetzt bleibt nur zu hoffen, dass die Code Verschiebung keine blöden Nebeneffekte erzeugt.

_Originally posted by @docolli in https://github.com/evcc-io/evcc/issues/22126#issuecomment-3017967367_

Im Ladelog passt es nicht. Fällt mir jetzt erst auf, weil ich mal 2 Tage lang keinen Neustart gemacht habe.
Anbei die Ladelogeinträge vom 4.7. und 5.7.. An beiden Tagen war die WP jeweils ca. 4 Std in Betrieb und hat jeweils ca. 2kWh verbraucht.
Am 5.7. werden aber die kummulierten Werte der beiden Tage angezeigt.

[image]
[image]

VolkerK62, 2025-07-05:
ok. Kannst du mir sagen von welchem Zeitraum?

LeonSol, 2025-07-05:
Moin, 

schaut euch mal den Zählerstand am 05.07. an! Dieser ist nur um 1,9 kWh hoch gegangen obwohl als Verbrauch 4,1 kWh angezeigt werden...

andig, 2025-07-05:
Es braucht noch die Config. Woher kommt die Energiemenge?

VolkerK62, 2025-07-05:
Charger ist ein Shelly EM (der kann auch messen).
Der ist immer eingeschaltet und plottet den Betrieb der WP mit.

```
charger
---
db:10 {Type:template Title: Icon: Product:} map[channel:0 heating:true host:192.168.178.21 icon:heatpump integrateddevice:true standbypower:10 template:shelly]

vehicle
---
db:24 {Type:custom Title: Icon: Product:} map[yaml:title: WasserWP
icon: heatpump
capacity: 12
soc:
  source: mqtt
  topic: Haus/Wasser_Temp
  timeout: 60m]

loadpoint
---
db:11 {Type: Title: Icon: Product:} map[charger:db:10 circuit: defaultMode:now limitSoc:60 maxCurrent:2 meter: minCurrent:2 mode:now phasesConfigured:1 priority:10 soc:map[estimate:false poll:map[interval:3.6e+12 mode:charging]] thresholds:map[disable:map[delay:1.8e+11 threshold:0] enable:map[delay:6e+10 threshold:0]] title:WP(L2)-Info- vehicle:db:24]
```

LeonSol, 2025-07-05:
Im jetzigen release 0.204.5 ist es so, dass der Ladevorgang einer Wärmepumpe, die bei mir "nicht" als vehicle definiert ist, beendet wird, wenn die Zieltemperatur erreicht ist = "off". Dann bleibt aber die geladene Energiemenge stehen und wird zum nächsten Ladevorgang unabhängig von der Zeit, die dazwischen vergangen ist (also auch der nächste Tag), addiert. Wenn ich aber vor der nächsten Ladung manuell auf "pv" gehe und sofort wieder zurück auf "off", wird die geladene Menge auf null gesetzt und ein Ladevorgang geschrieben. Bei diesem manuellen "off"-"pv"-"off" passiert also genau das, was auch passieren sollte, wenn der Ladevorgang aufgrund des Erreichens der Zieltemperatur beendet wird. 

Es funktioniert also schon, nur nicht an der richtigen Stelle. Just my2cent ohne Programmierkenntnisse.

LeonSol, 2025-07-05:
Das mache ich übrigens jeden Tag seit dem letzten release und es funktioniert wunderbar:

[image]

andig, 2025-07-05:
@VolkerK62 schau mal https://github.com/evcc-io/evcc/pull/22068:

> The switchsocket workaround remains untouched since it might be required outside of integrated heating devices.

M.E. hat sich bei dem Shelly erwartungsgemäß gar nichts geändert. Damit es das täte müsstest Du ihm gemäß PR noch das `heating` Feature verpassen.

/cc @naltatis

andig, 2025-07-09:
> Hat er doch (siehe config)

Stimmt, hab ich übersehen.

> Für "db" gibt es keine debug Einträge.

Bekommen wir DB auf trace dazu?

VolkerK62, 2025-07-09:
Hab log.level geändert.
Melde mich, wenn ich morgen oder übermorgen ein neues Log habe.

andig, 2025-07-09:
@VolkerK62 wenns reproduzierbar falsch ist könntest Du https://github.com/evcc-io/evcc/pull/22279 nochmal ausprobieren. Aber ich rate...

VolkerK62, 2025-07-10:
Gestern Abend noch Warmwasser gemacht
```
10 00:00:22 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 00:00:22 ----
Jul 10 00:00:22 raspberrypi evcc[336970]: [lp-4  ] DEBUG 2025/07/10 00:00:22 charge power: 0W
Jul 10 00:00:22 raspberrypi evcc[336970]: [circuit-load] DEBUG 2025/07/10 00:00:22 power: 0W
Jul 10 00:00:22 raspberrypi evcc[336970]: [circuit-load] DEBUG 2025/07/10 00:00:22 current: 0A
Jul 10 00:00:22 raspberrypi evcc[336970]: [circuit-WPSG] DEBUG 2025/07/10 00:00:22 power: 0W
Jul 10 00:00:22 raspberrypi evcc[336970]: [circuit-WPSG] DEBUG 2025/07/10 00:00:22 current: 0A
Jul 10 00:00:22 raspberrypi evcc[336970]: [circuit-main] DEBUG 2025/07/10 00:00:22 power: 47.002W
Jul 10 00:00:22 raspberrypi evcc[336970]: [circuit-main] DEBUG 2025/07/10 00:00:22 current: 0.778A
Jul 10 00:00:22 raspberrypi evcc[336970]: [lpc   ] DEBUG 2025/07/10 00:00:22 power: 47.002W
Jul 10 00:00:22 raspberrypi evcc[336970]: [lpc   ] DEBUG 2025/07/10 00:00:22 current: 0.778A
Jul 10 00:00:22 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 00:00:22 grid power: 47W
Jul 10 00:00:22 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 00:00:22 grid powers: [-135 25 157]W
Jul 10 00:00:22 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 00:00:22 grid currents: [-0.803 0.606 0.778]A
Jul 10 00:00:22 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 00:00:22 aux 1 power: 1W
Jul 10 00:00:22 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 00:00:22 battery 1 power: 207W
Jul 10 00:00:22 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 00:00:22 pv 1 power: 0W
Jul 10 00:00:22 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 00:00:22 battery 1 soc: 19%
Jul 10 00:00:22 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 00:00:22 site power: 475W
Jul 10 00:00:22 raspberrypi evcc[336970]: [lp-4  ] DEBUG 2025/07/10 00:00:22 charge total import: 153.165kWh
Jul 10 00:00:22 raspberrypi evcc[336970]: [lp-4  ] DEBUG 2025/07/10 00:00:22 charger status: B
Jul 10 00:00:22 raspberrypi evcc[336970]: [lp-4  ] DEBUG 2025/07/10 00:00:22 charge total import: 153.165kWh
Jul 10 00:00:22 raspberrypi evcc[336970]: [db    ] TRACE 2025/07/10 00:00:22 UPDATE `sessions` SET `created`="2025-07-09 20:07:52.841",`finished`="2025-07-10 00:00:22.801",`loadpoint`="WP(L2)-Info-",`identifier`="",`vehicle`="WasserWP",`odometer`=NULL,`meter_start_kwh`=152.7374,`meter_end_kwh`=153.16479999999999,`charged_kwh`=0.4210999999999956,`charge_duration`="52m0s",`solar_percentage`=90.97660633357113,`price`=0.05383914368234559,`price_per_kwh`=0.1278535827175164,`co2_per_kwh`=17.595617649536344 WHERE `id` = 1853 1 <nil>
Jul 10 00:00:22 raspberrypi evcc[336970]: [lp-4  ] DEBUG 2025/07/10 00:00:22 charge total import: 153.165kWh
Jul 10 00:00:22 raspberrypi evcc[336970]: [lp-4  ] DEBUG 2025/07/10 00:00:22 vehicle soc: 50%
Jul 10 00:00:22 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 00:00:22 solar forecast: accumulated 539.006kWh, produced 600.900kWh, scale 1.115
Jul 10 00:00:27 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 00:00:27 ----
```
So sah heute morgen die Webansicht aus.
Alles war genullt (geladen, Sonne, Preis, Kosten,CO2) nur die Ladedauer nicht.

[image]

Dann wurde heute morgen wieder geladen.
Hier Log vom Start und vom Ende
```
Jul 10 07:23:17 raspberrypi evcc[336970]: [lp-4  ] DEBUG 2025/07/10 07:23:17 charge power: 315W
Jul 10 07:23:17 raspberrypi evcc[336970]: [circuit-load] DEBUG 2025/07/10 07:23:17 power: 0W
Jul 10 07:23:17 raspberrypi evcc[336970]: [circuit-load] DEBUG 2025/07/10 07:23:17 current: 0A
Jul 10 07:23:17 raspberrypi evcc[336970]: [circuit-WPSG] DEBUG 2025/07/10 07:23:17 power: 0W
Jul 10 07:23:17 raspberrypi evcc[336970]: [circuit-WPSG] DEBUG 2025/07/10 07:23:17 current: 0A
Jul 10 07:23:17 raspberrypi evcc[336970]: [circuit-main] DEBUG 2025/07/10 07:23:17 power: 379.17W
Jul 10 07:23:17 raspberrypi evcc[336970]: [circuit-main] DEBUG 2025/07/10 07:23:17 current: 3.12A
Jul 10 07:23:17 raspberrypi evcc[336970]: [lpc   ] DEBUG 2025/07/10 07:23:17 power: 379.17W
Jul 10 07:23:17 raspberrypi evcc[336970]: [lpc   ] DEBUG 2025/07/10 07:23:17 current: 3.12A
Jul 10 07:23:17 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 07:23:17 grid power: 379W
Jul 10 07:23:17 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 07:23:17 grid powers: [662 -40 -243]W
Jul 10 07:23:17 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 07:23:17 grid currents: [3.12 -0.55 -1.19]A
Jul 10 07:23:17 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 07:23:17 aux 1 power: 1W
Jul 10 07:23:17 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 07:23:17 battery 1 power: -1015W
Jul 10 07:23:17 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 07:23:17 battery 1 soc: 60%
Jul 10 07:23:17 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 07:23:17 pv 1 power: 1291W
Jul 10 07:23:17 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 07:23:17 site power: -415W
Jul 10 07:23:17 raspberrypi evcc[336970]: [lp-4  ] DEBUG 2025/07/10 07:23:17 charge total import: 153.165kWh
Jul 10 07:23:17 raspberrypi evcc[336970]: [lp-4  ] DEBUG 2025/07/10 07:23:17 charger status: C
Jul 10 07:23:17 raspberrypi evcc[336970]: [lp-4  ] DEBUG 2025/07/10 07:23:17 charge start energy: 153.165kWh
Jul 10 07:23:17 raspberrypi evcc[336970]: [lp-4  ] INFO 2025/07/10 07:23:17 start charging ->
Jul 10 07:23:17 raspberrypi evcc[336970]: [lp-4  ] DEBUG 2025/07/10 07:23:17 wake-up timer: stop
[... cut ...] =0.20543234169295604,`co2_per_kwh`=104.82553688984854 WHERE `id` = 1854 1 <nil>
Jul 10 08:31:17 raspberrypi evcc[336970]: [lp-4  ] DEBUG 2025/07/10 08:31:17 vehicle soc: 50%
Jul 10 08:31:17 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 08:31:17 solar forecast: accumulated 541.642kWh, produced 574.100kWh, scale 1.060
Jul 10 08:31:22 raspberrypi evcc[336970]: [site  ] DEBUG 2025/07/10 08:31:22 ----
```
und die Webansicht danach

[image]

Dauer und Energy beinhalten wieder die gestrigen Werte

closed 2025-07-10 by andig