evcc-io/evcc #22240: Battery support for Pixii Powershaper / Home

erikarenhill, 2025-07-06:
Pixii PowerShaper / Home is a battery that supports both MQTT and Modbus TCP. 

They claim Sunspec 802 Base battery is implemented but that's only partially true as the polarity for power is reversed from what's expected. 

When power is negative the Pixii discharges, which is not what evcc expects when using the available Sunspec 802 connector, so discharge/charge is not reading correctly (switches place).

This means we need a custom implementation for Pixii for reading which changes the polarity, and it can also be controlled to charge/discharge using modbus tcp which would be great to have in evcc.

I am myself currently working on supporting this in a fork, but wanted to add an issue here for referencing it and tracking in the official project.

erikarenhill, 2025-07-06:
> Happy to take a pr- copy the 802 and add a scale to the inverted value. Anf of course open an issue with the vendor…

yes that would be a quickfix for adding read-support, but to control it, it needs to a proprietary implementation of writing other register and keeping it alive by writing an incremental heartbeat every x seconds (where x < message lifetime configured on a Pixii gateway level), so in the long run I think a custom Go implementation is necessary.

I'm currently working on it

erikarenhill, 2025-07-06:
My last comment was because I thought you meant adding a template, but maybe you just meant copying the sunspec implementation. Ill have a look at it and send a PR, starting with read only then adding control down the road, it requires some fiddeling with the heart beats...

closed 2025-07-19 by github-actions[bot]

erikarenhill, 2025-07-19:
I will contribute to implement this, please keep issue open

andig, 2025-07-20:
@erikarenhill looking forward to PR. You can still reference this issue, even when closed. Since we're using the open issues as our team backlog I'd appreciate if we could keep it closed. Any implementation issues, let's discuss them in the PR. Thank you!