evcc-io/evcc #22261: evcc 0.204.5 und 0.124.10 inkompatible with mbmd meters

wita-sec, 2025-07-08:
### Describe the bug

platform: ubuntu 24.04.2 LTS 
evcc 0.204.5

meter for inverter Fronius Symo 12-5-3-M no longer working  with the new evcc version
config details see below. 

evcc 0.124.10 works fine with my config and outputs:

    pv-aktuell
    ----------
    Power: 0W

    netzbezug-aktuell
    -----------------
    Power: 35W

--------------------
Searching the evcc dokumentation I found a config that works for "pv-aktuell".
-------------
my_ evcc.yaml:
-------------

meters:
- name: pv-aktuell
  type: template
  template: fronius-gen24
  #id: 240
  usage: pv
  host: 192.168.2.63 # IP-Adresse oder Hostname
  port: 502 # Port, optional

---------
result:
---------
  pv-aktuell
  ----------
  Power:  0W
  Energy: 38497.3kWh

--------------
Problem/question: However I could not find a working config to retrieve "netzbezug-aktuell",
                  How do I need to configure evcc 0.204.5 for these values?

### Steps to reproduce

1. run cmd: evcc -c my_evcc.yaml meter
2. or run evcc service with config as my_evcc.yaml 
3. check logs
...

### Configuration details

```yaml
meters:
# Fronius aktueller PV-Ertrag in Watt
- type: mbmd
  uri: 192.168.2.63:502
  id: 1 # this is the default if no value is submitted
  name: pv-aktuell
  power: Power

# Fronius aktueller Netzbezug/Einspeisung in Watt
  - type: mbmd
    #usage: pv
    uri: 192.168.2.63:502
    id: 240
    name: netzbezug-aktuell
    power: Power
```

### Log details

```text
Jul 08 20:09:57 chicago systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Jul 08 20:09:57 chicago systemd[1]: evcc.service: Failed with result 'exit-code'.
Jul 08 20:09:57 chicago systemd[1]: Stopped evcc.service - evcc.
Jul 08 20:09:57 chicago systemd[1]: Started evcc.service - evcc.
Jul 08 20:09:57 chicago evcc[722418]: [main  ] INFO 2025/07/08 20:09:57 evcc 0.204.5
Jul 08 20:09:57 chicago evcc[722418]: [main  ] INFO 2025/07/08 20:09:57 using config file: /etc/evcc.yaml
Jul 08 20:09:57 chicago evcc[722418]: [db    ] INFO 2025/07/08 20:09:57 using sqlite database: /var/lib/evcc/evcc.db
Jul 08 20:09:57 chicago evcc[722418]: [main  ] INFO 2025/07/08 20:09:57 listening at :7070
Jul 08 20:09:57 chicago evcc[722418]: [main  ] FATAL 2025/07/08 20:09:57 meter [netzbezug-aktuell] cannot create meter 'netzbezug-aktuell': cannot create meter type 'mbmd': unknown meter type:
Jul 08 20:09:57 chicago evcc[722418]: [main  ] FATAL 2025/07/08 20:09:57 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.204.5

mfuchs1984, 2025-07-08:
Why not https://docs.evcc.io/docs/devices/meters#symo-gen24-plus

```
meters:
  - name: my_grid
    type: template
    template: fronius-gen24
    usage: grid
    host: 192.0.2.2 # IP-Adresse oder Hostname
    port: 502 # Port, optional
```

andig, 2025-07-08:
> cannot create meter type 'mbmd': unknown meter type:

You config is missing the actual meter type