evcc-io/evcc #22285: evcc gibt falschen SOC zurück

hardl1, 2025-07-10:
### Describe the bug

Gestern habe ich die MercedesMe2020 v0.27.0-beta.1 installiert, da ich mich oft neu anmelden musste und auch gesperrt habe. Danach einmal bei Mercedes angemeldet. Ladestand und -limit wurden korrekt angezeigt.
Gestern hat alles einwandfrei funktioniert und es war auch keine weitere Anmeldung bei Mercedes erforderlich.

Heute Ladevorgang gestartet, Laden funktioniert, LOG Fehlermeldung siehe unten.
Keine Ausgabe von Ladestand und -limit.

In HomeAssistant wird  der STATE OF CHARGE korrekt angezeigt.

Bei Ladeende kein Fehler im LOG mehr, kein Ladestand und -limit.

### Steps to reproduce

1. Ladevorgang starten
2.
3.
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070
  
log: debug
levels:
  cache: error

database:
    type: sqlite
    dsn: /data/evcc.db
    
sponsortoken: xxx
interval: 30s # control cycle interval
  
# mqtt message broker #
mqtt:
  broker: 192.168.xxx
  user: xxx
  password: xxx
  clientid: evcc
  topic: evcc

meters:

- name: my_grid
  type: template
  template: tasmota-sml
  usage: grid
  host: xxx
  user: xxx
  password: xxx

- name: my_pv
  type: template
  template: solarlog 
  usage: xxx 
  host: xxx
  port: 502  
 
site:
  title: Mein Zuhause
  meters:
    grid: my_grid
    pv:
    - my_pv
    
chargers:
  - name: my_charger
    type: template
    template: keba-modbus
    # Modbus TCP
    modbus: tcpip
    id: 255
    host: xxx 
    port: 502

vehicles:
  - name: mercedes_EQA
    type: template
    template: mercedes
    title: Mercedes EQA
    capacity: 77
    user: xxx
    region: EMEA
    accesstoken: 
    refreshtoken: 
loadpoints:
  - title: Garage 
    charger: my_charger 
    mode: off 
    priority: 0 
    soc:
      poll:
        mode: charging
        interval: 60m
      estimate: true 
    enable: 
      delay: 1m 
      threshold: 0 
    disable: 
      delay: 3m
      threshold: 0

tariffs:
  currency: EUR 
  grid:
    type: fixed
    price: 0.263 # EUR/kWh

  feedin:
    type: fixed
    price: 0.148 # EUR/kWh
```

### Log details

```text
[lp-1 ] INFO 2025/07/10 12:19:06 start charging ->
[lp-1 ] DEBUG 2025/07/10 12:19:06 wake-up timer: stop
[lp-1 ] ERROR 2025/07/10 12:19:06 vehicle soc: unexpected status: 418 (I'm a teapot)
[lp-1 ] DEBUG 2025/07/10 12:19:06 pv charge current: 4.7A = 0A + 4.7A (-3241W @ 3p)
[lp-1 ] DEBUG 2025/07/10 12:19:06 pv enable timer reset
[lp-1 ] DEBUG 2025/07/10 12:19:06 pv timer inactive

nach Ladeende

[lp-1 ] DEBUG 2025/07/10 13:14:36 charge power: 0W
[site ] DEBUG 2025/07/10 13:14:36 pv 1 power: 7306W
[site ] DEBUG 2025/07/10 13:14:36 grid power: -6313W
[site ] DEBUG 2025/07/10 13:14:36 site power: -6313W
[lp-1 ] DEBUG 2025/07/10 13:14:36 charger status: B
[site ] DEBUG 2025/07/10 13:15:06 ----
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc v0.204.5

ReneNulschDE, 2025-07-10:
Fixed with #22284

hardl1, 2025-07-10:
Sorry, ich habe die früheren Meldungen trotz Suche nicht gesehen.

closed 2025-07-10 by hardl1