evcc-io/evcc #22301: Battery, allow discharge only for high prices

r4z0rhome, 2025-07-10:
Problem:
EVCC nows the current price and can already use it to allow cheap net charging (for wallbox and battery).
What it lacks: save the battery for the most expensive hours. This could improve the RoI for small batteries significantly.

Feature (1: simple, 3: hard)
1. Set min price to allow battery discharge
2. Adjust discharge price dynamically (e.g. 4 most expensive hours)
3. Automate optimized battery usage based on pv forecast and price

**Describe alternatives you've considered**
Use alternate tool.
Adjust battery with App manually.

**Additional context**

andig, 2025-07-11:
That‘s basically a discharge plan for the home battery, opposite of a charge plan. Since home consumption is not deterministic though there is no closed solution.

andig, 2025-07-16:
> The charge plan does not take into the account the expected energy balance either.

Not sure what you mean. It ensures energy demand is put into lowest cost slots. What we need to do here is put energy supply into highest cost slots. It's the same thing.

> Set min price to allow battery discharge

I don't think this is very helpful since you'd need to adjust that limit on a constant basis. With planner you don't need to do that. Also, with force-charging you don't need to do that, at least as long as you're happy with "doesn't cost anything" and don't want to go for "most profit". It is essentially the same issue.

> I guess, the implementation will be quite simple. Instead of triggering charging one has to lock the discharge similar to other cases when discharg is locked.

Happy to take a PR if we can get convinced that this is helpful.

andromedavpn, 2025-07-18:
> I don't think this is very helpful since you'd need to adjust that limit on a constant basis. With planner you don't need to do that. Also, with force-charging you don't need to do that, at least as [...]
Basically there is no difference to charge limit from grid. It is just a mirrored use case. If there will be another option to avoid discharge during low grid prices it is also fine. However, at least to me, it sounds quite consistent to use the same approach as it is IMO basically the completition of the current functionality.

My real life example is. I sell solar electricity for 12ct to grid. Because of the storage losses, I want to use my storage if and only if the grid price is higher than 20ct. If the grid price is below 17ct it is cheaper for me to buy electricity from instead of using the stored electricity. In fact, my electricity price is around 15ct  during the night 2-5. However around 7 it is quite expensive. So I prefer to use grid between 2 and 5 and discharge the storage only if the grid price is above 25ct (winter) / 20ct (summer) .  But now, the storage is discharged in the night before 6 and I have to pay high grid prices at 7 (Sept-Apr).

So basically there are three regimes.
-  charge if below 15ct (or from solar)
-  do not discharge below 25ct (because I want to save the energy for high prices time window. this price limits depend on the capacity. Higher capacity will usually mean lower value, e.g. 20ct)
-  allow discharge if price is above 25ct

The limits depend obviously on the capacity of the storage, the storage losses, feed-in compensation, etc.

> Happy to take a PR if we can get convinced that this is helpful. 
Happy to contribute, but might take a long time ;-)

andig, 2025-07-20:
> The limits depend obviously on the capacity of the storage, the storage losses, feed-in compensation, etc.

...and especially on varying daily usage. Instead of more limits we'd really need a goal-driven dynamic battery optimization (...that we currently don't have). I still maintain that a battery planner would be more useful than a fixed limit.

andig, 2025-07-20:
See https://github.com/evcc-io/evcc/discussions/22468 for context

andromedavpn, 2025-07-28:
> > Instead of more limits we'd really need a goal-driven dynamic battery optimization (...that we currently don't have). I still maintain that a battery planner would be more useful than a fixed limit.
As long as the goal-driven discharging will avoid uneconomical discharge, it will cover my use case, though it is much simpler. Until then as workaround I will just charge the car at low grid prices at the night, so the discharge of the battery is then locked during this time window.

andig, 2025-07-28:
We‘re looking into better algorithms in https://github.com/evcc-io/evcc/pull/22495. That should cover economic battery optimisation while at the same time removing the need for setting any limits.

VolkerK62, 2025-07-28:
@andromedavpn @r4z0rhome 
have a look to this workaround
https://github.com/evcc-io/evcc/wiki/aaa-Lifehacks#entladung-eines-steuerbaren-hausspeicher-preisgesteuert-oder-manuell-sperren

djfanatix, 2025-07-30:
You could also have a look at:

https://github.com/davidusb-geek/emhass

andig, 2025-08-02:
Depends on https://github.com/evcc-io/evcc/pull/22495. Let's recheck when that is done.

closed 2025-08-02 by andig