evcc-io/evcc #22333: Climater active when charging power is less then 6A

marcelGoerentz, 2025-07-12:
**Is your feature request related to a problem? Please describe.**
I don't like it that evcc does not recognize in short term when the climater is active

**Describe the solution you'd like**
Since the minimum charge current is 6A a power consumption of the charger less is mostly because of the climater. evcc should detect this and should indicate it in the UI.

**Describe alternatives you've considered**
None

**Additional context**
None

0x3d13f, 2025-07-12:
6A is the minimum the charger offers to the car. But the car can take less any time. For example my Hyundai Ioniq Classic sometimes charges around 5A which is a normal charge and not the climater.

marcelGoerentz, 2025-07-12:
> it should be shown in the UI, but only if the car provides this information
>

That's true, but it sometimes takes too long and the car is using its battery and gets charged (not very smart when there is enough energy from the pv)

 >6A is the minimum the charger offers to the car. But the car can take less any time. For example my Hyundai Ioniq Classic sometimes charges around 5A which is a normal charge and not the climater.
>

Interesting, but how much less could it be?

The climater of my BMW takes up to 1,2kW and is getting less during the climate control.

andig, 2025-07-14:
In general, climater detection is flaky. There's no good way to do it. Doing this via current is another flaky way of doing so. In general, using the planner, we will charge as late as possible (...and even reserve non-optimal time slots right before target time). Due to this late charging, climater detection should not be necessary at all since vehicle will already be active.

That said- we can't/wont add this logic on top.