evcc-io/evcc #22337: Planner: allow override of currently activated scheduled plan

marcelvanveen-prodapt, 2025-07-12:
**Is your feature request related to a problem? Please describe.**
* I currently have two active repeating charging plans:
(1) Mon-Fri, 7:00, 80% (plan # 2)
(2) Sat-Sun, 8:00, 80% (plan # 3)

* Now, when I connect my EV in the evening (e.g. Saturday at 17:00), evcc will configure a plan to have the car charged to 80% Sunday at 8:00 (this will be then be configured in charging plan # 1). 

* Now, let's say I know I will not be needing the car the next day before 13:00, so I update plan # 1 to be set to 13:00 instead of 8:00 and I click apply.

What I would expect is that evcc will now select the cheapest hours between Saturday 17:00 and Sunday 13:00, but instead it still selects the cheapest hours between Saturday 17:00 and Sunday 8:00 (probably taken from plan # 3). The only way to get evcc to select the cheapest hours between Saturday 17:00 and Sunday 13:00 is to disable repeating plan # 2 and # 3 (which of course is unwanted).

**Describe the solution you'd like**
I want to be able to override the active charging plan once they have been applied (after starting a charging session), without having to disable the repeating plans. So, I would expect functionality to works as follows:
1. User connects EV to loadpoint
2. evcc detects charging session and applies/copies appropriate repeated charging plan (# 3 in my example above) into plan # 1
3. evcc now follows only the configuration of plan # 1 and ignores all other repeating plans for the rest of the charging session

This would allow me to reconfigure the current charging session without updating the repeating plans.

**Describe alternatives you've considered**
I don't see any alternatives (temporarily disabling repeated plans is not really an option).

**Additional context**

StefanSchoof, 2025-07-12:
Similar topic: https://github.com/evcc-io/evcc/issues/20203

naltatis, 2025-07-13:
@marcelvanveen-prodapt So the mechanism you have in mind is, that the static plan (# 1) always takes precedence and overwrites all existing scheduled plans. I'm a little unsure if this would be general expected behavior and how we'd communicate this behavior clearly to the users. But I understand your point and currently don't see any practical scenarios or use-cases against this.

\cc @Maschga @andig

naltatis, 2025-07-17:
Nevertheless, I find the thought experiment, that the existence of an active one time plan temporarily disables all repeating plans interesting. This could also simplify our current implementation, wouldn't add more settings and address the pause use-case. Main thing I'm insure with if this behavior would match the expectation of most people and I still dont have a good idea how to communicate this elegantly. I'll keep this in mind for future iterations on planner.