evcc-io/evcc #22341: Übersicht über Ladepunkte

Razer2412, 2025-07-13:
Hallo zusammen! 

Ich nutze evcc schon seit einigen Jahren für einen 3-phasen Heizstab in Verbindung mit Shelly Relais als Überschusssteuerung.

Nun habe ich eine zusätzliche Wallbox angebracht.

evcc nutze ich hauptsächlich über ein Tablet an der Wand.
Da hier der Platz begrenzt ist, kann ich nur einen Ladepunkt darstellen und somit entweder die "Wassertemperatur" oder die Wallbox beobachten. 

Gibt es Pläne, die einzelnen Ladepunkte in der "oberen Übersicht" darzustellen? Ähnlich wie es bei mehreren "Erzeugern" bereits der Fall ist? 

[image]

Danke und viele Grüße
Pascal

VolkerK62, 2025-07-13:
Gibt es schon. Bin mir nur nicht sicher, ob das im letzten Release schon drin ist oder nur im Nightly.

[image]

naltatis, 2025-07-13:
Ja, das ist schon länger im Release. Alle aktiven(!) Ladepunkte werden zum Ausklappen oben angezeigt. Heißt in deinem Screenshot würde ich annehmen, dass gerade keine Energie bei den Ladepunkten fließt.

Razer2412, 2025-07-13:
Mein Fehler! 

Funktioniert - vielen Dank!

closed 2025-07-13 by naltatis