evcc-io/evcc #22345: HomeAssistant Smart Switch - support for input_boolean

marq24, 2025-07-13:
In the current implementation of the `homeassistant-switch.go` @ https://github.com/evcc-io/evcc/blob/7b3719e69fd871e10ee335bcb8e18de92d7afa69/charger/homeassistant-switch.go#L85 it's assume, that the configured `switchentity` is really a switch entity - therefor the service that will be called is hardcoded to the `switch` domain... [some sort of natural based on the naming of the property of the charger]...

Nevertheless - defining helper-switch entities in HA will result in a `input_boolean`  (and not a switch) - so right now evcc can not toggle such a ha-helper-switch (since `input_boolean != switch`. [In my context I use such a ha-helper, since I want to set the hotwater-temperature to a specific value (and switch it back to the default when there is no PV power)]

My "quick & dirty" solution would be to extract the domain from the `switchentity` (first string till `.`) and use this domain in the line https://github.com/evcc-io/evcc/blob/7b3719e69fd871e10ee335bcb8e18de92d7afa69/charger/homeassistant-switch.go#L93C38-L93C44 (instead of the hardcoded domain `switch`)

andig, 2025-07-13:
No, but I don't know HA. The whole thing comes down to "which kind of REST request do you need and what's the payload and response"?

andig, 2025-07-13:
> they are input_boolean entities

Seems their api is entirely different?

andig, 2025-07-13:
Ai- genearted answer? Your curl statements are sifferent then what‘s currently in go, hence the question.

marq24, 2025-07-13:
> ... Your curl statements are different then what‘s currently in go, hence the question.

?

assuming, that (for turn on):

- `c.switchEntity` = `input_boolean.wkh_hotwater_switch`  
- `c.baseURL` = `http://[my-ip]:8123`  

running then through the __suggested code__ (that is already now in the PR)...
```
	domain := strings.Split(c.switchEntity, ".")[0]
	uri := fmt.Sprintf("%s/api/services/%s/%s", c.baseURL, domain, service)
```

will result in `uri = http://[my-ip]:8123/api/services/input_boolean/turn_on` which is (if I am not totally blind in the same uri then in test curl command `curl -X POST "http://[my-ip]:8123/api/services/input_boolean/turn_on" ... `

andig, 2025-07-13:
Ah, thanks- I‘ve misread.

closed 2025-07-13 by andig