evcc-io/evcc #22346: Planer löst nicht aus.

Oleharder, 2025-07-13:
### Describe the bug

Wenn ich den Plan 1 für den Nächsten Tag 7:30 stelle zeigt er auch korekt an wie er Laden möchte. Tut es aber nicht.

[image]

[image]

### Steps to reproduce

1.Plan erstellt - 7:30
2.Ankunft auf 50 %  gestellt
3.Auto angeschlossen und staret automatisch auf 50 %
4. Stopt bei 50 %
5. nächsten morgen ist er immernoch auf 50 % ein manuelles Starten funktioniert dann.
6. 
...

### Configuration details

```yaml
# open evcc at http://evcc.local:70network:
Network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070
sponsortoken: *****
log: debug
levels:
  cache: error

modbusproxy:
  - port: 1502
    uri: 192.168.178.188:502
    readonly: true

# unique installation id
plant: *****

interval: 30s # control cycle interval

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

vehicles:
- type: template
  template: hyundai
  user: *****
  password: *****
  vin: *****
  title: Oles Auto
  capacity: 56
  language: de
  name: ev4

site:
  title: Rosenkehre24  auf pi 300625

Version: `0.204.5`
```

### Log details

```text
[main  ] INFO 2025/07/13 11:09:17 evcc 0.204.5
[main  ] INFO 2025/07/13 11:09:17 using config file: /etc/evcc.yaml
[db    ] INFO 2025/07/13 11:09:17 using sqlite database: /var/lib/evcc/evcc.db
[db    ] TRACE 2025/07/13 11:09:17 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/07/13 11:09:17 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/07/13 11:09:17 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/07/13 11:09:17 SELECT * FROM `settings` 18 <nil>
[db    ] TRACE 2025/07/13 11:09:17 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/07/13 11:09:17 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/07/13 11:09:17 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[main  ] INFO 2025/07/13 11:09:17 listening at :7070
[main  ] FATAL 2025/07/13 11:09:17 modbusproxy listen tcp :1502: bind: address already in use
[main  ] FATAL 2025/07/13 11:09:17 will attempt restart in: 15m0s
[main  ] TRACE 2025/07/13 11:09:17 GET https://api.github.com/repos/evcc-io/evcc/releases/latest
[main  ] FATAL 2025/07/13 11:09:17 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.5

naltatis, 2025-07-13:
@Oleharder Dein log sagt, dass evcc nicht starten konnte weil es schon läuft. Damit wir hier Debuggern können brauchen wir ein Log in den zur Diskussion stehenden Zeitpunkt (bspw. 5:52 Uhr, geplanter Start) erkennen kann. 

Hast du sichergestellt, dass du auch im PV-Modus bist? Bei "Aus" ist der Plan nicht aktiv.

naltatis, 2025-07-13:
We'll reopen once we can reproduce.

closed 2025-07-13 by naltatis