evcc-io/evcc #22348: Smartevse: Erwartete Umschaltung von 1p auf 3p erfolgt nicht

11-Andreas, 2025-07-13:
### Describe the bug

Smartevse: Die erwartete Umschaltung von 1p auf 3p erfolgt oft nicht.
Überschuß ist für 3p ausreichend, Hausbatterie ist bereits voll, es wird bereits mit 2 kW eingespeist und trotzdem bleibt die Wallbox auf 1p hängen.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 95b4e37254d836a0c6f5850a29a1d2dd2ffd635a860c98c6639b9c2ff1be7ee9

interval: 30s # control cycle interval

meters:
- type: template
  template: huawei-sun2000
  id: 1
  host: 192.168.0.92
  port: 502
  usage: grid
  modbus: tcpip
  timeout: 15s
  storageunit: 1
  name: grid1
- type: template
  template: huawei-sun2000
  id: 1
  host: 192.168.0.92
  port: 502
  usage: pv
  modbus: tcpip
  timeout: 15s
  storageunit: 1
  name: pv2
- type: template
  template: huawei-sun2000
  id: 1
  host: 192.168.0.92
  port: 502
  usage: battery
  modbus: tcpip
  timeout: 15s
  storageunit: 1
  name: battery3

chargers:
- type: template
  template: smartevse
  id: 2
  device: /dev/ttyUSB0
  baudrate: 9600
  comset: 8N1
  modbus: rs485serial
[... 466 lines cut ...]
[lp-1  ] DEBUG 2025/07/13 15:34:37 vehicle soc: 78%
[lp-1  ] DEBUG 2025/07/13 15:34:37 vehicle soc limit: 80%
[lp-1  ] DEBUG 2025/07/13 15:34:37 vehicle range: 264km
[smartevse] TRACE 2025/07/13 15:34:37 modbus: send 02 04 01 02 00 01 91 c5
[smartevse] TRACE 2025/07/13 15:34:37 modbus: recv 02 04 02 14 14 f2 3f
[lp-1  ] DEBUG 2025/07/13 15:34:37 pv charge current: 29A = 20A + 8.95A (-2059W @ 1p)
[site  ] DEBUG 2025/07/13 15:34:37 solar forecast: accumulated 1427.472kWh, produced 1205.990kWh, scale 0.845
[httpd ] TRACE 2025/07/13 15:34:39 GET /api/system/log
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.204.5

andig, 2025-07-13:
Probiers mal bitte mit

    evcc charger

11-Andreas, 2025-07-13:
pi@raspberrypi:/etc $ evcc charger
[main  ] INFO 2025/07/13 17:44:39 using config file: /etc/evcc.yaml
[main  ] FATAL 2025/07/13 17:44:39
Found systemd service database at "/var/lib/evcc/evcc.db", evcc has been invoked with no explicit database path.
Running the same config with multiple databases can lead to expiring vehicle tokens.

If you want to use the existing service database run the following command:

sudo evcc --database /var/lib/evcc/evcc.db

If you want to create a new user-space database run the following command:

evcc --database ~/.evcc/evcc.db

If you know what you're doing, you can skip the database check with the --ignore-db flag.

TobiasHuber1980, 2025-07-13:
> Probiers mal bitte mit
> 
>     evcc charger

sudo evcc charger --database /var/lib/evcc/evcc.db

andig, 2025-07-13:
Nein. Schau die mal die Optionen an. Damit schaltest du die Phasen testweise per Kommandozeile um.

premultiply, 2025-07-13:
Mit 99,4 V wird die Erkennung der anliegenden Phasen auch nicht funktionieren.
Was ist denn da los?

11-Andreas, 2025-07-13:
Die manuelle Umschaltung in UI funktioniert tadellos.
In der Stellung "automatischer Wechsel" funktioniert das runterschalten von 3p auf 1p korrekt. Jedoch nicht von 1p auf 3p.

99,4 V sieht tatsächlich komisch aus. Das war allerdings in dem Zustand, wo nicht geladen wurde (abends) und läßt sich nicht reproduzieren. Statische Aufladung offenen Potentials?

Siehe Ergebnisse bei erzwungenen Zuständen:

Zustand AUS:

[main  ] INFO 2025/07/13 20:55:39 using config file: /etc/evcc.yaml
[db    ] INFO 2025/07/13 20:55:39 using sqlite database: /var/lib/evcc/evcc.db
Power:          0W
Energy:         1040.2kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 233V 7.54V 7.45V
Charge status:  B
Enabled:        false
Charged:        5.2kWh

Zustand 1p, MIN+PV:

[main  ] INFO 2025/07/13 20:58:07 using config file: /etc/evcc.yaml
[db    ] INFO 2025/07/13 20:58:07 using sqlite database: /var/lib/evcc/evcc.db
Power:          1398W
Energy:         1040.2kWh
Current L1..L3: 6.04A 0.0352A 0.0586A
Voltage L1..L3: 232V 6.44V 6.75V
Charge status:  C
Enabled:        true
Charged:        5.2kWh

Zustand 3p, MIN+PV:

[main  ] INFO 2025/07/13 21:02:12 using config file: /etc/evcc.yaml
[db    ] INFO 2025/07/13 21:02:12 using sqlite database: /var/lib/evcc/evcc.db
Power:          4175W
Energy:         1040.3kWh
Current L1..L3: 6.01A 6.03A 5.92A
Voltage L1..L3: 229V 236V 232V
Charge status:  C
Enabled:        true
Charged:        5.3kWh

Zustand AUS:

[main  ] INFO 2025/07/13 21:04:20 using config file: /etc/evcc.yaml
[db    ] INFO 2025/07/13 21:04:20 using sqlite database: /var/lib/evcc/evcc.db
Power:          0W
Energy:         1040.4kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 231V 0.262V 0.484V
Charge status:  B
Enabled:        false
Charged:        5.4kWh

11-Andreas, 2025-07-13:
PS.: In dem TRACE-LOG ist die Spannung auf den 2 inaktiven Phasen immer zwischen 6 und 7V.
Ist das ein Grund für's Nichtumschalten?

andig, 2025-07-13:
Wie gesagt. Probiers auf der Kommandozeile aus.

11-Andreas, 2025-07-13:
Ich kenne die Kommandos nicht. Sehe auch keine Optionen. Brauche Unterstützung.

RenatusRo, 2025-07-13:
->  https://docs.evcc.io/docs/reference/cli/evcc_charger

andig, 2025-07-14:
Geht also. Dann brauchts ein debug Log für site/loadpoints bei dem man sieht wann/wo das nicht geht. In dem Kauderwelsch oben ist nichts zu erkennen.

11-Andreas, 2025-07-14:
Ich muss den Moment der erwarteten Umschaltung erst einfangen. Das kann einige Tage dauern.

11-Andreas, 2025-07-19:
[evcc-20250719-115223-debug_Nur Zeitfenster.log](https://github.com/user-attachments/files/21327571/evcc-20250719-115223-debug_Nur.Zeitfenster.log)

11-Andreas, 2025-07-19:
Ich hab nun die Konstellation für die erwartete Phasenumschaltung 2 Mal durch Veränderung der Hausverbraucher herbeigeführt. Siehe neuen Log.

1. Mal von 11:16 bis 11:18.
2. Mal von 11:25 bis 11:26.

Beobachtung:
Beides Mal das gleiche Verhalten: Die Hochschaltung wird durch den ablaufenden Timer angekündigt. Danach wird die Ladeleistung deutlich reduziert, geht jedoch nicht auf Null.
Dann wird die Leistung wieder hochgefahren, aber halt 1-phasig.

Folgende ergänzende Infos:
1. Wenn bereits beim Einstecken genug Überschuß für 3p vorhanden ist, dan wird auch mit 3p gestartet.
2. Das Runterschalten von 3p auf 1p im Automatik--Modus funktioniert immer zuverlässig.
3. Das manuelle Umschalten von 1p auf 3p funktioniert ebenso. (hier im Log um 11:47 getestet).
4. Der Log wurde bereits unter v0.205.0 durchgeführt.

PS: Hab auch einen TRACE-Log zu diesem Zeitraum, falls doch benötigt.

stephanmally, 2025-07-19:
Hi zusammen, Ich habe auch eine Huawei Anlage und seit 13.7 Schwierigkeiten - evtl liegt es an der Kombi mit Huawei?

swallage, 2025-07-20:
Das umschalten gehts oft nicht gut und hängt ab vom Wagen. Normaler Weise müssen Sie das Laden unterbrechen (abschalten), dann die Umschaltung and dann wieder Laden anfangen. Mein Renault Megane E-Tech gibt sehr oft ein Fehler während die Umschaltung. Es ist schon lange her dass ich die Umschaltung benutzt habe.

swallage, 2025-07-20:
Wenn ich hier oben lese dann glaub ich das die automatischen Umschaltung vielliecht zu schnell geht?

11-Andreas, 2025-07-20:
[evcc-20250719-115158-trace_Zeitfenster.txt](https://github.com/user-attachments/files/21336221/evcc-20250719-115158-trace_Zeitfenster.txt)

11-Andreas, 2025-07-20:
> Das ist unnötig, Du kannst das mittels `evcc charger` testen. Bitte probier mal:
> 
> 1. evcc aus
> 2. Ladevorgang 1p an der Box starten
> 3. Während (!) des Ladevorgangs mit `evcc charger` auf 3p umschalten
> 
> klappt das?

Hab es erneut ausprobiert - das klappt.
Es verhält sich genau so wie bei einer manuellen Umschaltung aus der UI.
Die Ladeleistung geht zuerst auf Null kW, dann geht es 3-phasig weiter.
Bei gescheiterten Fällen, wo die Umschalung NICHT manuell eingeleitet wird geht die Ladeleistung nicht auf Null.
Sie wird lediglich stark reduziert, aber eben ncht auf Null. Und dann bleibt die smartevse in 1p-Betrieb.

andig, 2025-07-23:
> Wenn ich hier oben lese dann glaub ich das die automatischen Umschaltung vielliecht zu schnell geht?

@swallage what is "too fast"? How much delay does it need?

swallage, 2025-08-05:
@andig : as a rule of thumb I would recommend switching like:
1. set current to 0 A
2. wait 2 seconds
3. change from/to 1/3p
4. wait 2 seconds
5. set current to available Amps
It makes sense to make this value configurable. From my experience switching time is fully dependent on the car.

premultiply, 2025-08-17:
Dann sollten wir die Funktion abschalten, da die Firmware ganz offensichtlich keinerlei interne Sicherungsmechanismen implementiert hat.

Eine Schaltung von Phasen im laufenden Betrieb muss die Firmware in jedem Fall blockieren.

Wir können somit davon ausgehen, dass die Firmware keine Phasenumschaltung unterstützt.

11-Andreas, 2025-09-26:
Die manuelle Umschaltung wäre keine Lösung, sondern Akzeptanz des Problems.
@andig mir ist völlig klar, dass ich keinen Anspruch auf die Lösung habe, ich würde mich aber sehr darüber freuen. Ich verstehe nicht warum es bei manuellen Imschaltung JEDES Mal funktioniert aber bei Automatik NIE. Wenn das so gut reproduzierbar/wiederholbar ist, dann ist es doch kein Zufall und die Ursache evtl. greifbar? Sind es unterschiedliche Befehle oder Befehlsfolgen?
Vielleicht könnte man die Abfolge realiesiern wie @swallage weiter oben vorgeschlagen hat, aber eine größere Pause zwischen 1p-AUS und 3p-EIN einbauen, z.B. 20 sek. ?
Vielleicht kann man nach dem 1p-AUS den Stromfluß abfragen um dann, wenn keiner mehr fließt wieder im 3p einzuschalten?

PR #24018 (2025-10-01): Smartevse: disable during phase switch
Fix #22348

If that doesn't work either, we will completely remove the phase switching for this charger.