evcc-io/evcc #22360: PV-Modus: Einspeisung bei maximaler Batterieladeleistung verhindern

Mananix, 2025-07-14:
**Is your feature request related to a problem? Please describe.**
Heute morgen hatte ich mal wieder festgestellt, das im Modus PV 1,5kw eingespeist wurden, da der Akku noch nicht bei 60% war aber seine maximal Ladeleistung von 5kw erreicht waren.

**Describe the solution you'd like**
Ideal wäre eine Option zu sagen, dass wenn dieses Szenario eintritt, auch mit minimaler Leistung angefangen wird zu laden auch wenn die Grenze für batterieunterstütztes Laden nicht erreicht ist. Dadurch soll ja nicht batterieunterstützt geladen werden sondern eher die Ladeleistung des Speichers reduziert und dieser trotzdem weiter geladen und die Einspeisung verhindert werden.

**Describe alternatives you've considered**
Aktuell behelfe ich mir damit manuell den Modus von PV auf Min+PV zu ändern.

Viele Grüße
Mananix

cschlipf, 2025-07-18:
Hm, also bei mir wird in diesem Fall in der Tat das Auto geladen, auch wenn der Akku noch nicht den Priority SoC erreicht hat.

Dumme Frage um das auszuschließen: Automatische Phasenumschaltung kann die Wallbox und der Minimalstrom steht auf 6A?

premultiply, 2025-07-18:
Das Fahrzeug wird geladen sobald der verbleibende Überschuss > als die notwendige Mindestleistung ist.

Ich kann nicht erkennen das daran etwas nicht funktionieren sollte.
Falls dem tatsächlich nicht so ist braucht es wie immer ein aussagekräftiges Log dazu.

closed 2025-07-18 by premultiply

premultiply, 2025-07-18:
Es gilt weiterhin https://github.com/evcc-io/evcc/issues/22360#issuecomment-3088615272