evcc-io/evcc #22388: Issue with latest nightly

SolarPower2024, 2025-07-15:
### Describe the bug

Is there a known bug in the latest nightly?

I just tried switching from latest stable to latest nightly - same config, same DB but the nightly version is not starting.

Switching back to stable is working again. 

### Steps to reproduce

1. Stop stable version 
2. start nightly version with same config / db

### Configuration details

```yaml
Full config in DB, but should not be relevant for this issue.
```

### Log details

```text
[main  ] FATAL 2025/07/15 18:44:04 database sql: Scan error on column index 10, name "charge_duration": converting driver.Value type string ("") to a int64: invalid syntax; sql: Scan error on column index 10, name "charge_duration": converting driver.Value type string ("") to a int64: invalid syntax; sql: Scan error on column index 10, name "charge_duration": converting driver.Value type string ("") to a int64: invalid syntax
[main  ] FATAL 2025/07/15 18:44:04 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-07-16:
Column definition is correct and there were no changes:

    ChargeDuration  *time.Duration `json:"chargeDuration" csv:"Charge Duration" gorm:"column:charge_duration"`

In any case, just change column type in sql and see if that takes care of it.

andig, 2025-07-16:
> Issue with latest nightly

It generally helps to give issues an easy-to-grasp title. Given we are in "Issues" here, adding "Issue with..." does not add any new information.

SolarPower2024, 2025-07-16:
> > Issue with latest nightly
> 
> It generally helps to give issues an easy-to-grasp title. Given we are in "Issues" here, adding "Issue with..." does not add any new information.

Sorry, my bad. Will improve next time. 

> Strange, sounds like your db's sessions table has `charge_duration` as type string and not integer. Can't reproduce that and we did not change anything directly in this mapping recently.
> 
> [@andig](https://github.com/andig) any ideas?

Ok, if you can't reproduce it's probably my DB. Will check my DB if I can find something odd in it. 

It's just strange, that the stable version starts without problems and issue occurs only with nightly version.

andig, 2025-07-16:
> It's just strange, that the stable version starts without problems and issue occurs only with nightly version.

Absolutely. But without further insights (or even your DB) nothing we can do. There are no obvious changes.

closed 2025-07-16 by andig