evcc-io/evcc #22394: Feed-in priority paused hours backwards

tehFred, 2025-07-16:
### Describe the bug

[image]

A feed in tarrif of -420.6c indicates I will be charged negative (i.e. earn money for feed in at this stage) on Amber dynamic tarrifs in Australia.

When I select that part of the chart, it indicates the pause hours will be the opposite of what I want - instead of pausing only when the price is high, it's pausing every other hour.

I assume this is an issue with negative values on Amber being opposite of the norm? Should we be inverting amber feed in prices to match what is expected? Or allow both less than or greater than selections for the feed in limit selection?

### Steps to reproduce

1.`
2.
3.
...

### Configuration details

```yaml
`
```

### Log details

```text
`
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.205.0

tehFred, 2025-07-16:
Grid use shows as a much higher value, so I believe it's configured correctly

[image]

ross-w, 2025-07-16:
Ah right sorry I misread consumption as import. Will have a look after the kids are in bed. Interesting how the UI is showing positive for feed in there….

tehFred, 2025-07-16:
Interesting indeed, should the feed in priority UI be referencing the grid export price here, which is in fact positive? (rather than the negative consumption value)

naltatis, 2025-07-17:
fixed with https://github.com/evcc-io/evcc/pull/22405

closed 2025-07-17 by naltatis