evcc-io/evcc #22400: Gemini CLI and MCP implementation are incompatible

andrewfoster, 2025-07-16:
### Describe the bug

Unlike Claude, Gemini's CLI refuses to work with EVCC's MCP implementation in EVCC 0.205.0

Gemini 0.1.12 reports:

Failed to list or register tools for MCP server 'evcc': Error: type and anyOf cannot be both populated.
No tools registered from MCP server 'evcc'. Closing connection.
MCP ERROR (evcc): Error: SSE stream disconnected: AbortError: This operation was aborted

In contrast on the same install claude works as per command in the PR

### Steps to reproduce

1. Define EVCC as an MCP server
2. Run gemini CLI with debugging.
3. Observe above error and no tools are registered.

### Configuration details

```yaml
mcp: true
```

### Log details

```text
Failed to list or register tools for MCP server 'evcc': Error: type and anyOf cannot be both populated.
No tools registered from MCP server 'evcc'. Closing connection.
MCP ERROR (evcc): Error: SSE stream disconnected: AbortError: This operation was aborted
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-07-16:
Closed in 614a6da6e61bd3405c4614ca84f7dae3597accb4. Please keep feedback on use cases and limitations coming.

closed 2025-07-16 by andig

andig, 2025-07-17:
Note this likely still won't work, due to defect on Gemini side: https://github.com/google-gemini/gemini-cli/issues/4127