evcc-io/evcc #22401: Support for Charge Amps LUNA

tclas, 2025-07-16:
Hello,

while I'm in the process of decission which wallbox to buy, the Charge Amps LUNA was placed on the short list. I've figured out, the Charge Amps HALO seems to be supported by evcc, but the LUNA, which is prepared for V2H and V2G UseCases is not. Is this support planned in the backlog or might it be supportet but not documented?

Best regards and thanks in advance
Ulf

premultiply, 2025-07-16:
If it supports OCPP and its OCPP implementation is not too broken it might be supported already like any other OCPP charger.

marcelGoerentz, 2025-07-16:
I think he wants to know if evcc supports V2H (vehicle to home) and V2G (vehicle to grid) charging.
And this is, as far as I know, not supported right now!

tclas, 2025-07-16:
@marcelGoerentz : This was not my question. The "BiDi-Ready"-Feature was one of the criterias for the wallbox, which is provided by the LUNA. My intention for opening this issue was, that I want to know, if this wallbox can be used in combination with evcc, beacause I want to use it for powering my vehicle by pv-surplus.