evcc-io/evcc #22423: EVCC doesn't start the charging

tasisale, 2025-07-17:
### Describe the bug

i will see diverent WARN in Log, but it will only start i change to Schnell(Quick)
--
````
[lp-1 ] WARN 2025/07/17 14:39:28 charger logic error: current mismatch (got 4A, expected 15A) - make sure your interval is at least 30s

````

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
evcc dump --cfg
[main  ] INFO 2025/07/17 14:45:20 using config file: /etc/evcc.yaml
[main  ] FATAL 2025/07/17 14:45:20
Found systemd service database at "/var/lib/evcc/evcc.db", evcc has been invoked                                                         with no explicit database path.
Running the same config with multiple databases can lead to expiring vehicle tok                                                        ens.

If you want to use the existing service database run the following command:

sudo evcc --database /var/lib/evcc/evcc.db

If you want to create a new user-space database run the following command:

evcc --database ~/.evcc/evcc.db

If you know what you're doing, you can skip the database check with the --ignore                                                        -db flag.
```

### Log details

```text
[lp-1 ] WARN 2025/07/17 14:36:30 charger logic error: phases mismatch (got 3, expected 1)

Easee-Home Tür: charger logic error: current mismatch (got 4A, expected 15A) - make sure your interval is at least 30s
--
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.205.0

naltatis, 2025-07-17:
Please provide a proper configuration. You've posted a warning message instead of your config.

The most likely cause of your error is, that you've configured 4A as min current which will not work for EV charging. The minimum is 6A. See also https://docs.evcc.io/en/docs/features/solar-charging

closed 2025-07-17 by naltatis

tasisale, 2025-07-17:
i have 6A as min and 16A as max.

[image]

naltatis, 2025-07-17:
Then you need to check your charger. It reports 4A. But without config and proper logs we're guessing here.