evcc-io/evcc #22425: Solar Mode stoppt nicht

meniku, 2025-07-17:
### Describe the bug

Seit ich von v 0.1xx auf 0.20x geupdated habe, stoppt der Solar mode nicht mehr. (einmal hab ich es erst bemerkt als der Hausakku nur noch 1% hatte, trotzdem hat es fröhlich weiter gemacht).
Aktuell bin ich bei 0.205, aber es trat auch schon bei zwei weiteren Versionen auf. 

Mit version 0.1xx hatte ich keine solche Probleme. Die Konfiguration habe ich nicht angepasst

Hier ein Beispiel wo es nicht gestoppt hat, obwohl hier der Schwellwert vom batterieunterstützten Laden bereits erreicht war:

[image]

[image]

[image]

### Steps to reproduce

1. Wait for Solar Mode to trigger
2. See that it never stops

### Configuration details

```yaml
evcc.yaml

# open evcc at http://evcc.local:7070/
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: XXXXXX

interval: 30s # control cycle interval

meters:
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.178.74  
  port: 1502  
  usage: grid  
  modbus: tcpip
  name: grid1
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.178.74  
  port: 1502  
  usage: pv  
  modbus: tcpip
  name: pv2
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.178.74  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery3

vehicles:
- type: template
  template: renault 
  title: Zoe  
[... 900 lines cut ...]
[site  ] DEBUG 2025/07/16 17:03:38 site power: 2633W
[lp-1  ] DEBUG 2025/07/16 17:03:38 detected active phases: 3p
[lp-1  ] DEBUG 2025/07/16 17:03:38 charge total import: 3573.967kWh
[lp-1  ] DEBUG 2025/07/16 17:03:38 charger status: C
[lp-1  ] DEBUG 2025/07/16 17:03:38 soc estimated: 74.34% (vehicle: 73.00%)
[lp-1  ] DEBUG 2025/07/16 17:03:38 vehicle soc: 74%
[lp-1  ] DEBUG 2025/07/16 17:03:38 vehicle range: 234km
[lp-1  ] DEBUG 2025/07/16 17:03:38 pv charge current: 5.18A = 9A + -3.82A (2633W @ 3p)
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

 0.205

andig, 2025-07-17:
> pv charge current: min 9A > 3.8A (3589W @ 3p, battery: true)

Wo kommt der Minimalstrom her?

meniku, 2025-07-18:
so sieht es aus wenn ich den service neu starte
[evcc-start.log](https://github.com/user-attachments/files/21318701/evcc-start.log)

meniku, 2025-07-18:
ja ist identisch

VolkerK62, 2025-07-18:
unabhängig von den Datenbank Hinweisen
Du hast wohl noch eine yaml   `using config file: /root/evcc.yaml`

meniku, 2025-07-18:
ja das stimmt, die ist ziemlich alt. Denke nicht dass die tatsächlich benutzt wird. Ich lösche sie mal und schau nochmal

mfuchs1984, 2025-07-19:
Wenn es das nächste mal passiert, stell hier bitte die Ausgabe von http://evcc:7070/api/state zur Verfügung. Damit lässt sich hoffentlich rekonstruieren, welche Bedingung zur Nichtabschaltung führt.

dkc-services, 2025-07-19:
I have the exact same issue. I think it started after upgrading to 0.205 . It worked before that.

mfuchs1984, 2025-07-19:
@dkc-services then please add a log file and the output of api/state when it happens the next time.

dkc-services, 2025-07-19:
> [@dkc-services](https://github.com/dkc-services) then please add a log file and the output of api/state when it happens the next time.

```
[site ] DEBUG 2025/07/19 20:36:20 site power: 326W

[lp-1 ] DEBUG 2025/07/19 20:36:23 set charge mode: pv
[site ] DEBUG 2025/07/19 20:36:23 ----

[lp-1 ] DEBUG 2025/07/19 20:36:23 charge power: 0W
[lp-1 ] DEBUG 2025/07/19 20:36:23 charge currents: [0 0 0]A
[site ] DEBUG 2025/07/19 20:36:23 grid power: -7W
[site ] DEBUG 2025/07/19 20:36:23 battery 1 power: 332W
[site ] DEBUG 2025/07/19 20:36:23 pv 1 power: 257W
[site ] DEBUG 2025/07/19 20:36:23 battery 1 soc: 87%
[site ] DEBUG 2025/07/19 20:36:23 grid currents: [0.59 1.27 1.58]A
[site ] DEBUG 2025/07/19 20:36:23 site power: 325W
[lp-1 ] DEBUG 2025/07/19 20:36:23 charge voltages: [228 228 227]V
[lp-1 ] DEBUG 2025/07/19 20:36:23 charge total import: 16372.662kWh
[lp-1 ] DEBUG 2025/07/19 20:36:23 charger status: B
[lp-1 ] DEBUG 2025/07/19 20:36:23 pv charge current: min 6A > 0A (325W @ 1p, battery: true)
[lp-1 ] DEBUG 2025/07/19 20:36:24 set charge current limit: 6A
[lp-1 ] DEBUG 2025/07/19 20:36:24 charger enable
[lp-1 ] DEBUG 2025/07/19 20:36:24 wake-up timer: start
[site ] DEBUG 2025/07/19 20:36:25 ----
[lp-2 ] DEBUG 2025/07/19 20:36:25 charge power: 0W
[lp-1 ] DEBUG 2025/07/19 20:36:25 charge power: 182W
[lp-1 ] DEBUG 2025/07/19 20:36:25 charge currents: [0.245 0 0]A
[site ] DEBUG 2025/07/19 20:36:25 grid power: -17W
[site ] DEBUG 2025/07/19 20:36:25 pv 1 power: 232W
[site ] DEBUG 2025/07/19 20:36:25 battery 1 power: 396W
[site ] DEBUG 2025/07/19 20:36:25 grid currents: [0.6 1.28 1.59]A
[site ] DEBUG 2025/07/19 20:36:25 battery 1 soc: 87%
[site ] DEBUG 2025/07/19 20:36:25 site power: 379W
[lp-1 ] DEBUG 2025/07/19 20:36:26 charge voltages: [228 228 227]V
[lp-1 ] DEBUG 2025/07/19 20:36:26 charge total import: 16372.662kWh
[lp-1 ] DEBUG 2025/07/19 20:36:26 charger status: C
[lp-1 ] DEBUG 2025/07/19 20:36:26 charge start energy: 16372.662kWh
[lp-1 ] INFO 2025/07/19 20:36:26 start charging ->
[lp-1 ] DEBUG 2025/07/19 20:36:26 wake-up timer: stop
[lp-1 ] DEBUG 2025/07/19 20:36:26 soc estimated: 61.12% (vehicle: 61.00%)
[lp-1 ] DEBUG 2025/07/19 20:36:26 vehicle soc: 61%
[lp-1 ] DEBUG 2025/07/19 20:36:26 vehicle soc limit: 80%
[lp-1 ] DEBUG 2025/07/19 20:36:26 vehicle range: 223km
[lp-1 ] DEBUG 2025/07/19 20:36:26 pv charge current: min 6A > 0.597A (379W @ 1p, battery: true)
[site ] DEBUG 2025/07/19 20:36:30 ----

[lp-1 ] DEBUG 2025/07/19 20:36:30 charge power: 0W
[lp-1 ] DEBUG 2025/07/19 20:36:30 charge currents: [0.472 0 0]A
[site ] DEBUG 2025/07/19 20:36:30 grid power: -11W
[site ] DEBUG 2025/07/19 20:36:30 pv 1 power: 257W
[site ] DEBUG 2025/07/19 20:36:30 battery 1 power: 348W
[site ] DEBUG 2025/07/19 20:36:30 grid currents: [0.59 1.29 1.59]A
[site ] DEBUG 2025/07/19 20:36:30 battery 1 soc: 87%
[site ] DEBUG 2025/07/19 20:36:30 site power: 337W

[site ] DEBUG 2025/07/19 20:36:35 ----

[lp-1 ] DEBUG 2025/07/19 20:36:35 charge power: 1354W
[lp-1 ] DEBUG 2025/07/19 20:36:35 charge currents: [6 0 0]A
[site ] DEBUG 2025/07/19 20:36:35 grid power: -4W
[site ] DEBUG 2025/07/19 20:36:35 pv 1 power: 268W
[site ] DEBUG 2025/07/19 20:36:35 battery 1 power: 1610W
[site ] DEBUG 2025/07/19 20:36:35 grid currents: [6.05 1.28 1.58]A
[site ] DEBUG 2025/07/19 20:36:35 battery 1 soc: 87%
[site ] DEBUG 2025/07/19 20:36:35 site power: 1606W
[lp-1 ] DEBUG 2025/07/19 20:36:36 charge voltages: [227 228 227]V
[lp-1 ] DEBUG 2025/07/19 20:36:36 detected active phases: 1p
[lp-1 ] DEBUG 2025/07/19 20:36:36 charge total import: 16372.663kWh
[lp-1 ] DEBUG 2025/07/19 20:36:36 charger status: C
[lp-1 ] DEBUG 2025/07/19 20:36:36 soc estimated: 61.12% (vehicle: 61.00%)
[lp-1 ] DEBUG 2025/07/19 20:36:36 vehicle soc: 61%
[lp-1 ] DEBUG 2025/07/19 20:36:36 vehicle soc limit: 80%
[lp-1 ] DEBUG 2025/07/19 20:36:36 vehicle range: 223km
[lp-1 ] DEBUG 2025/07/19 20:36:36 pv charge current: min 6A > 0A (1606W @ 1p, battery: true)
[site ] DEBUG 2025/07/19 20:36:40 ----
[lp-1 ] DEBUG 2025/07/19 20:36:40 charge power: 1353W
[lp-1 ] DEBUG 2025/07/19 20:36:40 charge currents: [5.95 0 0]A
[site ] DEBUG 2025/07/19 20:36:40 grid power: 7W
[site ] DEBUG 2025/07/19 20:36:40 pv 1 power: 189W
[site ] DEBUG 2025/07/19 20:36:40 battery 1 power: 1745W
[site ] DEBUG 2025/07/19 20:36:40 battery 1 soc: 87%
[site ] DEBUG 2025/07/19 20:36:40 grid currents: [6 1.27 1.58]A
[site ] DEBUG 2025/07/19 20:36:40 site power: 1752W
[site ] DEBUG 2025/07/19 20:36:45 ----
[lp-2 ] DEBUG 2025/07/19 20:36:45 charge power: 0W
[lp-1 ] DEBUG 2025/07/19 20:36:45 charge power: 1342W
[lp-1 ] DEBUG 2025/07/19 20:36:45 charge currents: [5.98 0 0]A
[site ] DEBUG 2025/07/19 20:36:45 grid power: -4W
[site ] DEBUG 2025/07/19 20:36:45 pv 1 power: 195W
[site ] DEBUG 2025/07/19 20:36:45 battery 1 power: 1744W
[site ] DEBUG 2025/07/19 20:36:45 battery 1 soc: 87%
[site ] DEBUG 2025/07/19 20:36:45 grid currents: [6.03 1.27 1.57]A
[site ] DEBUG 2025/07/19 20:36:45 site power: 1740W
[lp-1 ] DEBUG 2025/07/19 20:36:46 charge voltages: [227 228 227]V
[lp-1 ] DEBUG 2025/07/19 20:36:46 detected active phases: 1p
[lp-1 ] DEBUG 2025/07/19 20:36:46 charge total import: 16372.667kWh
[lp-1 ] DEBUG 2025/07/19 20:36:46 charger status: C
[lp-1 ] DEBUG 2025/07/19 20:36:46 soc estimated: 61.12% (vehicle: 61.00%)
[lp-1 ] DEBUG 2025/07/19 20:36:46 vehicle soc: 61%
[lp-1 ] DEBUG 2025/07/19 20:36:46 vehicle soc limit: 80%
[lp-1 ] DEBUG 2025/07/19 20:36:46 vehicle range: 223km
```

removed lp-2 logs for convenience

mfuchs1984, 2025-07-19:
[lp-1 ] DEBUG 2025/07/19 20:36:36 pv charge current: min 6A > 0A (1606W @ 1p, battery: true)

You are within your configure battery supported charging range.

mfuchs1984, 2025-07-19:
This seems to be off-topic here. Create a discussion with the details in case you need help.

dkc-services, 2025-07-19:
> This seems to be off-topic here. Create a discussion with the details in case you need help.

Meanwhile was able to verify this. 

I didn’t realize this as my home battery hasn’t been under 80 a lot. 

So I can confirm the cause is identical

andig, 2025-07-20:
Seems this has been clarified. I'm wondering why we're not seeing the battery-supported charging displayed in the UI @naltatis ?

mfuchs1984, 2025-07-20:
@andig bei @meniku, dem issue Ersteller, ist es meines Erachtens noch nicht klar. Nur beim Log, das @dkc-services bereitgestellt hatte, war battery supported charging aktiv.

meniku, 2025-07-20:
How habe es reproduziert. Habe die Werte etwas angepasst

[image]

und lädt fröhlich weiter

Hier der result vom API state @mfuchs1984 [](url)

> {"result":{"aux":[],"battery":[{"power":1283,"capacity":0,"soc":81.1111068725586,"controllable":true}],"batteryCapacity":1,"batteryDischargeControl":false,"batteryEnergy":0,"batteryGridChargeActive":false,"batteryGridChargeLimit":-0.005,"batteryMode":"unknown","batteryPower":1283,"batterySoc":81.111,"bufferSoc":90,"bufferStartSoc":95,"currency":"EUR","eebus":false,"ext":[],"forecast":{"grid":[{"start":"2025-07-20T00:00:00+02:00","end":"2025-07-20T01:00:00+02:00","value":0.3232},{"start":"2025-07-20T01:00:00+02:00","end":"2025-07-20T02:00:00+02:00","value":0.3204},{"start":"2025-07-20T02:00:00+02:00","end":"2025-07-20T03:00:00+02:00","value":0.3208},{"start":"2025-07-20T03:00:00+02:00","end":"2025-07-20T04:00:00+02:00","value":0.334},{"start":"2025-07-20T04:00:00+02:00","end":"2025-07-20T05:00:00+02:00","value":0.3356},{"start":"2025-07-20T05:00:00+02:00","end":"2025-07-20T06:00:00+02:00","value":0.3315},{"start":"2025-07-20T06:00:00+02:00","end":"2025-07-20T07:00:00+02:00","value":0.3302},{"start":"2025-07-20T07:00:00+02:00","end":"2025-07-20T08:00:00+02:00","value":0.3104},{"start":"2025-07-20T08:00:00+02:00","end":"2025-07-20T09:00:00+02:00","value":0.3072},{"start":"2025-07-20T09:00:00+02:00","end":"2025-07-20T10:00:00+02:00","value":0.2676},{"start":"2025-07-20T10:00:00+02:00","end":"2025-07-20T11:00:00+02:00","value":0.2253},{"start":"2025-07-20T11:00:00+02:00","end":"2025-07-20T12:00:00+02:00","value":0.217},{"start":"2025-07-20T12:00:00+02:00","end":"2025-07-20T13:00:00+02:00","value":0.2145},{"start":"2025-07-20T13:00:00+02:00","end":"2025-07-20T14:00:00+02:00","value":0.2145},{"start":"2025-07-20T14:00:00+02:00","end":"2025-07-20T15:00:00+02:00","value":0.2145},{"start":"2025-07-20T15:00:00+02:00","end":"2025-07-20T16:00:00+02:00","value":0.2254},{"start":"2025-07-20T16:00:00+02:00","end":"2025-07-20T17:00:00+02:00","value":0.297},{"start":"2025-07-20T17:00:00+02:00","end":"2025-07-20T18:00:00+02:00","value":0.3249},{"start":"2025-07-20T18:00:00+02:00","end":"2025-07-20T19:00:00+02:00","value":0.3338},{"start":"2025-07-20T19:00:00+02:00","end":"2025-07-20T20:00:00+02:00","value":0.3478},{"start":"2025-07-20T20:00:00+02:00","end":"2025-07-20T21:00:00+02:00","value":0.3526},{"start":"2025-07-20T21:00:00+02:00","end":"2025-07-20T22:00:00+02:00","value":0.3514},{"start":"2025-07-20T22:00:00+02:00","end":"2025-07-20T23:00:00+02:00","value":0.3512},{"start":"2025-07-20T23:00:00+02:00","end":"2025-07-21T00:00:00+02:00","value":0.3395},{"start":"2025-07-21T00:00:00+02:00","end":"2025-07-21T01:00:00+02:00","value":0.3182},{"start":"2025-07-21T01:00:00+02:00","end":"2025-07-21T02:00:00+02:00","value":0.3094},{"start":"2025-07-21T02:00:00+02:00","end":"2025-07-21T03:00:00+02:00","value":0.3139},{"start":"2025-07-21T03:00:00+02:00","end":"2025-07-21T04:00:00+02:00","value":0.3137},{"start":"2025-07-21T04:00:00+02:00","end":"2025-07-21T05:00:00+02:00","value":0.3139},{"start":"2025-07-21T05:00:00+02:00","end":"2025-07-21T06:00:00+02:00","value":0.329},{"start":"2025-07-21T06:00:00+02:00","end":"2025-07-21T07:00:00+02:00","value":0.3501},{"start":"2025-07-21T07:00:00+02:00","end":"2025-07-21T08:00:00+02:00","value":0.3651},{"start":"2025-07-21T08:00:00+02:00","end":"2025-07-21T09:00:00+02:00","value":0.362},{"start":"2025-07-21T09:00:00+02:00","end":"2025-07-21T10:00:00+02:00","value":0.345},{"start":"2025-07-21T10:00:00+02:00","end":"2025-07-21T11:00:00+02:00","value":0.3407},{"start":"2025-07-21T11:00:00+02:00","end":"2025-07-21T12:00:00+02:00","value":0.3322},{"start":"2025-07-21T12:00:00+02:00","end":"2025-07-21T13:00:00+02:00","value":0.3156},{"start":"2025-07-21T13:00:00+02:00","end":"2025-07-21T14:00:00+02:00","value":0.3118},{"start":"2025-07-21T14:00:00+02:00","end":"2025-07-21T15:00:00+02:00","value":0.3166},{"start":"2025-07-21T15:00:00+02:00","end":"2025-07-21T16:00:00+02:00","value":0.3192},{"start":"2025-07-21T16:00:00+02:00","end":"2025-07-21T17:00:00+02:00","value":0.3228},{"start":"2025-07-21T17:00:00+02:00","end":"2025-07-21T18:00:00+02:00","value":0.3387},{"start":"2025-07-21T18:00:00+02:00","end":"2025-07-21T19:00:00+02:00","value":0.3405},{"start":"2025-07-21T19:00:00+02:00","end":"2025-07-21T20:00:00+02:00","value":0.357},{"start":"2025-07-21T20:00:00+02:00","end":"2025-07-21T21:00:00+02:00","value":0.3612},{"start":"2025-07-21T21:00:00+02:00","end":"2025-07-21T22:00:00+02:00","value":0.3574},{"start":"2025-07-21T22:00:00+02:00","end":"2025-07-21T23:00:00+02:00","value":0.344},{"start":"2025-07-21T23:00:00+02:00","end":"2025-07-22T00:00:00+02:00","value":0.3336}],"planner":[{"start":"2025-07-20T00:00:00+02:00","end":"2025-07-20T01:00:00+02:00","value":0.3232},{"start":"2025-07-20T01:00:00+02:00","end":"2025-07-20T02:00:00+02:00","value":0.3204},{"start":"2025-07-20T02:00:00+02:00","end":"2025-07-20T03:00:00+02:00","value":0.3208},{"start":"2025-07-20T03:00:00+02:00","end":"2025-07-20T04:00:00+02:00","value":0.334},{"start":"2025-07-20T04:00:00+02:00","end":"2025-07-20T05:00:00+02:00","value":0.3356},{"start":"2025-07-20T05:00:00+02:00","end":"2025-07-20T06:00:00+02:00","value":0.3315},{"start":"2025-07-20T06:00:00+02:00","end":"2025-07-20T07:00:00+02:00","value":0.3302},{"start":"2025-07-20T07:00:00+02:00","end":"2025-07-20T08:00:00+02:00","value":0.3104},{"start":"2025-07-20T08:00:00+02:00","end":"2025-07-20T09:00:00+02:00","value":0.30 [... cut ...] 00026,"solarPercentage":57.19959069112406},"total":{"avgCo2":0,"avgPrice":0.21236517105168962,"chargedKWh":3164.549169252065,"solarPercentage":55.77342933780538}},"tariffFeedIn":0.08,"tariffGrid":0.297,"tariffPriceHome":0.08,"tariffPriceLoadpoints":0.083,"vehicles":{"ev4":{"title":"Zoe","icon":"car","capacity":52,"phases":3,"minCurrent":9,"maxCurrent":16,"repeatingPlans":[{"weekdays":[1,2,3,4,5],"time":"06:30","tz":"Europe/Zurich","soc":25,"precondition":0,"active":true}]}},"version":"0.205.0"}}

mfuchs1984, 2025-07-21:
Auch ganz unabhängig vom Vorzeichen passen die Werte überhaupt nicht, sowohl die powers zu power als auch die currents zu den powers.

stopcoder, 2025-07-24:
Ich habe genau dasselbe Problem mit Version 205. Ich kann aber nicht sagen, mit welcher Version hat es funktioniert. Heute Morgen konnte ich es erneut reproduzieren.

In EVCC habe ich folgende Einstellungen für den Speicher konfiguriert: Der Speicher soll zunächst bis 90 % geladen werden. Erst danach soll das Auto geladen werden.

[image]

Der SoC des Speichers lag bei etwa 50 %, und die Photovoltaikleistung betrug bereits über 4,1 kW. Dennoch wurde die Ladeleistung auf mindestens 4,1 kW gehalten, und der restliche Strom floss in den Speicher. Eigentlich sollte die gesamte Photovoltaikleistung zunächst vollständig in den Speicher geladen werden, bis dieser einen SoC von 90 % erreicht.

Wenn die Photovoltaikleistung unter 4,1 kW liegt, bleibt die Ladeleistung trotzdem auf mindestens 4,1 kW. Die Differenz wird dann aus dem Speicher entnommen – selbst wenn der SoC noch unter 90 % liegt.

Hier ist ein Screenshot von EVCC:

[image]

Und hier das JSON-Objekt von der Schnittstelle /api/state:

```json
{
    "result": {
        "aux": [],
        "battery": [
            {
                "power": -407,
                "capacity": 0,
                "soc": 50,
                "controllable": true
            }
        ],
        "batteryCapacity": 1,
        "batteryDischargeControl": true,
        "batteryEnergy": 0,
        "batteryGridChargeActive": false,
        "batteryMode": "unknown",
        "batteryPower": -407,
        "batterySoc": 50,
        "bufferSoc": 95,
        "bufferStartSoc": 95,
        "currency": "EUR",
        "eebus": false,
        "ext": [],
        "forecast": {},
        "greenShareHome": 1,
        "greenShareLoadpoints": 0.985,
        "grid": {
            "power": 63,
            "energy": 31767.7
        },
        "gridConfigured": true,
        "hems": {},
        "homePower": 374,
        "influx": {
            "url": "",
            "database": "",
            "token": "",
            "org": "",
            "user": "",
            "password": "",
            "insecure": false
        },
        "interval": 30,
        "loadpoints": [
            {
                "batteryBoost": false,
                "chargeCurrents": [
                    5.89,
                    5.85,
                    6.07
                ],
                "chargeDuration": 570,
                "chargePower": 4148,
[... 131 lines cut ...]
                "chargedKWh": 293.6999999999998,
                "solarPercentage": 97.35241206334825
            },
            "total": {
                "avgCo2": 0,
                "avgPrice": 0.008113381204748684,
                "chargedKWh": 534.9999999999997,
                "solarPercentage": 98.3238689817338
            }
        },
        "tariffGrid": 0.295,
        "tariffPriceHome": 0,
        "tariffPriceLoadpoints": 0.004,
        "vehicles": {
            "ID.4": {
                "title": "Blue Crystal",
                "icon": "car",
                "capacity": 77,
                "repeatingPlans": []
            }
        },
        "version": "0.205.0"
    }
}
```

VolkerK62, 2025-07-24:
@stopcoder 
das ist eine mögliche Fehlerursache `  "residualPower": 0,`

Stell mal Residualpower auf mindestens +100.

mfuchs1984, 2025-07-24:
Ok, der Timer läuft los, wir dann aber plötzlich zurückgesetzt, ich schau mal weiter
```
Jul 20 15:51:37 evcc evcc[693]: [lp-1  ] DEBUG 2025/07/20 15:51:37 pv disable in 2m30s
Jul 20 15:51:37 evcc evcc[693]: [lp-1  ] DEBUG 2025/07/20 15:51:37 pv disable timer remaining: 2m30s

2025/07/20 15:52:07 pv charge current: 4.57A = 9A + -4.43A (3057W @ 3p)
Jul 20 15:52:07 evcc evcc[693]: [lp-1  ] DEBUG 2025/07/20 15:52:07 pv disable timer reset
```

mfuchs1984, 2025-07-24:
Muss an der projectedSitePower liegen. Ich versuche nachzuvollziehen welchen Wert die annimmt.

mfuchs1984, 2025-07-24:
Ok:

```
Jul 20 15:51:07 evcc evcc[693]: [lp-1  ] DEBUG 2025/07/20 15:51:07 pv charge current: 2.35A = 9A + -6.65A (4585W @ 3p)
Jul 20 15:51:07 evcc evcc[693]: [lp-1  ] DEBUG 2025/07/20 15:51:07 projected site power 445W >= 0W disable threshold
```
```
Jul 20 15:51:37 evcc evcc[693]: [lp-1  ] DEBUG 2025/07/20 15:51:37 pv charge current: 2.19A = 9A + -6.81A (4702W @ 3p)
Jul 20 15:51:37 evcc evcc[693]: [lp-1  ] DEBUG 2025/07/20 15:51:37 projected site power 562W >= 0W disable threshold
```
```
Jul 20 15:53:37 evcc evcc[693]: [lp-1  ] DEBUG 2025/07/20 15:53:37 pv charge current: 2.72A = 9A + -6.28A (4336W @ 3p)
Jul 20 15:53:37 evcc evcc[693]: [lp-1  ] DEBUG 2025/07/20 15:53:37 projected site power 196W >= 0W disable threshold
```

Es liegen jeweils 4140W zwischen site power (4XXXW @ 3p) und projected site power in der nächsten Zeile, woher kommt das:

[image]

230 V * 9 A * 2 = 4140W 

Es wird aus irgendeinem Grund davon ausgegangen, dass eine Phasenumschaltung stattfindet.

mfuchs1984, 2025-07-24:
@meniku hast du davor irgendwann battery boost verwendet?

mfuchs1984, 2025-07-24:
@andig ich denke das Problem ist diese Zeile in `func (lp *Loadpoint) boostPower(batteryBoostPower float64)`:
https://github.com/evcc-io/evcc/blob/ba29e4c16ec50eb436fbdd34266c9f35c7755f6c/core/loadpoint.go#L1352

Hier wird der phaseTimer auf "elapsed" gesetzt, was nicht "Zero" ist. Das passiert an der Stelle auch, wenn der charger gar kein Phase switching unterstützt.

meniku, 2025-07-24:
Ja den boost hab ich schon ab und zu benutzt

andig, 2025-07-24:
Die Frage hier ist: war der Boost zum Zeitpunkt des Problems aktiv?

meniku, 2025-07-24:
Nein, zumindest ist der Boost laut UI immer nur für die Session aktiv, in der er aktiviert wird. Und im Screenshot ganz oben ist es auch ersichtlich, dass der Boost in der UI deaktiviert war.
Ob da jetzt irgendein Backend State nicht korrekt zurückgesetzt wird, kann ich nicht beurteilen.

mfuchs1984, 2025-07-24:
> Die Frage hier ist: war der Boost zum Zeitpunkt des Problems aktiv?

Nein, es reicht aber soweit ich es verstehe, wenn dieser vorher aktiv war, da dann der Phase timer nicht mehr Zero ist.

closed 2025-07-25 by andig

meniku, 2025-07-25:
Danke vielmals :-)

meniku, 2025-07-30:
behebt das auch das Problem mit dem Boost Modus, dass der Solarmode nicht stoppt, wenn der Hausakku leer ist?