evcc-io/evcc #22431: Zusatzeinstellung für "Batterieunterstützes Laden" (min / max)

michiproep, 2025-07-18:
Wenn sich mein Akku im oberen Prozentbereich (Batterieunterstützes Laden) befindet und dann zu wenig von Dach kommt, wird zwar wie erwartet weitergeladen, aber soweit ich das erkennen kann immer mit der Einstellung "Min. Ladestrom" (pro Loadpoint).

Es wäre cool, wenn man pro Ladepunkt einstellen könnte, ob im in diesem Fall mit max. oder min. weitergeladen werden soll.

Im Fall vom Auto wäre (zumindest bei mir) "min" die richtige Einstellung, bei kleineren Ladepunkten, die sowieso nur zwischen 1kW und 3kw ziehen, würde ich gerne mit max weiterladen.

Natürlich immer unter der Voraussetzung, dass die max. Hybridleistung des WR nicht überschritten wird.

Dadurch könnte ich meinen Eigenverbrauch deutlich erhöhen (hab aktuell 5 loadpoints)

Denkt ihr, das wäre sinnvoll/machbar?

andig, 2025-07-18:
> Es wäre cool, wenn man pro Ladepunkt einstellen könnte, ob im in diesem Fall mit max. oder min. weitergeladen werden soll.

Funktioniert der experimentelle Battery Boost bei Dir? Problem der Batterie ist *IMMER* dass es dazu keine Leistungserwartung  gibt und evcc nicht von sich "mehr Batterie" anfordern kann.

premultiply, 2025-07-18:
Batterieunterstütztes Laden verhindert "nur" die Unterbrechung des Ladevorgangs bei zu wenig PV-Überschuss.
Schnelle Wolkenwechsel oder taktende, größere Verbraucher im Haushalt sind ein typischer Anwendungsfall.

Ist quasi ein temporärer Min+PV-Modus so lange der Speicher genug Vorrat hat.

Das was du suchst ist "Battery Boost". Das findest du unter den experimentellen Funktionen.

closed 2025-07-18 by premultiply

cschlipf, 2025-07-18:
Was nett wäre beim Battery Boost wäre eine untere SoC Grenze ab der er sich wieder abschaltet.

michiproep, 2025-07-18:
So, kurzen Test gemacht. Das ist leider nicht das was ich suche.
Wenn ich den an habe, dann zieht er die Batterie leer ohne auf die Batterieeinstellung zu achten.
Wie gesagt, wäre für mich ein cooles Feature, aber meine Vorschläge werden dann meistens ein halbes Jahr später implementiert sobald es jemand anders vorschlägt ( ;-) kleiner Spaß am Rande)

Kleiner Hinweis/Bug: Der Boost Schalter wird bei mir nur auf dem Desktop, nicht aber in der mobilen Ansicht (iphone mini und iPad mini) gezeigt.

premultiply, 2025-07-18:
> Der Boost Schalter wird bei mir nur auf dem Desktop, nicht aber in der mobilen Ansicht (iphone mini und iPad mini) gezeigt.

Die experimentellen Einstellungen gelten pro Gerät.

mfuchs1984, 2025-07-19:
> Wenn ich den an habe, dann zieht er die Batterie leer ohne auf die Batterieeinstellung zu achten.

Dazu gibt's schon https://github.com/evcc-io/evcc/issues/21318