evcc-io/evcc #22437: Batterie Netzladen: keine CO2 Auswahl

VolkerK62, 2025-07-18:
Hallo, ich habe Version 0.205.0 installiert und wollte das Netzladen testen. Grundsätzlich funktioniert meine aktive Batteriesteuerung. Nur wenn ich die Batterie Netzladen will bekomme ich bei CO2 als Grenzwert nur 0 angeboten. 

_Originally posted by @MikeRow1972 in https://github.com/evcc-io/evcc/discussions/22427_

naltatis, 2025-07-18:
Ich schau mal rein ...

andig, 2025-07-18:
Hier fehlt die Config für die Bestätigung. These: falscher Tarif.

VolkerK62, 2025-07-18:
> These: falscher Tarif.

Wenn es doch im Loadpoint funktioniert????

[image]

andig, 2025-07-18:
@naltatis GSI ist Typ co2, daran kann es also nicht liegen.