evcc-io/evcc #22442: Support for Zendure SolarFlow 2400 AC Series

JonnyJST, 2025-07-18:
Please add support for the Zendure SolarFlow 2400 AC Series. The goal is to integrate this power station with EVCC for better charging control and energy management.

andig, 2025-07-18:
What's missing?

remb0, 2025-09-30:
I think he mean:   there is support for a zendure hyper 2000  but in the dropdown list the ac2400 is not listed (both are different in communication)

jannehoffmann, 2025-10-05:
I'm considering buying Zendure SolarFlow 2400 AC. Is it right, that evcc doesn't support it yet?
Does the template for Hyper 2000 not work with SolarFlow 2400?

AndySchoe, 2025-10-31:
@andig  what is needed to contribute a template?
as I own a SolarFlow 2400 AC now I would like to have this integrated into EVCC.

Unfortunately I am not knowledgeable/experienced enough to do the implementation via configuration.
Is there something to support the integration into EVCC?

andig, 2025-11-01:
All you need is a text editor.

andig, 2025-11-01:
Sure. Not sure i can comment sind I don‘t know any of its api. Http? Look t templates using the http plugin.

MvsBluewin, 2025-12-10:
I have a Zendure Solarflow 2400 battery and it works perfectly with evcc with the following simple additions to the evcc.yaml file:

Under 'meters' add:  (change the IP addresses to the one you have for your Zendure, change the name and battery capacity.

- name: battery1
  type: custom
  power:
    source: calc
    add:
    - source: http
      uri: http://10.128.0.22/properties/report
      jq: .properties.packInputPower
      scale: 1
    - source: http
      uri: http://10.128.0.22/properties/report
      jq: .properties.outputPackPower
      scale: -1
  soc:
    source: http
    uri: http://10.128.0.22/properties/report
    jq: .properties.electricLevel
  capacity: 5.760

then add to 'site' 'meters' : 

battery: battery1 (or the name you have chosen)

Make sure the spacings in the yaml are correct.