evcc-io/evcc #22455: minpv (Solar and Battery supported charging) seems not to work with an OCPP based charger

olger, 2025-07-19:
### Describe the bug

In my setup, I like to use the solar / battery supported charging but the charging seems to start and stop frequently - even when there is enough capacity available in the battery. 

### Steps to reproduce

1. Setup with 
  1.1 an OCPP based charger (Blink EQ200 -eNovates based like a lot of the Shell Recharge series) 
  1.2 SolarEdge PV and Battery (Hybrid) 
2. Car is a Tesla but I am now testing with the Guest car to keep that out of equation
3. Plug in the Car. This will automatically start a transaction. 
4. UI States: "Ready. Waiting for vehicle…" Charging starts and stops stating "Charging equipment not ready". 
5. This after a while, this happens again. 

I've tried changes with phase settings and timer timeout (that is 60s now). Also happy to help out in more detail (I know go, have built multiple charge point backends (including shell recharge)  in the past) 

Checked the nightly build but will do so today. 

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: controller.localdomain # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  site: trace
  lp-1: trace
  cache: error
  tesla: trace

sponsortoken: ***************

# unique installation id
plant: ********************

interval: 60s # control cycle interval

meters:
  - name: grid1
    type: template
    template: solaredge-hybrid
    id: 1
    host: 192.168.10.177
    port: 1502
    usage: grid
    modbus: tcpip
    timeout: 10s
    watchdog: 60s
  - name: pv2
    type: template
    template: solaredge-hybrid
    id: 1
    host: 192.168.10.177
    port: 1502
    usage: pv
    modbus: tcpip
    timeout: 10s
  - name: battery3
    type: template
    template: solaredge-hybrid
    id: 1
    host: 192.168.10.177
    port: 1502
    usage: battery
    modbus: tcpip
    timeout: 10s

vehicles:
  - name: ev4
    type: template
    template: tesla
    title: Tesseke de 3e
    phases: 3
    capacity: 80
    clientId: 29e91e8d-ffe3-4d91-88e8-c00a9e2670a7
    accessToken: ************
    refreshToken: **********

chargers:
  - name: eq200
    type: template
    template: ocpp
    stationid: ENOGEN22SB-*********
    idtag: FFFFFFFF
    connector: 1
site:
  title: Vlak42
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

loadpoints:
  - title: Garage
    charger: eq200
    vehicle: ev4
    phases: 3

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.25 # EUR/kWh
  feedin:
    type: fixed
    price: 0.12 # EUR/kWh
```

### Log details

```text
[lp-1 ] INFO 2025/07/19 14:28:20 start charging ->
[lp-1 ] DEBUG 2025/07/19 14:28:20 wake-up timer: stop
[lp-1 ] WARN 2025/07/19 14:28:20 charger logic error: current mismatch (got 10A, expected 13.5A) - make sure your interval is at least 30s
[lp-1 ] DEBUG 2025/07/19 14:28:20 pv charge current: min 7A > 0A (4163W @ 1p, battery: true)
[lp-1 ] DEBUG 2025/07/19 14:28:21 set charge current limit: 7A
[site ] DEBUG 2025/07/19 14:28:59 ----
[lp-1 ] DEBUG 2025/07/19 14:28:59 charge power: 0W
[lp-1 ] DEBUG 2025/07/19 14:28:59 charge currents: [0 0 0]A
[site ] DEBUG 2025/07/19 14:28:59 battery 1 power: -0W
[site ] DEBUG 2025/07/19 14:28:59 grid power: -2966W
[site ] DEBUG 2025/07/19 14:28:59 battery 1 soc: 99%
[site ] DEBUG 2025/07/19 14:29:00 pv 1 power: 3164W
[site ] DEBUG 2025/07/19 14:29:00 grid powers: [-1039 -937 -990]W
[site ] DEBUG 2025/07/19 14:29:00 grid currents: [-0.966 -0.987 -0.69]A
[site ] DEBUG 2025/07/19 14:29:00 site power: -2966W
[lp-1 ] DEBUG 2025/07/19 14:29:00 charge voltages: [241 243 236]V
[lp-1 ] DEBUG 2025/07/19 14:29:00 charge total import: 6332.491kWh
[lp-1 ] DEBUG 2025/07/19 14:29:00 charger status: B
[lp-1 ] DEBUG 2025/07/19 14:29:00 charge final energy: 2.081kWh
[lp-1 ] INFO 2025/07/19 14:29:00 stop charging <-
[lp-1 ] DEBUG 2025/07/19 14:29:00 charge total import: 6332.491kWh
[lp-1 ] WARN 2025/07/19 14:29:00 charger out of sync: expected enabled, got disabled
[lp-1 ] DEBUG 2025/07/19 14:29:00 charger disable
[lp-1 ] DEBUG 2025/07/19 14:29:00 pv charge current: 12.9A = 0A + 12.9A (-2966W @ 1p)
[lp-1 ] DEBUG 2025/07/19 14:29:00 set charge current limit: 12.9A
[lp-1 ] DEBUG 2025/07/19 14:29:01 charger enable
[lp-1 ] DEBUG 2025/07/19 14:29:01 wake-up timer: start
[site ] DEBUG 2025/07/19 14:29:39 ----
[lp-1 ] DEBUG 2025/07/19 14:29:39 charge power: 0W
[lp-1 ] DEBUG 2025/07/19 14:29:39 charge currents: [0 0 0]A
[site ] DEBUG 2025/07/19 14:29:39 battery 1 power: -0W
[site ] DEBUG 2025/07/19 14:29:39 grid power: -2616W
[site ] DEBUG 2025/07/19 14:29:40 battery 1 soc: 99%
[site ] DEBUG 2025/07/19 14:29:40 pv 1 power: 2805W
[site ] DEBUG 2025/07/19 14:29:40 grid powers: [-923 -823 -869]W
[site ] DEBUG 2025/07/19 14:29:40 grid currents: [-0.966 -0.987 -0.69]A
[site ] DEBUG 2025/07/19 14:29:40 site power: -2616W
[lp-1 ] DEBUG 2025/07/19 14:29:40 charge voltages: [241 243 236]V
[lp-1 ] DEBUG 2025/07/19 14:29:40 charge total import: 6332.491kWh
[lp-1 ] DEBUG 2025/07/19 14:29:40 charger status: B
[lp-1 ] DEBUG 2025/07/19 14:29:40 pv charge current: 11.4A = 0A + 11.4A (-2616W @ 1p)
[lp-1 ] DEBUG 2025/07/19 14:29:40 set charge current limit: 11.4A
[site ] DEBUG 2025/07/19 14:30:19 ----
[lp-1 ] DEBUG 2025/07/19 14:30:19 charge power: 0W
[lp-1 ] DEBUG 2025/07/19 14:30:19 charge currents: [0 0 0]A
[site ] DEBUG 2025/07/19 14:30:19 battery 1 power: -0W
[site ] DEBUG 2025/07/19 14:30:19 grid power: -2792W
[site ] DEBUG 2025/07/19 14:30:19 battery 1 soc: 99%
[site ] DEBUG 2025/07/19 14:30:20 pv 1 power: 2981W
[site ] DEBUG 2025/07/19 14:30:20 grid powers: [-978 -880 -931]W
[site ] DEBUG 2025/07/19 14:30:20 grid currents: [-0.966 -0.987 -0.69]A
[site ] DEBUG 2025/07/19 14:30:20 site power: -2792W
[lp-1 ] DEBUG 2025/07/19 14:30:20 charge voltages: [241 243 236]V
[lp-1 ] DEBUG 2025/07/19 14:30:20 charge total import: 6332.491kWh
[lp-1 ] DEBUG 2025/07/19 14:30:20 charger status: B
[lp-1 ] WARN 2025/07/19 14:30:20 charger out of sync: expected enabled, got disabled
[lp-1 ] DEBUG 2025/07/19 14:30:20 charger disable
[lp-1 ] DEBUG 2025/07/19 14:30:20 pv charge current: 12.1A = 0A + 12.1A (-2792W @ 1p)
[lp-1 ] DEBUG 2025/07/19 14:30:21 set charge current limit: 12.1A
[lp-1 ] DEBUG 2025/07/19 14:30:22 charger enable
[lp-1 ] DEBUG 2025/07/19 14:30:22 wake-up timer: start
[site ] DEBUG 2025/07/19 14:30:59 ----
[lp-1 ] DEBUG 2025/07/19 14:30:59 charge power: 0W
[lp-1 ] DEBUG 2025/07/19 14:30:59 charge currents: [0 0 0]A
[site ] DEBUG 2025/07/19 14:30:59 battery 1 power: -0W
[site ] DEBUG 2025/07/19 14:30:59 grid power: -2875W
[site ] DEBUG 2025/07/19 14:30:59 battery 1 soc: 99%
[site ] DEBUG 2025/07/19 14:31:00 pv 1 power: 3128W
[site ] DEBUG 2025/07/19 14:31:00 grid powers: [-1031 -893 -986]W
[site ] DEBUG 2025/07/19 14:31:00 grid currents: [-0.966 -0.987 -0.69]A
[site ] DEBUG 2025/07/19 14:31:00 site power: -2875W
[lp-1 ] DEBUG 2025/07/19 14:31:00 charge voltages: [241 243 236]V
[lp-1 ] DEBUG 2025/07/19 14:31:00 charge total import: 6332.491kWh
[lp-1 ] DEBUG 2025/07/19 14:31:00 charger status: B
[lp-1 ] DEBUG 2025/07/19 14:31:00 pv charge current: 12.5A = 0A + 12.5A (-2875W @ 1p)
[lp-1 ] DEBUG 2025/07/19 14:31:00 set charge current limit: 12.5A

[lp-1 ] DEBUG 2025/07/19 14:28:20 pv charge current: min 7A > 0A (4163W @ 1p, battery: true)

Battery is seen but not taken into account ?
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.205.0

olger, 2025-07-19:
Instralled latest via apt-get: evcc 0.205.0 (0bbd3933)

`Jul 19 16:04:04 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:04 set charge mode: minpv
Jul 19 16:04:04 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:04:04 ----
Jul 19 16:04:04 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:04 charge power: 0W
Jul 19 16:04:04 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:04 charge currents: [0 0.24 0]A
Jul 19 16:04:04 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:04:04 grid power: -2766W
Jul 19 16:04:04 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:04:04 battery 1 power: -0W
Jul 19 16:04:04 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:04:04 battery 1 soc: 99%
Jul 19 16:04:04 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:04:04 pv 1 power: 2952W
Jul 19 16:04:04 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:04:04 grid powers: [-977 -866 -917]W
Jul 19 16:04:04 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:04:04 grid currents: [-0.966 -0.987 -0.69]A
Jul 19 16:04:04 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:04:04 site power: -2766W
Jul 19 16:04:04 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:04 charge voltages: [237 237 236]V
Jul 19 16:04:04 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:04 charge total import: 6332.979kWh
Jul 19 16:04:04 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:04 charger status: B
Jul 19 16:04:04 controller evcc[965246]: [lp-1  ] WARN 2025/07/19 16:04:04 ignoring inconsistent phases: 1p < 2p observed active
Jul 19 16:04:04 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:04 pv charge current: 12A = 0A + 12A (-2766W @ 1p)
Jul 19 16:04:05 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:05 set charge current limit: 12A
Jul 19 16:04:05 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:05 charger enable
Jul 19 16:04:05 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:05 wake-up timer: start
Jul 19 16:04:39 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:04:39 ----
Jul 19 16:04:39 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:39 charge power: 4547W
Jul 19 16:04:39 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:39 charge currents: [9.95 10.2 0.21]A
Jul 19 16:04:39 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:04:39 battery 1 power: 2168W
Jul 19 16:04:39 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:04:39 grid power: 2402W
Jul 19 16:04:39 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:04:39 pv 1 power: 2749W
Jul 19 16:04:39 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:04:39 battery 1 soc: 99%
Jul 19 16:04:40 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:04:40 grid powers: [722 860 820]W
Jul 19 16:04:40 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:04:40 grid currents: [0.966 0.987 0.69]A
Jul 19 16:04:40 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:04:40 site power: 4570W
Jul 19 16:04:40 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:40 charge voltages: [234 236 236]V
Jul 19 16:04:40 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:40 charge total import: 6332.990kWh
Jul 19 16:04:40 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:40 charger status: C
Jul 19 16:04:40 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:40 charge start energy: 6332.990kWh
Jul 19 16:04:40 controller evcc[965246]: [lp-1  ] INFO 2025/07/19 16:04:40 start charging ->
Jul 19 16:04:40 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:40 wake-up timer: stop
Jul 19 16:04:40 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:40 pv charge current: min 6A > 0A (4570W @ 1p, battery: false)
Jul 19 16:04:40 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:04:40 set charge current limit: 6A
`
Above first start and stop (as seen in the tesla app

Now 'Charging equipment not ready in the Tesla app:

`Jul 19 16:05:39 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:05:39 ----
Jul 19 16:05:39 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:05:39 charge power: 0W
Jul 19 16:05:39 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:05:39 charge currents: [0 0 0]A
Jul 19 16:05:39 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:05:39 battery 1 power: -0W
Jul 19 16:05:39 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:05:39 grid power: -2747W
Jul 19 16:05:39 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:05:39 battery 1 soc: 99%
Jul 19 16:05:39 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:05:39 pv 1 power: 2926W
Jul 19 16:05:40 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:05:40 grid powers: [-970 -867 -910]W
Jul 19 16:05:40 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:05:40 grid currents: [-0.966 -0.987 -0.69]A
Jul 19 16:05:40 controller evcc[965246]: [site  ] DEBUG 2025/07/19 16:05:40 site power: -2747W
Jul 19 16:05:40 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:05:40 charge voltages: [234 238 235]V
Jul 19 16:05:40 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:05:40 charge total import: 6333.011kWh
Jul 19 16:05:40 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:05:40 charger status: B
Jul 19 16:05:40 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:05:40 charge final energy: 0.032kWh
Jul 19 16:05:40 controller evcc[965246]: [lp-1  ] INFO 2025/07/19 16:05:40 stop charging <-
[... cut ...]
Jul 19 16:09:40 controller evcc[965246]: [lp-1  ] DEBUG 2025/07/19 16:09:40 set charge current limit: 6A
`

And it stops charging. 

And so on...

Note that the charger is connected to the Smart Meter with its own connection. 
So there may be a difference between power asked and power given (not the situation at this moment as there is hardly any use of power inside the house) 

One other note. With the integrated SolarEdge setup, the battery provides power when there is 'ask' on the house grid.
So it will automatically start delivering the power needed.

mfuchs1984, 2025-07-19:
Is your charger configured to free charging or did you use a RFID tag to authorize? If not, reconfigure it or use RFID. 
When it is still not charging, create an ocpp trace log (and please format your logs, copy them into GitHub, select all and then press <> to do so.)

olger, 2025-07-19:
Hi @mfuchs1984 

It’s configured with a default RFID to start the charge. I’ll figure out how to get trace logs.

olger, 2025-07-19:
```
Jul 19 18:08:40 controller evcc[965738]: [site  ] DEBUG 2025/07/19 18:08:40 ----
Jul 19 18:08:40 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:08:40 charge power: 4W
Jul 19 18:08:40 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:08:40 charge currents: [0.01 0.06 0]A
Jul 19 18:08:40 controller evcc[965738]: [site  ] DEBUG 2025/07/19 18:08:40 grid power: -59W
Jul 19 18:08:40 controller evcc[965738]: [site  ] DEBUG 2025/07/19 18:08:40 pv 1 power: 579W
Jul 19 18:08:40 controller evcc[965738]: [site  ] DEBUG 2025/07/19 18:08:40 battery 1 power: -0W
Jul 19 18:08:40 controller evcc[965738]: [site  ] DEBUG 2025/07/19 18:08:40 battery 1 soc: 99%
Jul 19 18:08:40 controller evcc[965738]: [site  ] DEBUG 2025/07/19 18:08:40 grid powers: [-205 287 -140]W
Jul 19 18:08:40 controller evcc[965738]: [site  ] DEBUG 2025/07/19 18:08:40 grid currents: [-0.966 0.987 -0.69]A
Jul 19 18:08:40 controller evcc[965738]: [site  ] DEBUG 2025/07/19 18:08:40 site power: -59W
Jul 19 18:08:40 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:08:40 charge voltages: [234 236 233]V
Jul 19 18:08:40 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:08:40 charge total import: 6333.080kWh
Jul 19 18:08:40 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:08:40 charger status: B
Jul 19 18:08:40 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:08:40 charge start energy: 6333.080kWh
Jul 19 18:08:40 controller evcc[965738]: [lp-1  ] INFO 2025/07/19 18:08:40 car connected
Jul 19 18:08:40 controller evcc[965738]: [lp-1  ] INFO 2025/07/19 18:08:40 vehicle updated: unknown -> Tesseke de 3e
Jul 19 18:08:40 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:08:40 pv timer elapse
Jul 19 18:08:40 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:08:40 pv timer inactive
Jul 19 18:08:40 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:08:40 charge total import: 6333.080kWh
Jul 19 18:08:40 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:08:40 charger vehicle id: FFFFFFFF
Jul 19 18:08:41 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:08:41 vehicle status: B
Jul 19 18:08:41 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:08:41 vehicle soc: 77%
Jul 19 18:08:41 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:08:41 vehicle soc limit: 91%
Jul 19 18:08:41 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:08:41 vehicle range: 330km
Jul 19 18:08:41 controller evcc[965738]: [ocpp  ] TRACE 2025/07/19 18:08:41 send ENOGEN22SB-JB-0005433: [2,"3566342299","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
Jul 19 18:08:41 controller evcc[965738]: [ocpp  ] TRACE 2025/07/19 18:08:41 recv ENOGEN22SB-JB-0005433: [3,"3566342299",{"status":"Accepted"}]
Jul 19 18:08:41 controller evcc[965738]: [ocpp  ] TRACE 2025/07/19 18:08:41 recv ENOGEN22SB-JB-0005433: [2,"18241","MeterValues",{"connectorId":1,"transactionId":1752940878,"meterValue":[{"timestamp":"2025-07-19T16:08:39Z","sampledValue":[{"value":"0.01","context":"Trigger","format":"Raw","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"0.06","context":"Trigger","format":"Raw","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"0","context":"Trigger","format":"Raw","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"0","context":"Trigger","format":"Raw","measurand":"Current.Offered","unit":"A"},{"value":"6333080","context":"Trigger","format":"Raw","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"4","context":"Trigger","format":"Raw","measurand":"Power.Active.Import","unit":"W"},{"value":"2","context":"Trigger","format":"Raw","measurand":"Power.Active.Import","phase":"L1","unit":"W"},{"value":"2","context":"Trigger","format":"Raw","measurand":"Power.Active.Import","phase":"L2","unit":"W"},{"value":"0","context":"Trigger","format":"Raw","measurand":"Power.Active.Import","phase":"L3","unit":"W"},{"value":"234.38","context":"Trigger","format":"Raw","measurand":"Voltage","phase":"L1-N","unit":"V"},{"value":"235.99","context":"Trigger","format":"Raw","measurand":"Voltage","phase":"L2-N","unit":"V"},{"value":"232.98","context":"Trigger","format":"Raw","measurand":"Voltage","phase":"L3-N","unit":"V"}]}]}]
```

This is when I plug it in. 

```
Jul 19 18:12:05 controller evcc[965738]: [ocpp  ] TRACE 2025/07/19 18:12:05 send ENOGEN22SB-JB-0005433: [3,"18285",{}]
Jul 19 18:12:07 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:12:07 set charge mode: minpv
Jul 19 18:12:07 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:12:07 pv timer reset
Jul 19 18:12:07 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:12:07 pv timer inactive
Jul 19 18:12:07 controller evcc[965738]: [site  ] DEBUG 2025/07/19 18:12:07 ----
Jul 19 18:12:07 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:12:07 charge power: 4W
Jul 19 18:12:07 controller evcc[965738]: [lp-1  ] DEBUG 2025/07/19 18:12:07 charge currents: [0.01 0.06 0]A
Jul 19 18:12:07 controller evcc[965738]: [site  ] DEBUG 2025/07/19 18:12:07 battery 1 power: -0W
Jul 19 18:12:07 controller evcc[965738]: [site  ] DEBUG 2025/07/19 18:12:07 battery 1 soc: 99%
Jul 19 18:12:07 controller evcc[965738]: [site  ] DEBUG 2025/07/19 18:12:07 grid power: -152W
Jul 19 18:12:07 controller evcc[965738]: [site  ] DEBUG 2025/07/19 18:12:07 pv 1 power: 656W
Jul 19 18:12:07 controller evcc[965738]: [site  ] DEBUG 2025/07/19 18:12:07 grid powers: [-225 278 -158]W
[... cut ...] "Raw","measurand":"Voltage","phase":"L2-N","unit":"V"},{"value":"232.98","context":"Sample.Periodic","format":"Raw","measurand":"Voltage","phase":"L3-N","unit":"V"}]}]}]
Jul 19 18:12:24 controller evcc[965738]: [ocpp  ] TRACE 2025/07/19 18:12:24 send ENOGEN22SB-JB-0005433: [3,"18289",{}]

```

Till this point the Tesla App states a few things:
 - Chargint equipment not ready
 - Charging Error - No Power
 
Thereafter it keeps on logging meter requests. 

The UI states: Ready, Waiting for vehicle.

olger, 2025-07-19:
> Current.Offered should go to 6 Ampere after evcc sends the SetChargingProfile. But although the charger accepts it, it stays on 0. This is a failure of the charger to comply to ocpp smart charging.

Thanks for the clarification. I’ll see if I can check with eNovates