evcc-io/evcc #22465: Kalkulation von Ladekosten und Überschusssteuerung bei Einspeiselimit

ReinholdEckert, 2025-07-20:
Aktuell wird in evcc beim Überschussladen angenommen, dass der Strom ansonsten in das Netz eingespeist werden könnte - entsprechend wird auch die aktuelle Einspeisevergütung als Kosten für das Laden dargestellt.

Nun ist es bei vielen aber so, dass zB ein Einspeiselimit von 10kW besteht. Meine Wechselrichter müssen sich entsprechend runterregeln, wenn dieses Limit erreicht wird. Die Energie wird also verschenkt bzw. nicht genutzt.
Wenn ich mein Auto aber in diesem Zeitfenster lade, wo ich zB 18kW erzeugen kann, kann ich 8kW kostenlos nutzen, da ich sie ohnehin nicht einspeisen könnte. Die Kosten müssten also mit 0 berechnet werden, wenn evcc ein hinterlegtes Einspeiselimit erkennt.

Noch schöner wäre natürlich, wenn es einen Lademodus gäbe, der überhaupt nur in diesem Bereich regelt. Also wenn 10kW Einspeiselimit hinterlegt sind, dann könnte sich evcc über einen dynamischen Bereich der Stromstärke hintasten, ob es zB mit 6 oder 8 oder 10 Ampere so laden könnte, dass die Wechselrichter ausgelastet sind und trotzdem 10kW eingespeist werden.
Das wäre halt dann absolut kostenoptimiert. Aktuell mache ich diesen Schritt manuell.

Vielen Dank vorab!

andig, 2025-07-21:
> Ich könnte mir aber vorstellen, dass - sofern ein Einspeiselimit hinterlegt ist und im Lademodus irgendwie der "Über-Überschuss"-Modus gewählt werden kann - evcc einfach versucht die Ladeleistung auf den Wert hin zu optimieren, bei welchem man möglichst nahe an das Einspeiselimit kommt.

Das wäre so ähnlich wie der Battery Boost, nur eben nicht mit Abregelung auf 0W sondern auf 10kW.

andig, 2025-08-02:
@naltatis wenns hier kein Todo bei der Kostenrechnung gibt können wir zu machen?

@ReinholdEckert weitere Dynamisierung der Optimierung abhängig von https://github.com/evcc-io/evcc/pull/22495

closed 2025-08-02 by andig