evcc-io/evcc #22466: EVCC crashing since 0.204.5

juz22, 2025-07-20:
### Describe the bug

I'm running EVCC in Synology Container Manager for the past 7 months, and since 0.204.5 and 0.205.0 have encountered frequent crashes (between 1 and 3 times per 24 hours) that never happened previously.

### Steps to reproduce

1. I haven't been able to narrow down the crashes to any particular action. It appears to occur out of the blue and can happen whilst no car is connected or whilst the car is charging.

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

interval: 30s

log: debug

meters:
  - name: SungrowGrid
    type: template
    template: sungrow-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: 192.168.0.60
    port: 502
    timeout: 10s

  - name: SungrowBattery
    type: template
    template: sungrow-hybrid
    usage: battery
    modbus: tcpip
    id: 1
    host: 192.168.0.60
    port: 502
    timeout: 10s

  - name: SungrowSolar
    type: template
    template: sungrow-hybrid
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.0.60
    port: 502
    timeout: 10s

chargers:
  - name: froniuswattpilot
    type: template
    template: fronius-wattpilot
    host: 192.168.0.16
    password: ***

vehicles:
  - name: polestar2
    type: template
    template: polestar
    vin: ***
    title: Polestar 2
    user: ***
    password: ***
#    capacity: 78
    icon: car
    phases: 3
    cache: 15m
[... 238 lines cut ...]
2025/07/16 23:50:05	stdout	[site  ] DEBUG 2025/07/16 23:50:05 battery 1 soc: 82%
2025/07/16 23:50:05	stdout	[site  ] DEBUG 2025/07/16 23:50:05 grid currents: [-0.323 0.496 -0.146]A
2025/07/16 23:50:05	stdout	[site  ] DEBUG 2025/07/16 23:50:05 battery 1 power: 562W
2025/07/16 23:50:05	stdout	[site  ] DEBUG 2025/07/16 23:50:05 pv 1 power: 0W
2025/07/16 23:50:05	stdout	[site  ] DEBUG 2025/07/16 23:50:05 grid power: -0W
2025/07/16 23:50:05	stdout	[lp-1  ] DEBUG 2025/07/16 23:50:05 charge currents: [0 0 0]A
2025/07/16 23:50:05	stdout	[lp-1  ] DEBUG 2025/07/16 23:50:05 charge power: 0W
2025/07/16 23:50:05	stdout	[site  ] DEBUG 2025/07/16 23:50:05 ----
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.205.0

andig, 2025-07-20:
/cc @mabunixda it would still be nice if it didn't crash ;)

mabunixda, 2025-07-22:
Yeah ... i am not using this API in combination with evcc anymore - I switched to ocpp.
but I gonna review that because I am still using it for prometheus exporter job(s)

andig, 2025-07-22:
Much appreciated, thank you!

closed 2025-07-22 by andig

juz22, 2025-08-05:
Just wanted to say thank you, this appears to have fixed my crashing issue. I will figure out how to update to use OCPP at some stage but it's great that I can get continued functionality in the meantime.