evcc-io/evcc #22479: Missing template sma-sbs-modbus in docker environment

mkernforvia, 2025-07-21:
### Describe the bug

I am trying to run EVCC using docker-compose. I've pulled the latest `evcc/evcc:latest` image.

Upon running `docker compose up` I get template not found? Do I have to do anything else here?

Thanks

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:                                                                                                                                                                                                                                  
  - name: my_battery                                                                                                                                                                                                                       
    type: template                                                                                                                                                                                                                         
    template: sma-sbs-modbus                                                                                                                                                                                                               
    usage: battery                                                                                                                                                                                                                         
    # Modbus TCP                                                                                                                                                                                                                           
    modbus: tcpip                                                                                                                                                                                                                          
    id: 3                                                                                                                                                                                                                                  
    host: 192.168.178.44 # Hostname                                                                                                                                                                                                        
    port: 502 # Port                                                                                                                                                                                                                       
    capacity: 50 # Akkukapazität (kWh), optional                                                                                                                                                                                           
    maxchargepower: 4200 # Maximale Ladeleistung (W), Für erzwungenes Laden des Speichers. (optional)                                                                                                                                      
    watchdog: 60s # Watchdog, optional
```

### Log details

```text
evcc  | [main  ] INFO 2025/07/21 12:21:23 evcc 0.205.0
evcc  | [main  ] INFO 2025/07/21 12:21:23 using config file: /etc/evcc.yaml
evcc  | [db    ] INFO 2025/07/21 12:21:23 using sqlite database: /root/.evcc/evcc.db
evcc  | [main  ] INFO 2025/07/21 12:21:23 UI listening at :7070
evcc  | [main  ] FATAL 2025/07/21 12:21:23 charger [my_battery] cannot create charger 'my_battery': cannot create charger type 'template': template not found: sma-sbs-modbus
evcc  | [main  ] FATAL 2025/07/21 12:21:23 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.205.0

andig, 2025-07-21:
I can repro this. What's the last docker version that worked for you?

andig, 2025-07-21:
Update: a battery is a `meter`, not a `charger`!

closed 2025-07-21 by andig

joekerna, 2025-07-21:
I switched Github users.... 
I don't know. I haven't used it in over a year and wanted to try again.

joekerna, 2025-07-21:
> Update: a battery is a `meter`, not a `charger`!

of course you're right. Sorry about that.

Thanks a lot