evcc-io/evcc #22491: Kia Support in Australia

adam-80, 2025-07-22:
### Describe the bug

I have a Kia EV5 in Australia, it appears that the Kia Bluelink will not work. I am sponsoring the project so that I can get Wallbox and Kia integrations working, the wallbox is fine.

It appears the Bluelink integration hits the European servers only.  I am using the Kia API in Home Assistant successfully using the Kia Uvo code - https://github.com/Hyundai-Kia-Connect/kia_uvo

The main difference is the ability to add a country code to connect to the correct API. 

### Steps to reproduce

1. Create a connection using the Add Vehicle option
2. Select Kia Bluelink
3. Fill in the details as per Kia account
4. Get error "cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)"
...

### Configuration details

```yaml
Adding via GUI so no config to post.
```

### Log details

```text
[kia   ] TRACE 2025/07/22 17:29:32 POST https://eu-account.kia.com/auth/realms/eukiaidm/login-actions/authenticate?session_code=vEAjVFHT1iLtKz4_3olm1MzvAopEx4HZTd6HrctCGJs&execution=8eee4545-fa39-4c5c-8722-20e40c8b1710&client_id=572e0304-5f8d-4b4c-9dd5-41aa84eed160&tab_id=iwqjoAtR9tU
[kia   ] TRACE 2025/07/22 17:29:33 credentialId=&password=***&rememberMe=on&username=***
--
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN"  "http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd">
    <html xmlns="http://www.w3.org/1999/xhtml" class="login-pf"
                    lang="en"
    >

    <head>
        <meta charset="utf-8">
        <meta http-equiv="Content-Type" content="text/html; charset=UTF-8" />
        <meta name="robots" content="noindex, nofollow">
        <meta http-equiv="Expires" content="Mon, 06 Jan 1990 00:00:01 GMT">
        <meta http-equiv="Expires" content="-1">
        <meta http-equiv="Pragma" content="no-cache">
        <meta http-equiv="Cache-Control" content="no-cache">

                <meta name="viewport" content="width=device-width,initial-scale=1"/>
        <title>Log in to Kia Account</title>
        <link rel="icon" href="/auth/resources/19ggl/login/kia-account-2022/img/favicon.ico?v=2" />

            <!-- OneTrust Cookies Consent Notice start for eu-account.kia.com -->
            <script type="text/javascript" src="https://cdn.cookielaw.org/consent/dea61314-e1fe-4d63-8605-6b28e56f24ea/OtAutoBlock.js?v=1" ></script>
            <script src="https://cdn.cookielaw.org/scripttemplates/otSDKStub.js?v=1" data-document-language="true" type="text/javascript" charset="UTF-8" data-domain-script="dea61314-e1fe-4d63-8605-6b28e56f24ea" ></script>
            <script type="text/javascript">function OptanonWrapper() { }</script>
            <!-- OneTrust Cookies Consent Notice end for eu-account.kia.com -->

                <link href="/auth/resources/19ggl/login/kia-account-2022/node_modules/bootstrap/dist/css/bootstrap.css?v=10" rel="stylesheet" />
                <link href="/auth/resources/19ggl/login/kia-account-2022/css/kia-login-2022.css?v=10" rel="stylesheet" />

                <script src="/auth/resources/19ggl/login/kia-account-2022/node_modules/jquery/dist/jquery.min.js" type="text/javascript"></script>
                <script src="/auth/resources/19ggl/login/kia-account-2022/node_modules/bootstrap/dist/js/bootstrap.min.js" type="text/javascript"></script>
                <script src="/auth/resources/19ggl/login/kia-account-2022/js/utils.js?v=3" type="text/javascript"></script>
                <script src="/auth/resources/19ggl/login/kia-account-2022/js/css-vars-ponyfill.js" type="text/javascript"></script>
        <script type="text/javascript">
            cssVars({});

            const bgImgPath = [];

            bgImgPath[0] = [
                "/auth/resources/19ggl/login/kia-account-2022/img/kia-600x400-1.jpg?v=3"
                , "/auth/resources/19ggl/login/kia-account-2022/img/kia-600x400-2.jpg?v=3"
                , "/auth/resources/19ggl/login/kia-account-2022/img/kia-600x400-3.jpg?v=3"
                , "/auth/resources/19ggl/login/kia-account-2022/img/kia-600x400-4.jpg?v=3"
                , "/auth/resources/19ggl/login/kia-account-2022/img/kia-600x400-5.jpg?v=3"
[... 97 lines cut ...]
            }

            function startShowBGImgs() {
                const bgImgList = [];

                bgImgList[0] = {idx:1, interval:3000, idPrefix:"#BG_Img1"};
                bgImgList[1] = {idx:1, interval:5000, idPrefix:"#BG_Img2"};
                bgImgList[2] = {idx:1, interval:7000, idPrefix:"#BG_Img3"};
[kia   ] TRACE 2025/07/22 17:29:33 POST https://prd.eu-ccapi.kia.com:8080/api/v1/user/signin
[kia   ] TRACE 2025/07/22 17:29:34 {"email":"***","password":"QY5^SN!T05SZ\u0026bY$!4Io"}
--
{"errId":"15858854-b8c7-4996-8667-10249199e82c","errCode":"4003","errMsg":"Invalid values"}
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-07-22:
Could you kindly check which domains are being used for login and actual api? I couldn't quite figure out from looking at the code.

adam-80, 2025-07-22:
> Could you kindly check which domains are being used for login and actual api? I couldn't quite figure out from looking at the code.

Yep, it's actually at the bottom of the logs I posted. It's hitting Europe by the looks of it: https://prd.eu-ccapi.kia.com:8080/api/v1/user/signin

When setting up the Kia Uvo integration in Home Assistant you are asked to select a country, which routes you to the correct API endpoint.

adam-80, 2025-07-22:
For reference, this is the python package that was written to simplify dealing with multiple regions and is based on the bluelinky package: https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api 

I am not a coder, but looking at the Bluelink code in evcc it seems like both of these packages are referenced.

andig, 2025-07-23:
I meant: what are the correct urls to hit?

andig, 2025-07-23:
Added a fix, but smdb. needs to confirm this is working.

andig, 2025-07-25:
Anyone with a Kia/Hyunday, preferably from non-EU.

adam-80, 2025-07-28:
Hi Again, just wondering if there's a way I can help to test the new code?

bsmout, 2025-07-30:
KIA EV6 no longer connecting here as well:
cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: unexpected status: 401 (Unauthorized): Require authentication (4010)

RustyDust, 2025-08-14:
> Hi Again, just wondering if there's a way I can help to test the new code?

If you're still interested, mind testing this -> https://github.com/RustyDust/evcc/tree/kia_hyundai_oauth2 ?
I just added code, that _might_ support Kia/Hyundai logins in AU/NZ but for me being in Europe I obviously lack the necessary credentials for testing.
In case you need a binary for testing let me know where you intend to run it and I'll see what I can do.