evcc-io/evcc #22494: Template für openWB EVU Kit lovato dme d330 meter

rehajel, 2025-07-23:
Hi,

ich habe ein Lovato dme d330 Meter mit Elfin-EE11 modbus converter im Einsatz.
Die Meter/Converter Combo wurde mal bei openWB als "EVU Kit" verkauft und ist so auch dort vorkonfiguriert.

Ich habe die Konfiguration via modbus plugin bei mir eingebunden und bekomme korrekte Daten. Mit Hilfe von AI habe ich auch ein Template erstellt, dieses aber nicht getestet.

Snippet aus der evvc.yaml mit meiner Meter Konfiguration:

```
meters:
  - name: grid
    type: custom
    power:
      source: calc
      add:
        - source: modbus
          uri: 192.168.0.46:8899
          id: 2
          timeout: 4s
          scale: 0.01
          register:
            address: 19
            type: input
            decode: int32
        - source: modbus
          uri: 192.168.0.46:8899
          id: 2
          timeout: 4s
          scale: 0.01
          register:
            address: 21
            type: input
            decode: int32
        - source: modbus
          uri: 192.168.0.46:8899
          id: 2
          timeout: 4s
          scale: 0.01
          register:
            address: 23
            type: input
            decode: int32
    powers:
      - source: modbus
        uri: 192.168.0.46:8899
        id: 2
        timeout: 4s
        scale: 0.01
        register:
          address: 19
          type: input
          decode: int32
      - source: modbus
        uri: 192.168.0.46:8899
        id: 2
        timeout: 4s
        scale: 0.01
        register:
          address: 21
          type: input
          decode: int32
      - source: modbus
        uri: 192.168.0.46:8899
        id: 2
        timeout: 4s
        scale: 0.01
        register:
          address: 23
          type: input
          decode: int32
    voltages:
      - source: modbus
        uri: 192.168.0.46:8899
        id: 2
        timeout: 4s
        scale: 0.01
        register:
          address: 1
          type: input
          decode: int32
      - source: modbus
        uri: 192.168.0.46:8899
        id: 2
        timeout: 4s
        scale: 0.01
        register:
          address: 3
          type: input
          decode: int32
      - source: modbus
        uri: 192.168.0.46:8899
        id: 2
        timeout: 4s
        scale: 0.01
        register:
          address: 5
          type: input
          decode: int32
    currents:
      - source: modbus
        uri: 192.168.0.46:8899
        id: 2
        timeout: 4s
        scale: 0.0001
        register:
          address: 7
          type: input
          decode: int32
      - source: modbus
        uri: 192.168.0.46:8899
        id: 2
        timeout: 4s
        scale: 0.0001
        register:
          address: 9
          type: input
          decode: int32
      - source: modbus
        uri: 192.168.0.46:8899
        id: 2
        timeout: 4s
        scale: 0.0001
        register:
          address: 11
          type: input
          decode: int32

``` 

Mögliches Template File:

```
template: lovato-dme-d330
products:
  - brand: Lovato
    description: DME D330 3-phase energy meter
    
params:
  - name: host
    default: 192.168.193.15
  - name: port
    default: 8899
  - name: id
    default: 2

render: |
  type: custom
  power:
    source: calc
    add:
      - source: modbus
        uri: {{ .host }}:{{ .port }}
        id: {{ .id }}
        scale: 0.01
        register:
          address: 19
          type: input
          decode: int32
      - source: modbus
        uri: {{ .host }}:{{ .port }}
        id: {{ .id }}
        scale: 0.01
        register:
          address: 21
          type: input
          decode: int32
      - source: modbus
        uri: {{ .host }}:{{ .port }}
        id: {{ .id }}
        scale: 0.01
        register:
          address: 23
          type: input
          decode: int32
  powers:
    - source: modbus
      uri: {{ .host }}:{{ .port }}
      id: {{ .id }}
      scale: 0.01
      register:
        address: 19
        type: input
        decode: int32
    - source: modbus
      uri: {{ .host }}:{{ .port }}
      id: {{ .id }}
      scale: 0.01
      register:
        address: 21
        type: input
        decode: int32
    - source: modbus
      uri: {{ .host }}:{{ .port }}
      id: {{ .id }}
      scale: 0.01
      register:
        address: 23
        type: input
        decode: int32
  voltages:
    - source: modbus
      uri: {{ .host }}:{{ .port }}
      id: {{ .id }}
      scale: 0.01
      register:
        address: 1
        type: input
        decode: int32
    - source: modbus
      uri: {{ .host }}:{{ .port }}
      id: {{ .id }}
      scale: 0.01
      register:
        address: 3
        type: input
        decode: int32
    - source: modbus
      uri: {{ .host }}:{{ .port }}
      id: {{ .id }}
      scale: 0.01
      register:
        address: 5
        type: input
        decode: int32
  currents:
    - source: modbus
      uri: {{ .host }}:{{ .port }}
      id: {{ .id }}
      scale: 0.0001
      register:
        address: 7
        type: input
        decode: int32
    - source: modbus
      uri: {{ .host }}:{{ .port }}
      id: {{ .id }}
      scale: 0.0001
      register:
        address: 9
        type: input
        decode: int32
    - source: modbus
      uri: {{ .host }}:{{ .port }}
      id: {{ .id }}
      scale: 0.0001
      register:
        address: 11
        type: input
        decode: int32

``` 

Meine Frage wäre - kann/soll ich für dieses Template einen PR erstellen, oder ist das nicht erwünscht, weil es AI generiert ist und/oder das Kit von einem anderen Produkt ist?

Technisch noch etwas Erklärung:

Der Elfin-EE11 Converter mit openWB Konfiguration ist standardmäßig unter der IP 192.168.193.15 auf port 8899 erreichbar (dieses netz ist standardmäßig mit secondary ip auf der openwb konfiguriert).
StandardUser / PW sind im Netz findbar - damit lässt sich dann auch IP/Port ändern.
Die Modbus ID für das lovato dme d330 Meter ist wohl 2.

Noch eine (unverschämte?) Frage zum Schluss: Wäre das genug contribution für einen developer token? ;)

premultiply, 2025-07-23:
Für die integrierten Modbus-Zähler verwenden wir üblicherweise als Grundlage [mbmd](https://github.com/volkszaehler/mbmd).
Falls der Zähler da noch nicht drin ist dann gerne dort einen entsprechenden PR erstellen.

Die externe Konfiguration ist nicht so wichtig und kann ggf. dann später in einem Template oder dessen Beschreibung vermittelt werden. Das muss man dann sehen was wie wem konkret weiterhilft oder eher ein Sonderfall ist.

andig, 2025-07-23:
Added a fix, but smdb. needs to confirm this is working.

andig, 2025-07-25:
> bin aber bisher aber gescheitert.

Das ist arg unspezifisch. 

> nur nicht bevorzugt

Genau

In jedem Fall muss dine Lösung getestet werden. Ich mach mal zu bis es dazu Feedback gibt.

closed 2025-07-25 by andig