evcc-io/evcc #22521: Amber electric 5 minutes billing change

Audiobuzz, 2025-07-25:
**Is your feature request related to a problem? Please describe.**

Amber electric recently changed from 30 minutes billing to 5 minutes billing. From what I can tell evcc is still using the 30 minute predicted price rather than the live 5 minutes actual price. 

**Describe the solution you'd like**

Use the 5 minute actual price for import and export pricing so that charging can be more responsive and accurate to live price changes

andig, 2025-07-26:
This would mean using the forecast value for t>0 but always re-requesting the instand value for the current rate. I'm not sure if that can be done/is helpful. /cc @ross-w

ross-w, 2025-07-26:
Yep that’s correct we are still using the 30 minute values. But as I understand it evcc still uses hourly planning? So I’m not sure how helpful switching to the 5 minute values would be

ross-w, 2025-07-26:
Good point. Ok, easily done I think.

andig, 2025-07-26:
I'm not familiar with the code, but typicially you would cache the forecast for ~1h. If the _current_ price changes against the forecast that would not be picked up?