evcc-io/evcc #22544: Incorrect kWh recorded for Keba P40 wallbox with firmware 1.2.2 and evcc 0.205.0

MFs-1, 2025-07-27:
### Describe the bug

I recently updated evcc to 0.205.0 and Keba released a new Firmware 1.2.2 for the P40 wallbox. Since both updates the recorded kWh for a charging session deviates in evcc by factor 10 from the real value. Example for my last (only) charging session: evcc recorded 118.2 kWh, the wallbox web interface in the emobility portal reported 11.9 kWh. Battery size of my car is ~60 kWh, I charged ~18%. I did not do a charging where only one of the two was updated.
Previous charges with old evcc and Keba firmware were consistent.

### Steps to reproduce

1. charge a car with P40 firmware 1.2.2 and evcc 0.205.0
2. compare the kWh values in evcc and the P40 web interface (emobility portal)

### Configuration details

```yaml
[main  ] INFO 2025/07/27 06:42:11 using config file: C:\evcc\evcc.yaml
[db    ] INFO 2025/07/27 06:42:11 using sqlite database: C:\Users\www\.evcc\evcc.db
[lp-1  ] DEBUG 2025/07/27 06:42:13 charge total import: 422.754kWh

Konfiguration (C:\evcc\evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- name: grid1
  type: template
  template: senec-home 
  usage: grid  
  host: 192.168.178.31  
  schema: https  
- name: pv2
  type: template
  template: senec-home 
  usage: pv  
  host: 192.168.178.31  
  schema: https  
- name: battery3
  type: template
  template: senec-home 
  usage: battery  
  host: 192.168.178.31  
  schema: https  
- name: grid2
  type: template
  template: keba-kecontact
  id: 1
  host: 192.168.178.68
  port: 502
  usage: grid
  modbus: tcpip

chargers:
- type: template
  template: keba-modbus 
  id: 255  
  host: 192.168.178.71  
  port: 502  
  modbus: tcpip  
  name: wallbox5
  welcomecharge: # Laden bei Verbindung, Wallbox gibt kurzzeitige Ladefreigabe bei Fahrzeugverbindung. Das ermöglicht es Fahrzeugen, die eine Stromversorgung beim Anschließen benötigen, einen Fehlerzustand zu vermeiden. (optional)
    
vehicles:
- type: template
  template: cupra 
  icon: car                           
  user: *****
  password: *****
  title: Cupra Born  
  capacity: 59  
  phases: 3  
  cache: 15m  
  mode: minpv  
  welcomecharge: false           
  name: ev4

loadpoints:
- title: Keba P40
  charger: wallbox5
  mode: off

site:
  title: Mein Zuhause
  residualpower: 100
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

Version: `0.205.0`
```

### Log details

```text
Sorry, not available. I detected the issue after charging was finished.
```

### What type of operating system or environment does evcc run on?

Windows

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.205.0

TobiasHuber1980, 2025-07-27:
https://github.com/evcc-io/evcc/pull/20907

andig, 2025-07-27:
Looks like this was fixed ages ago. Will need trace log to analyse.

/cc @breiti

closed 2025-07-27 by andig

breiti, 2025-07-29:
After updating to 1.2.2 I can also see this behavior. Something seems to be
wrong with the fix above. It will take some time until I can look into it,
so I'd be happy if somebody else could have a look.

MFs-1, 2025-07-29:
Just finished a new (quite short) loading session. This time evcc and my Wallbox reported the same values. I created a log this time, maybe it helps. I have also the logs at TRACE level, but at least there is the VIN in it. I don't know which other data it contains that I don't want to see in the public. I can extract the details by timestamp if requested.

[log.log](https://github.com/user-attachments/files/21488289/log.log)