evcc-io/evcc #22549: Volvo authentication is expiring every 7 days

yareblo, 2025-07-27:
### Describe the bug

Hi, I am using evcc 0.205.0 with the new Volvo connector. Normally it is working very good. Unfortunately, every ~7 days the connection is lost. It can be restarted manually very quickly by:

1) restart the evcc-service
2) scan the log and call http://openhabian:7070/oauth/login?id=SuperSecret123
3) approve everything until you get to the error page
4) remove the "s" from the callback (httpS://openhabian...)
5) restart the evcc-service

This works quite reliable. However, it is a bit annoying that evcc does not respect the charging limit and (maybe) charging plans when the connection is lost. It charges the vehicle to 100%. 

Is there a way to avoid the expitation? Is the "s" in httpS the problem and how can it be fixed? Or, as a workaround, get the expiration date somewhere and refresh the token manually before it expires?

### Steps to reproduce

1. Use the Volvo connector
2. Wait 7 days
3.
...

### Configuration details

```yaml
[main  ] INFO 2025/07/27 10:15:29 using config file: /etc/evcc.yaml
[db    ] INFO 2025/07/27 10:15:29 using sqlite database: /var/lib/evcc/evcc.db
[oauth-generic] DEBUG 2025/07/27 10:15:29 loading token for Blubber from database
[oauth2redirect] INFO 2025/07/27 10:15:29 registering oauth provider at /oauth/login?id=Blubber 
[lp-1  ] WARN 2025/07/27 10:15:29 poll mode '{always 6m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] DEBUG 2025/07/27 10:15:30 charge total import: 674.259kWh

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: info

# unique installation id
plant: *****

interval: 30s # control cycle interval

meters:
- type: template
  template: rct-power
  usage: grid
  host: 192.168.178.6
  name: grid1
- type: template
  template: rct-power
  usage: pv
  host: 192.168.178.6
  name: pv2
- type: template
  template: rct-power
  usage: battery
  host: 192.168.178.6
  name: battery3

chargers:
- type: template
  template: homematic
  host: 192.168.178.4
  port: 2010
  device: '*****'
  user: *****
  password: *****
  standbypower: 15
  name: wallbox

loadpoints:
- title: Garage
  charger: wallbox
  mode: pv
  vehicle: db:1
  soc:
    poll:
      mode: always
      interval: 6m
  enable:
    threshold: 0
    delay: 10m
  disable:
    threshold: 0
    delay: 15m

site:
  title: My home
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.***** # EUR/kWh
  feedin:
    type: fixed
    price: 0.08 # EUR/kWh
  co2:
    type: template
    template: grünstromindex # Beispiel GrünstromIndex (nur Deutschland)
    zip: *****
    token: *****
  solar:
    - type: template
      template: forecast-solar
      lat: *****
      lon: *****
      dec: ***** # 0 = horizontal, 90 = vertikal
      kwp: ***** 
      az: - ***** # -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden
    - type: template
      template: forecast-solar
      lat: *****
      lon: *****
      dec: ***** # 0 = horizontal, 90 = vertikal
      kwp: ***** 
      az: ***** # -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden

vehicles:
  - name: guest_car
    title: Gast-Auto
    type: template
    template: offline
    capacity: *****  # in kWh

Version: `0.205.0`
```

### Log details

```text
Maybe this is the part:

Jul 26 16:42:38 openhabian evcc[3929]: [oauth-generic] DEBUG 2025/07/26 16:42:38 refreshing token for ****
Jul 26 16:42:39 openhabian evcc[3929]: [lp-1  ] WARN 2025/07/26 16:42:39 vehicle soc: Get "https://api.volvocars.com/energy/v2/vehicles/****/state": oauth2: "invalid_grant" "unknown, invalid, or expired re>
Jul 26 16:42:39 openhabian evcc[3929]: [lp-1  ] DEBUG 2025/07/26 16:42:39 soc estimated: 85.00% (vehicle: 85.00%)
Jul 26 16:42:39 openhabian evcc[3929]: [lp-1  ] DEBUG 2025/07/26 16:42:39 vehicle soc: 85%
Jul 26 16:42:39 openhabian evcc[3929]: [lp-1  ] ERROR 2025/07/26 16:42:39 vehicle soc limit: Get "https://api.volvocars.com/energy/v2/vehicles/****/state": oauth2: "invalid_grant" "unknown, invalid, or exp>
Jul 26 16:42:39 openhabian evcc[3929]: [lp-1  ] ERROR 2025/07/26 16:42:39 vehicle range: Get "https://api.volvocars.com/energy/v2/vehicles/****/state": oauth2: "invalid_grant" "unknown, invalid, or expired>
Jul 26 16:42:39 openhabian evcc[3929]: [lp-1  ] DEBUG 2025/07/26 16:42:39 limitSoc reached: 85.0% > 85%
J
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.205.0

andig, 2025-07-27:
> Is there a way to avoid the expitation? 

As Volvo... We canm only make it simpler with https://github.com/evcc-io/evcc/pull/21266.

closed 2025-07-27 by andig

MartijnvanGeffen13, 2025-08-07:
As the developer of Volvo4EVCC i am also running into this same issue , 

And this is against the documentation of Volvo , as they claim with refresh key rotation the tokens should be valid longer than 7 days. 

I am expirimenting with cold tokens to not keep the token hot to see if that brings a differance and we are in conversation with Volvo as this seems a bug / not correctly implemented from their end when they switched to the V2 and shortend the lifespan of the active token. 

But they are on holiday for atleast another month so we have to wait till volvo somes back.

andig, 2025-10-30:
> I tried going through the code - but couldn't find where it does the refresh handling of the refresh token that quickly - but I would recommended trying to refresh the token every 3 days - that way we are in the refresh time limit (hopefully) and it should keep running

Added in https://github.com/evcc-io/evcc/pull/24857