evcc-io/evcc #22563: Mennekes Smart Premium 22 (ACU V4) 2. Ladepunkt lässt sich nicht hinzufügen,

MasterOfPuppets111, 2025-07-27:
### Describe the bug

Hallo zusammen,

Ich nutze seit längerem eine Mennekes Smart Premium 22 Ladesäule mit 2x 22kW Typ 2 Ladepunkten.
Seit Mitte letzter Woche versuche ich EVCC zum laufen zu bringen mit dem Testtoken um erstmal zu schauen ob es das alles so bei mir funktioniert.

Wenn ich einen Ladepunkt hinzufüge funktioniert das prima, sobald ich aber den zweiten Ladepunkt hinzufügen möchte geht er immer in einen Timeout und will nach 15min EVCC neu starten, Log Level habe ich auf Trace gesetzt um zu schauen was passiert, das komische ist, es kommen Daten von Connector ID 2 an, trotz vorherigem Timeout.
Dieser Fehler:
```
[main ] FATAL 2025/07/27 17:31:04 charger [wallbox4] cannot create charger 'wallbox4': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[main ] FATAL 2025/07/27 17:31:04 will attempt restart in: 15m0s
```

Eventuell weiß ja jemand was der Fehler ist. anbei meine Config und mein Log.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
levels:
  cache: error

# unique installation id
plant: xxx

interval: 30s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.1.132  
  port: 502  
  meterid: 1  
  name: grid1
- type: template
  template: victron-energy 
  usage: pv  
  host: 192.168.1.132  
  port: 502  
  meterid: 1  
  name: pv2
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.1.132  
  port: 502  
  meterid: 1  
  name: battery3

[... 118 lines cut ...]
[ocpp ] TRACE 2025/07/27 17:30:53 send Mennekes: [2,"503565216","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":2}]
[ocpp ] TRACE 2025/07/27 17:30:54 recv Mennekes: [3, "503565216", {"status": "Accepted"}]
[ocpp ] TRACE 2025/07/27 17:30:55 send Mennekes: [2,"3041953947","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":2}]
[ocpp ] TRACE 2025/07/27 17:30:56 recv Mennekes: [3, "3041953947", {"status": "Accepted"}]
[ocpp ] TRACE 2025/07/27 17:30:56 recv Mennekes: [2, "bb7d94ae-6afe-11f0-8ae2-000a14851572", "MeterValues", {"meterValue": [{"timestamp": "2025-07-27T15:31:13.000Z", "sampledValue": [{"measurand": "Energy.Active.Import.Register", "unit": "Wh", "value": "5429689"}]}], "connectorId": 2}]
[ocpp ] TRACE 2025/07/27 17:30:56 send Mennekes: [3,"bb7d94ae-6afe-11f0-8ae2-000a14851572",{}]
[main ] FATAL 2025/07/27 17:31:04 charger [wallbox4] cannot create charger 'wallbox4': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[main ] FATAL 2025/07/27 17:31:04 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.205.0

VolkerK62, 2025-07-27:
Deine Charger heißen beide "wallbox4"!!??

MasterOfPuppets111, 2025-07-27:
Korrekt, einer von tausenden versuchen, das Ding ans laufen zu bekommen, habe jetzt den charger von connector 2 auf wallbox5 geändert, so wie anfangs auch nicht geklappt hat.... immernoch timeout 😞

MasterOfPuppets111, 2025-07-27:
Merkwürdig, jetzt geht es mit dieser Config... aber erst nachdem ich die ACU der Ladesäule mehrmals neu gestartet habe.
Doofe Frage, kann es an einem instabilen Wlan liegen? die Ladesäule war bisher "Dumm" da rum gestanden, ein Netwerkkabel habe ich auf die schnelle nicht dort hin legen können, deshalb habe ich einen Wlan Repeater als client konfiguriert um eine Netzwerkverbindung herzustellen, können davon die Timeout Probleme kommen, obwohl sich die Ladesäule immer super über das Webinterface erreichen ließ?

[image]

Config:
```
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
levels:
  cache: error

# unique installation id
plant: xxxx

interval: 30s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.1.132  
  port: 502  
  meterid: 1  
  name: grid1
- type: template
  template: victron-energy 
  usage: pv  
  host: 192.168.1.132  
  port: 502  
  meterid: 1  
  name: pv2
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.1.132  
  port: 502  
  meterid: 1  
  name: battery3

chargers:
- type: template
  template: ocpp-mennekes-acu
  stationid: Mennekes
  connector:  2 
  remotestart: false  
  connecttimeout: 5m  
  meterinterval: 10s  
  name: wallbox5
- type: template
  template: ocpp-mennekes-acu
  stationid: Mennekes
  connector: 1
  remotestart: false  
  connecttimeout: 5m  
  meterinterval: 10s  
  name: wallbox4

loadpoints:
[... 1424 lines cut ...]
[pv2   ] TRACE 2025/07/27 19:45:10 modbus: send 01 27 00 00 00 06 64 04 03 2a 00 01
[pv2   ] TRACE 2025/07/27 19:45:10 modbus: recv 01 27 00 00 00 05 64 04 02 00 00
[pv2   ] TRACE 2025/07/27 19:45:10 modbus: send 01 28 00 00 00 06 64 04 03 2b 00 01
[pv2   ] TRACE 2025/07/27 19:45:10 modbus: recv 01 28 00 00 00 05 64 04 02 00 48
[grid1 ] TRACE 2025/07/27 19:45:10 modbus: send 01 29 00 00 00 06 01 04 0a 3d 00 01
[grid1 ] TRACE 2025/07/27 19:45:10 modbus: recv 01 29 00 00 00 05 01 04 02 00 0d
[site  ] DEBUG 2025/07/27 19:45:10 grid currents: [1.2 1.7 1.3]A
[pv2   ] TRACE 2025/07/27 19:45:10 modbus: send 01 2a 00 00 00 06 64 04 03 2c 00 01
[pv2   ] TRACE 2025/07/27 19:45:10 modbus: recv 01 2a 00 00 00 05 64 04 02 00 50
[pv2   ] TRACE 2025/07/27 19:45:10 modbus: send 01 2b 00 00 00 06 64 04 03 2d 00 01
[pv2   ] TRACE 2025/07/27 19:45:10 modbus: recv 01 2b 00 00 00 05 64 04 02 00 2e
[pv2   ] TRACE 2025/07/27 19:45:10 modbus: send 01 2c 00 00 00 06 64 04 03 52 00 01
[pv2   ] TRACE 2025/07/27 19:45:10 modbus: recv 01 2c 00 00 00 05 64 04 02 00 00
[site  ] DEBUG 2025/07/27 19:45:10 pv 1 power: 198W
[site  ] DEBUG 2025/07/27 19:45:10 site power: 330W
[lp-2  ] DEBUG 2025/07/27 19:45:10 charger status: A
[ocpp  ] TRACE 2025/07/27 19:45:10 send Mennekes: [2,"2831904394","GetCompositeSchedule",{"connectorId":1,"duration":60}]
[ocpp  ] TRACE 2025/07/27 19:45:10 recv Mennekes: [4, "2831904394", "NotImplemented", "Action not known.", {}]
[ocpp  ] TRACE 2025/07/27 19:45:11 send Mennekes: [2,"2945620592","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/07/27 19:45:11 recv Mennekes: [3, "2945620592", {"status": "Accepted"}]
[ocpp  ] TRACE 2025/07/27 19:45:13 send Mennekes: [2,"196766584","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/07/27 19:45:13 recv Mennekes: [3, "196766584", {"status": "Accepted"}]
[httpd ] TRACE 2025/07/27 19:45:14 GET /api/system/log

```

closed 2025-07-27 by premultiply