evcc-io/evcc #22565: Kermi x-center 40 Unterstützung

Offroadbiker, 2025-07-27:
Hallo zusammen,

ich betreibe eine Wärmepumpe von Kermi mit der Steuerung x-center x40 Regelung (nicht "pro"). Die Steuerung hat einen PV-Modus, mit dem die Wärmepumpe den überschüssigen PV-Strom zum überheizen des Trinkwassers bzw. des Heizungswassers verwenden kann. 
Der Überschuss kann per Modbus an die Wärmepumpe übergeben werden. Hierzu ist ein Modbus TCP-->RTU Converter notwendig. 
Ich habe einen entsprechenden Converter im Einsatz und kann über eine Automation in Home Assistant den Überschuss an die Wärmepumpe senden. Die Wärmepumpe entscheidet dann je nach höhe des Überschusses, ob das Trinkwasser erwärmt wird oder ob das Heizwasser erhöht wird. Schöner wäre es, wenn der Überschuss über EVCC übermittelt werden kann, um so eine Priorisierung mit den Ladepunkten zu erreichen. 

Kann diese Wärmepumpenansteuerung in EVCC ergänzt werden? Also schlussendlich "nur" die Übermittlung des aktuellen PV-Überschusses und ggfs. das Anzeigen der aktuellen Heizungstemperaturen. Die Anbindung für Kermi pro Wärmepumpen ist ja bereits möglich, funktioniert aber nicht mit den älteren x-center x40 Regelungen. Die Modbusadressen kann ich gerne zu Verfügung stellen. 
Vielen Dank.

Viele Grüße
Max

andig, 2025-07-27:
Das kannst Du selbst machen, siehe Template für die Lambda.

Offroadbiker, 2025-07-27:
Danke für die schnelle Rückmeldung. Das bedeutet ich würde mit nachfolgendem Template den Überschuss per Modbus übermitteln?

chargers:
  - name: my_charger
    type: template
    template: lambda-zewotherm
    host: 192.0.2.2 # IP-Adresse oder Hostname
    tempsource: # Temperaturquelle, [warmwater_top, warmwater_bottom, buffer_top, buffer_bottom] (optional)
    excess: plus # E-Überschuss ("plus" oder "minus"), [plus, minus] (optional)
    watchdog: 60s # Watchdog, optional

An welcher Stelle wird hier das Register eingetragen?

andig, 2025-07-28:
Siehe https://github.com/evcc-io/evcc/blob/2d2bc58e6e0128d76abf4439245c89fda4515a5e/templates/definition/charger/lambda-zewotherm.yaml

Offroadbiker, 2025-07-31:
Hi, vielen Dank für den Link. Ich werde mir diesen mal anschauen ob ich damit weiter komme. Ich habe den Waveshare RS485 to RJ45 Ethernet Converter Module im Einsatz. Dieser hat einen LAN-Anschluss. 
https://www.amazon.de/dp/B0BGHVRMPJ/ref=pe_27091401_487027711_TE_SCE_dp_i1?th=1

VG

koch86, 2025-09-07:
Hallo @Offroadbiker,

ich habe auch eine Kermi WP mit x40 Steuerung und arbeite an der Integrations ins SmartHome.
Bisher mache ich das über HTTP und HomeAssistant, aber würde lieber Modbus und evcc verwenden.

- Wie hast du modbus an der x40 verkabelt? Am Interfacemodul (wo genau ansgeschlossen?) Oder direkt an der CAREL Steuerung? Die Reihenklemmen führen bei mir nämlich kein Modbus "raus". 
- Hast du technische Dokumentation für die x40? Sowohl Register als auch elektrisch?

Danke!

koch86, 2025-10-29:
Hallo @lfleck,
ich nutze den gleichen Host/Port wie die x40 Weboberfläche und App - also Port 80 / 443 und die IP der Steuerung.
Wenn man sich in den Entwicklertools (Firefox -> Netzwerkanalyse) anschaut, welche Anfragen die Web-App an die Steuerung schickt, dann kann man reverse-engineeren über welche Endpunkte man die Statuswerte abfragen kann.
Die IDs der Statuswerte kann man auch darüber ermitteln (zugegeben, erstmal viel manuelle Arbeit - aber in meinem Fall habe ich die 15-20 relevanten IDs einfach so ermitteln können). Die IDs bleiben im übrigen stabil.

Konkret sieht das bei mir so aus:

Öffnet man eine Status-Seite, dann sieht man wiederholende Anfragen an `https://**device-ip**/api/Datapoint/ReadValues/**device-id**` zu den einzelnen Werten:

[image]

Anhand der Datenpunkt-IDs habe ich mir entsprechende Anfragen gebastelt, die alle relevanten Daten abfragen und per MQTT für Home Assistant bereitstellen.

Kann man natürlich alles noch viel weiter perfektionieren, aber bisher hat mir dazu einfach die Zeit gefehlt.

Grüße!

rilezi, 2025-10-31:
> Hi [@koch86](https://github.com/koch86), super, dass hat mir weitergeholfen. Http ist tatsächlich nicht als Protokoll hinterlegt und ich kann es auch nicht hinzufügen. Scheinbar wird es von meinem konkreten Modell nicht unterstützt, aber ich probiere es nochmal über den Kundensupport.
> 
> Besten Dank!

M.w. gibt es das xcenter mit und ohne Interface Modul. Über das Interface Modul wird modbus TCP und wohl auch der http Zugriff lokal bereitgestellt. Vermute bei @koch86 ist das Modul verbaut. 
Bei @lfleck vermutlich nicht.

rilezi, 2025-10-31:
> Hallo [@Offroadbiker](https://github.com/Offroadbiker), ich habe nahezu das gleiche Setup. Das Auslesen einzelner Register klappt aber nicht. Bei Test z.B. über modbustester kommt lediglich connection error. Ich wäre für Hilfe sehr dankbar. Setup: Xcenter40 (nicht pro), homeassistant, Waveshare modbus-RTU-to TCP Converter (allerdings [diesen hier](https://www.amazon.de/-/en/Ethernet-Converter-Industrial-Insulated-Function/dp/B0BNKZ1LNJ/ref=sr_1_5?crid=W3C1DNOBC2YX&dib=eyJ2IjoiMSJ9.iS0qtDiz_XdP5Sdxs7gy8IkeTbmK5Xme_L8B3OSWcIhyn-eIX2x5YWwqz_Yv014R4Nd4N_Fyu66TSToEVC7JQBEMvuZJesPgZZBjuA4BSYiZwkDBEIs93TZyVkiqEj_wfUyte-HI2Y4beKbMC825wNdAr55ys_JdGwdmkqMaF_pwJpqp0B64v1vFqDrSLTSjGlZVzd7idr5lxRkfAxhLtNspTxs7iMxOMxwyWT0YvM_vhCbWBfu8KqWxKAPDNCHu8ovQdT8X8ptHwT1Gmqu5yuhanWoye5bnp-F5tduLD0A.RhOOsXH5fGMZrDxQksfgh3ZwmAnw5CHS7BClzPrXsX4&dib_tag=se&keywords=waveshare+modbus+converter&qid=1759672319&s=computers&sprefix=waveshare+modbus+converter%2Ccomputers%2C78&sr=1-5)
> 
> Habe den Kontakt TA des Waveshare mit 3 beim XBUS und TB mit 4 beim XBUS verbunden. Die Active und Link LEDs leuchten blau. Waveshare ist über TCP erreichbar. Baudrate 38.400, Databits 8, Parität none, Stopbist one. Szenen "Grundeinstellungen Freigabe ext. Steuergerät 2.0" und "Freigabe ext. Steuergerät 2.0) sind aktiv (durch Kermi gesetzt).
> 
> Woran könnte es liegen, dass ich die Register nicht auslesen kann? Ist ggf. noch eine Einstellung im XCenter zu ändern?
> 
> Vielen Dank schon mal.

Kann mir hier jemand weiterhelfen?