evcc-io/evcc #22568: logic error, but charging

Lechuza365, 2025-07-27:
### Describe the bug

Hello,

I am using evcc with a Zaptec Go 2 Wallbox and an the Car Skoda Enyaq 2025.
When I change the charge Mode, then I have an Error:
„charger logic error: disabled but charging“.

Is this a communication problem of the Zaptec Go 2 Wallbox with evcc?

Thanks.

Best regards
Marcus

### Steps to reproduce

1. change charge mode from PV to off
2.
3.
...

### Configuration details

```yaml
# Sponsortoken, valid until 2028-04-25
sponsortoken: ****

#Stromrarif
#tariffs:
#currency: EUR # (default EUR)
#  grid:
    # static grid price
#    type: fixed
#    price: 0.29 # EUR/kWh
    
#  feedin:
  # rate for feedin excess (pv) energy for the grid
#  type: fixed
#  price: 0.08 # EUR/kWh
```

### Log details

```text
[lp-1 ] DEBUG 2025/07/27 22:06:53 set charge mode: off
[lp-1 ] DEBUG 2025/07/27 22:06:53 pv timer reset
[lp-1 ] DEBUG 2025/07/27 22:06:53 pv timer inactive
[site ] DEBUG 2025/07/27 22:06:53 ----
[lp-1 ] DEBUG 2025/07/27 22:06:53 charge power: 10790W
[lp-1 ] DEBUG 2025/07/27 22:06:53 charge currents: [15.3 15.5 16]A
[site ] DEBUG 2025/07/27 22:06:53 grid power: 11410W
[site ] DEBUG 2025/07/27 22:06:53 grid currents: [0 0 0]A
[site ] DEBUG 2025/07/27 22:06:53 pv power: 0W
[site ] DEBUG 2025/07/27 22:06:53 site power: 11410W
[lp-1 ] DEBUG 2025/07/27 22:06:53 detected active phases: 3p
[lp-1 ] DEBUG 2025/07/27 22:06:53 charger status: C
[lp-1 ] DEBUG 2025/07/27 22:06:53 vehicle soc: 47%
[lp-1 ] DEBUG 2025/07/27 22:06:53 vehicle soc limit: 80%
[lp-1 ] DEBUG 2025/07/27 22:06:53 vehicle range: 249km
[lp-1 ] DEBUG 2025/07/27 22:06:53 plan: charge 2h33m27s between 2025-07-28 04:26:33 +0200 CEST until 2025-07-28 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.000)
[lp-1 ] DEBUG 2025/07/27 22:06:55 charger disable
[lp-1 ] DEBUG 2025/07/27 22:06:55 wake-up timer: stop
[lp-1 ] DEBUG 2025/07/27 22:06:56 set charge mode: pv
[site ] DEBUG 2025/07/27 22:06:56 ----
[lp-1 ] DEBUG 2025/07/27 22:06:56 charge power: 10790W
[lp-1 ] DEBUG 2025/07/27 22:06:56 charge currents: [15.3 15.5 16]A
[site ] DEBUG 2025/07/27 22:06:56 grid power: 11421W
[site ] DEBUG 2025/07/27 22:06:56 grid currents: [0 0 0]A
[site ] DEBUG 2025/07/27 22:06:56 pv power: 0W
[site ] DEBUG 2025/07/27 22:06:56 site power: 11421W
[lp-1 ] DEBUG 2025/07/27 22:06:56 detected active phases: 3p
[lp-1 ] DEBUG 2025/07/27 22:06:56 charger status: C
[lp-1 ] DEBUG 2025/07/27 22:06:56 vehicle soc: 47%
[lp-1 ] DEBUG 2025/07/27 22:06:56 vehicle soc limit: 80%
[lp-1 ] DEBUG 2025/07/27 22:06:56 vehicle range: 249km
[lp-1 ] WARN 2025/07/27 22:06:56 charger logic error: disabled but charging
[lp-1 ] DEBUG 2025/07/27 22:06:56 plan: charge 2h33m27s between 2025-07-28 04:26:33 +0200 CEST until 2025-07-28 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.000)
[lp-1 ] DEBUG 2025/07/27 22:06:56 available power -631W < 4140W min 3p threshold
[lp-1 ] DEBUG 2025/07/27 22:06:56 start phase scale1p timer
[lp-1 ] DEBUG 2025/07/27 22:06:56 phase scale1p in 3m0s
[lp-1 ] DEBUG 2025/07/27 22:06:56 pv charge current: 0A = 0A + -16.6A (11421W @ 3p)
[site ] DEBUG 2025/07/27 22:07:11 ----
[lp-1 ] DEBUG 2025/07/27 22:07:11 charge power: 0W
[lp-1 ] DEBUG 2025/07/27 22:07:11 charge currents: [0.02 0.022 0.03]A
[site ] DEBUG 2025/07/27 22:07:11 grid power: 554W
[site ] DEBUG 2025/07/27 22:07:11 grid currents: [0 0 0]A
[site ] DEBUG 2025/07/27 22:07:11 pv power: 0W
[site ] DEBUG 2025/07/27 22:07:11 site power: 554W
[lp-1 ] DEBUG 2025/07/27 22:07:11 charger status: B
[lp-1 ] INFO 2025/07/27 22:07:11 stop charging <-
[lp-1 ] DEBUG 2025/07/27 22:07:12 vehicle soc: 50%
[lp-1 ] DEBUG 2025/07/27 22:07:12 vehicle soc limit: 80%
[lp-1 ] DEBUG 2025/07/27 22:07:12 vehicle range: 266km
[lp-1 ] DEBUG 2025/07/27 22:07:12 plan: charge 2h19m30s between 2025-07-28 04:40:30 +0200 CEST until 2025-07-28 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.000)
[lp-1 ] DEBUG 2025/07/27 22:07:12 available power -554W < 4140W min 3p threshold
[lp-1 ] DEBUG 2025/07/27 22:07:12 phase scale1p in 0s
[lp-1 ] DEBUG 2025/07/27 22:07:12 switched phases: 1p
[lp-1 ] DEBUG 2025/07/27 22:07:12 phase timer inactive
[lp-1 ] DEBUG 2025/07/27 22:07:12 pv charge current: 0A = 0A + -2.41A (554W @ 1p)
[site ] DEBUG 2025/07/27 22:07:41 ----
[lp-1 ] DEBUG 2025/07/27 22:07:41 charge power: 0W
[lp-1 ] DEBUG 2025/07/27 22:07:41 charge currents: [0.02 0.022 0.03]A
[site ] DEBUG 2025/07/27 22:07:41 grid power: 546W
[site ] DEBUG 2025/07/27 22:07:41 grid currents: [0 0 0]A
[site ] DEBUG 2025/07/27 22:07:41 pv power: 0W
[site ] DEBUG 2025/07/27 22:07:41 site power: 546W
[lp-1 ] DEBUG 2025/07/27 22:07:41 charger status: B
[lp-1 ] DEBUG 2025/07/27 22:07:41 plan: charge 2h19m30s between 2025-07-28 04:40:30 +0200 CEST until 2025-07-28 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.000)
[lp-1 ] DEBUG 2025/07/27 22:07:41 pv charge current: 0A = 0A + -2.37A (546W @ 1p)
[site ] DEBUG 2025/07/27 22:08:11 ----
[lp-1 ] DEBUG 2025/07/27 22:08:11 charge power: 0W
[lp-1 ] DEBUG 2025/07/27 22:08:11 charge currents: [0.02 0.022 0.03]A
[site ] DEBUG 2025/07/27 22:08:11 grid power: 552W
[site ] DEBUG 2025/07/27 22:08:11 grid currents: [0 0 0]A
[site ] DEBUG 2025/07/27 22:08:11 pv power: 0W
[site ] DEBUG 2025/07/27 22:08:11 site power: 552W
[lp-1 ] DEBUG 2025/07/27 22:08:11 charger status: B
[lp-1 ] DEBUG 2025/07/27 22:08:11 plan: charge 2h19m30s between 2025-07-28 04:40:30 +0200 CEST until 2025-07-28 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.000)
[lp-1 ] DEBUG 2025/07/27 22:08:11 pv charge current: 0A = 0A + -2.4A (552W @ 1p)
[site ] DEBUG 2025/07/27 22:08:41 ----
[lp-1 ] DEBUG 2025/07/27 22:08:41 charge power: 0W
[lp-1 ] DEBUG 2025/07/27 22:08:41 charge currents: [0.02 0.022 0.03]A
[site ] DEBUG 2025/07/27 22:08:41 grid power: 591W
[site ] DEBUG 2025/07/27 22:08:41 grid currents: [0 0 0]A
[site ] DEBUG 2025/07/27 22:08:41 pv power: 0W
[site ] DEBUG 2025/07/27 22:08:41 site power: 591W
[lp-1 ] DEBUG 2025/07/27 22:08:41 charger status: B
[lp-1 ] DEBUG 2025/07/27 22:08:41 plan: charge 2h19m30s between 2025-07-28 04:40:30 +0200 CEST until 2025-07-28 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.000)
[lp-1 ] DEBUG 2025/07/27 22:08:41 pv charge current: 0A = 0A + -2.57A (591W @ 1p)
[site ] DEBUG 2025/07/27 22:09:11 ----
[lp-1 ] DEBUG 2025/07/27 22:09:11 charge power: 0W
[lp-1 ] DEBUG 2025/07/27 22:09:11 charge currents: [0.02 0.022 0.03]A
[site ] DEBUG 2025/07/27 22:09:11 grid power: 588W
[site ] DEBUG 2025/07/27 22:09:11 grid currents: [0 0 0]A
[site ] DEBUG 2025/07/27 22:09:11 pv power: 0W
[site ] DEBUG 2025/07/27 22:09:11 site power: 588W
[lp-1 ] DEBUG 2025/07/27 22:09:11 charger status: B
[lp-1 ] DEBUG 2025/07/27 22:09:11 plan: charge 2h19m30s between 2025-07-28 04:40:30 +0200 CEST until 2025-07-28 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.000)
[lp-1 ] DEBUG 2025/07/27 22:09:11 pv charge current: 0A = 0A + -2.56A (588W @ 1p)
[site ] DEBUG 2025/07/27 22:09:41 ----
[lp-1 ] DEBUG 2025/07/27 22:09:41 charge power: 0W
[lp-1 ] DEBUG 2025/07/27 22:09:41 charge currents: [0.02 0.022 0.03]A
[site ] DEBUG 2025/07/27 22:09:41 grid power: 581W
[site ] DEBUG 2025/07/27 22:09:41 grid currents: [0 0 0]A
[site ] DEBUG 2025/07/27 22:09:41 pv power: 0W
[site ] DEBUG 2025/07/27 22:09:41 site power: 581W
[lp-1 ] DEBUG 2025/07/27 22:09:41 charger status: B
[lp-1 ] DEBUG 2025/07/27 22:09:41 plan: charge 2h19m30s between 2025-07-28 04:40:30 +0200 CEST until 2025-07-28 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.000)
[lp-1 ] DEBUG 2025/07/27 22:09:41 pv charge current: 0A = 0A + -2.53A (581W @ 1p)
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.205.0

mfuchs1984, 2025-07-27:
It looks like you switched from fast to off and then pv mode within just 2 seconds which triggered the warning since the charger was not able to react in such a short time.

Lechuza365, 2025-07-27:
That is correct. I will change my behavior and see if the message appears again. Thanks for the quick help.