evcc-io/evcc #22612: Mennekes Smart Premium 22 (ACU V4) Charge Power not available

MasterOfPuppets111, 2025-07-29:
### Describe the bug

Hallo Zusammen,

Ich bekomme in evcc leider von beiden Ladepunkten den Fehler "chargepower not available"

[image]

In der Ladesäule sind 2 EHZ Zähler für jeden Ladepunkt verbaut, die rückseitig über die optische Schnitstelle ausgelesen werden. In den Einstellungen der beiden SCU's (Socet Control Unit) gibt es folgende Zählertypen zur Auswahl:

[image]

Nur wenn der richtige Zählertyp ausgewählt ist (EHZ ELD21) wird zumindest der Zählerstand über das OCPP Protokoll über die ACU and EVCC übertragen, zumindest findet man den Zählerstand in den Logs von EVCC, leider jedoch nicht die Aktuelle Leistung....

Auch verwundert es mich, dass in den Logs von EVCC  bei der Antwort von der Ladesäule als "Meter Typ" IEC61107 angegeben ist. Weiß einer woran das liegt, oder gibt es eine Möglichkeit herauszufinden, ob die Ladesäule überhaupt die Momentanleistung über OCPP ausgibt? Der EHZ kann es über die optische Schnittstelle auf jeden Fall (Neben Spannungs- & Stromwerten), die Frage ist nur, ob die ACU von Mennekes nur den Zählerstand über OCPP weiter gibt oder auch mehr und es einfach nicht richtig in EVCC implementiert ist bei dem Mennekes template....

Ich hoffe ihr könnt mir auch hier helfen 🙂

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
levels:
  cache: error

# unique installation id
plant: xxx

interval: 30s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.1.132  
  port: 502  
  meterid: 1  
  name: grid1
- type: template
  template: victron-energy 
  usage: pv  
  host: 192.168.1.132  
  port: 502  
  meterid: 1  
  name: pv2
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.1.132  
  port: 502  
  meterid: 1  
[... 114 lines cut ...]
[ocpp ] TRACE 2025/07/27 17:30:49 send Mennekes: [2,"2169605327","TriggerMessage",{"requestedMessage":"StatusNotification","connectorId":1}]
[ocpp ] TRACE 2025/07/27 17:30:49 recv Mennekes: [3, "2169605327", {"status": "Accepted"}]
[ocpp ] TRACE 2025/07/27 17:30:49 send Mennekes: [2,"1492539085","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":2}]
[ocpp ] TRACE 2025/07/27 17:30:50 recv Mennekes: [3, "1492539085", {"status": "Accepted"}]
[ocpp ] TRACE 2025/07/27 17:30:51 send Mennekes: [2,"3690268563","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":2}]
[ocpp ] TRACE 2025/07/27 17:30:52 recv Mennekes: [3, "3690268563", {"status": "Accepted"}]
[ocpp ] TRACE 2025/07/27 17:30:53 send Mennekes: [2,"503565216","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":2}]
[ocpp ] TRACE 2025/07/27 17:30:54 recv Mennekes: [3, "503565216", {"status": "Accepted"}]
[ocpp ] TRACE 2025/07/27 17:30:55 send Mennekes: [2,"3041953947","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":2}]
[ocpp ] TRACE 2025/07/27 17:30:56 recv Mennekes: [3, "3041953947", {"status": "Accepted"}]
[ocpp ] TRACE 2025/07/27 17:30:56 recv Mennekes: [2, "bb7d94ae-6afe-11f0-8ae2-000a14851572", "MeterValues", {"meterValue": [{"timestamp": "2025-07-27T15:31:13.000Z", "sampledValue": [{"measurand": "Energy.Active.Import.Register", "unit": "Wh", "value": "5429689"}]}], "connectorId": 2}]
[ocpp ] TRACE 2025/07/27 17:30:56 send Mennekes: [3,"bb7d94ae-6afe-11f0-8ae2-000a14851572",{}]
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

mfuchs1984, 2025-07-29:
Die wallbox meldet, dass sie nur einen einzigen Messwert übertragen kann

`{"readonly": true, "key": "metervaluessampleddatamaxlength", "value": "1"}`

mfuchs1984, 2025-07-29:
Die Fehlermeldung wirst du wahrscheinlich los, indem du `metervalues: Energy.Active.Import.Register` zur charger Konfiguration hinzufügst.

premultiply, 2025-07-29:
Wenn die Firmware nicht so mies wäre könnte sie wohl theoretisch noch die Leistung übermitteln.

```
[ocpp ] TRACE 2025/07/27 17:30:32 send Mennekes: [2,"2748903973","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import"}]
[ocpp ] TRACE 2025/07/27 17:30:32 recv Mennekes: [3, "2748903973", {"status": "Accepted"}]
[ocpp ] TRACE 2025/07/27 17:30:32 send Mennekes: [2,"1751246152","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
[ocpp ] TRACE 2025/07/27 17:30:32 recv Mennekes: [3, "1751246152", {"status": "Accepted"}]
```

Im Zweifelsfall halt einfach einen zusätzlichen Zähler pro Ladepunkt davor und diesen direkt in evcc einbinden.
Dann kommen auch sonst alle verfügbaren und sinnvollen die der Zähler liefert.

/cc @benesolar

benesolar, 2025-07-30:
Ich verwende eine Smart T Ladesäule.
Zählertype ist bei mir EMH E-Moc.

Software der Station ist zumindest mit der Version 3633 (3656 hab ich noch nicht drauf) im Klartext auf der SD Karte abgespeichert.

MasterOfPuppets111, 2025-07-30:
Danke euch 2 für eure Antworten, d.h. den Intervall am besten mal auf 60 Sekunden in der config stellen? Evtl. Sind dann auch die Timeouts beim ersten verbinden der Ladesäule mit evcc weg... Ich muss evcc & die ACU in einem bestimmten Abstand mehrmals nacheinander starten, damit sie connecten, ansonsten kommt es seitens evcc immer zu einem Timeout...
'''
[main ] FATAL 2025/07/27 17:31:04 charger [wallbox4] cannot create charger 'wallbox4': cannot create charger type 'template': cannot create charger type 'ocpp': timeout [main ] FATAL 2025/07/27 17:31:04 will attempt restart in: 15m0s
'''
 ...und evcc kann nicht starten, steht die Verbindung dann jedoch einmal ist alles gut und sie bleibt bestehen.

Ja ich hatte ein Auto mal eingesteckt, habe aber leider im Moment keines da...

Habe noch folgende Info vom Mennekes Support bekommen, auch wenn die Firmware nicht die beste ist, sind die Leute da extrem hilfsbereit:

Die 3656 war an sich eine Reaktion auf ein geändertes EU-Gesetz und deshalb schon eine Ausnahme.

So eine Gesetzesänderung kommt vermutlich auch nochmal und wird auch noch ein Update nach sich ziehen, allerdings werden wir im Funktionsumfang nichts mehr verändern oder ergänzen.

 

Der Updateserver ist schon eine ganze weile heruntergefahren. Man kann aber noch selbst Updates einspielen, über Hiddenlinks:

http://IP_Adresse/admin/admin_local_acu_upgrade

http://IP_Adresse/admin/admin_local_scu_upgrade

Allerdings ist die 3656, wie schon gesagt, aktuell das Ende der Fahnenstange.

 

Soweit ich weiß, kann der Optokoppler zwischen Zähler und SCU Leistungs- und Stromdaten übermitteln. Spannung und Frequenz sind da soweit ich weiß nicht möglich.

Das OCPP Protokoll ist von diesen System leider wirklich etwas hinterher und in der genauen Ausführung bin ich mir auch unsicher was genau die ACU übermitteln kann und was nicht.

 

Ich werde mal einen Kollegen ansprechen, was bei der ACU im Bereich OCPP alles möglich ist. Die EDL21 Zähler sind nämlich auch schon etwas älter. Die neueren ab 2018 verwenden E-MOC.

Ab diesen Modellen ging es dann nämlich um Eichrecht und da wurden diese Dinge auf einmal relevant.

"'

Eventuell ist es sinnvoll die Zähler durch welche zu ersetzen, die E-MOC unterstützen. Ich versuche aber erstmal Meterintervall auf 60s zu setzen und hänge dann nochmal ein Auto an.

MasterOfPuppets111, 2025-07-30:
Ah okay, kommt es sich aber evtl. in die Quere mit dieser config?
  ```
remotestart: false  
  connecttimeout: 5m  
  meterinterval: 10s  
```

Habe den Teil selbst nicht gesetzt, dass hat der Assistent gemacht.

mfuchs1984, 2025-07-30:
> Eventuell ist es sinnvoll die Zähler durch welche zu ersetzen, die E-MOC unterstützen. 

Da die Box nur einen einzigen Wert pro MeterValues Nachricht übermitteln kann, kannst du dir Versuche mit internen Zählern wohl sparen. Sofern du wirklich unbedingt die echte Charge Power brauchst, installiere wie on @premultiply vorgeschlagen, einen externen Zähler, den evcc unterstützt.

mfuchs1984, 2025-07-30:
> dann übermittelt die box eventuell mehr

Deine Box kann die Power laut ocpp trace theoretisch jetzt schon übertragen, tut es aber nicht, das die konkrete Implementierung des ocpp Protokolls auf einen Messwert limitiert ist.

MasterOfPuppets111, 2025-07-30:
Hier wäre halt die Frage, ob die ganze OCPP Softwareimplementierung auf der SD-Karte der ACU liegt (ein Backup werde ich nachher definitiv machen) evtl. kann @benesolar etwas dazu sagen, bei ihm werden ja anscheinend die Momentanwerte übertragen, evtl kann man die Implementierung auf der SD-Karte glatt ziehen.

MasterOfPuppets111, 2025-07-30:
Neue Antwort vom Support, echt bemüht die Jungs und ziemlich fix 😀

```
Eine Anfrage akzeptieren und Sie auch beantworten können sind zwei paar Schuhe 😉

Bei Fragen geben Sie aber alles gerne durch. Was ich nicht selbst beantworten kann erfrage ich dann.

 

Die Hardware in der die ACUs verbaut wurden ist sehr hartnäckig und wird uns noch viele Jahre begleiten.

Deshalb ist es sicher nicht verkehrt hier auch nach Abkündigung dieser Produkte selbst noch was dazu zu lernen.

 

Auf der SD Karte ist alles an Software was von uns kommt. Ohne die gibt es keine Funktion.

 

Hat der User mit der Smart T denn eine mit ACU? Die gibt es nämlich auch mit den neueren Bender Controllern – welche wesentlich mehr können.

Und bevor die Frage kommt, lässt sich das leider nicht umrüsten ☹

```

@benesolar hast du tatsächlich eine ACU oder einen Bender Controller?

benesolar, 2025-08-01:
Ja, kannst du.

Soweit ich das verstehe bedeutet E-MOC mit Funktionen für Eichrechtskonforme Ladestationen.

Dieser Punkt aus dem Datenblatt ist für uns interessant:
[image]

MasterOfPuppets111, 2025-12-30:
Hallo @benesolar habe mittlerweile einen E-Moc fähigen Zähler ergattern können. Welcher Zählertyp ist denn bei dir im SCU Setup eingestellt? Tatsächlich E-Moc oder trotzdem ELD? Mit E-Moc Einstellung überträgt er leider gar keine Werte mehr und geht in der ACU auf Störung :(.

Gruß Pierre

benesolar, 2025-12-30:
EMH e-moc.

Die richtigen Leseköpfe hinter den Zählern sind verbaut und an den SCUs angeschlossen?
https://emh-metering.com/produkte/kommunikation/okk-bke-gen-f/

Das sind die Einstellungen bei mir:
[image]