evcc-io/evcc #22651: Kia template can not create after update on 0.206

musikus24, 2025-08-01:
### Describe the bug

Nach dem Update wird mein Auto ein Kia nicht mehr verbunden. Wenn ich die Anmeldedaten eingeben und überprüfen lasse, kommt immer folgende Fehlermeldung:

cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: unexpected status: 401 (Unauthorized): Require authentication (4010)

### Steps to reproduce

1.update durchführen
2.IN der UI auf Konfiguration
3. Anmeldedaten erneut eingeben und prüfen
...

### Configuration details

```yaml
Alles über die UI eingegeben.
```

### Log details

```text
Keine Möglichkeit
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.206

hans-peter-st, 2025-08-02:
Gleiches Login Problem bei Hyundai!

andig, 2025-08-02:
Danke dass ihr Euch die Mühe gemacht habt, das exakt identische Issue direkt unter diesem zu finden und nicht doppelt zu posten.