evcc-io/evcc #22665: falsche Werte und Anzeige ab PV-Erzeugung über 10 kW, unterhalb 10 kW stimmen Werte und Anzeige

mrlehm, 2025-08-01:
### Describe the bug

Hallo zusammen,
ich habe unter HomeAssistant zuerst das evcc-Addon und später ha-evcc eingebunden. Seit Beginn habe ich das Problem, dass evcc bis zu PV-Erzeugungswerten von etwa 10 kW alles ordentlich darstellt, über ca. 10 kW PV-Erzeugung aber immer Netzbezug in falscher Größe anstatt Einspeisung anzeigt. Es ist dauerhaft reproduzierbar und ich finde weder einen helfenden Forumsbeitrag, noch komme ich selbst auf einen Lösungsansatz. Alle weiteren Nutzer der Werte wie das Energie-Dashboard und das Energieflussdiagramm zeigen dauerhaft richtige Werte an. Also scheint die Datenbasis ok zu sein. Folgend habe ich versucht mit Protokolleinträgen und Screenshots den Zustand möglichst genau zu dokumentieren. Über eure Unterstützung freue ich mich sehr - gerne auch ausführlich, da ich in dieser Umgebung neu bin und erst ins Thema kommen muss - Danke.

[image]

Protokoll mit 3 fehlerhaften Datenblöcken = PV-Ertrag größer 10 kW = falsche Werte und Anzeige,
folgend von einem korrekten Datenblock = PV-Ertrag unter 10 kW = richtige Werte (- Vorzeichen) und Anzeige
folgend wieder 1 fehlerhafter Datenblock = PV-Ertrag größer 10 kW = falsche Werte und Anzeige

[image]

evcc mit korrekter Anzeige bei PV-Ertrag unter 10 kW

[image]

Energieflussdiagramm bei PV-Ertrag unter 10 kW = korrekt

[image]

evcc mit fehlerhafter Anzeige bei PV-Ertrag über 10 kW, PV-Ertrag von 13,6 kW ist korrekt, Netzbezug ist real nicht vorhanden und Einspeisewert von ca. 11,6 kW wird nicht angezeigt

[image]

Energieflussdiagramm bei PV-Ertrag über 10 kW = korrekt

Die Werte in den Screenshots sollen nur das Problem widerspiegeln. Da die Screenshots hintereinander gemacht wurden, differieren die Werte ein wenig. 

Was kann die Ursache sein?

### Steps to reproduce

1. bis ca. 10 kW PV-Ertrag (Erzeugung) werden Werte im Protokoll und in der Anzeige richtig ausgegeben
2. über ca. 10 kW PV-Ertrag (Erzeugung) werden Werte im Protokoll und in der Anzeige fehlerhaft ausgegeben
3.
...

### Configuration details

```yaml
wenn benötigt gerne später
```

### Log details

```text
siehe oben
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

nightly.20250801-dc7d1a7

andig, 2025-08-01:
> wenn benötigt gerne später

Sie werden benötigt und deshalb angefragt. Issues ohne werden mangels Kontext geschlossen. Logs bitte als Text. Screenshots sind nicht durchsuchbar.

mrlehm, 2025-08-01:
Werden umgehend nachgereicht...