evcc-io/evcc #22668: evcc zeigt über PV-Ertrag von ca. 10 kW falsche Werte, unter ca. 10 kW PV-Ertrag wird alles ordnungsgemäß angezeigt

mrlehm, 2025-08-01:
### Describe the bug

Hallo zusammen,
ich habe unter HomeAssistant zuerst das evcc-Addon und später ha-evcc eingebunden. Seit Beginn habe ich das Problem, dass evcc bis zu PV-Erzeugungswerten von etwa 10 kW alles ordentlich darstellt, über ca. 10 kW PV Erzeugung aber immer Netzbezug in falscher Größe anstatt Einspeisung anzeigt. Es ist dauerhaft reproduzierbar und ich finde weder einen helfenden Forumsbeitrag, noch komme ich selbst auf einen Lösungsansatz. Alle weiteren Nutzer der Werte wie das Energie-Dashboard und das Energieflussdiagramm zeigen dauerhaft richtige Werte an. Also scheint die Datenbasis ok zu sein. Folgend habe ich versucht mit Protokolleinträgen und Screenshots den Zustand möglichst genau zu dokumentieren. Über eure Unterstützung freue ich mich sehr - gerne auch ausführlich, da ich in dieser Umgebung neu bin und erst ins Thema kommen muss - Danke.

[image]

evcc zeigt unter ca. 10 kW PV-Ertrag die richtigen Werte

[image]

Energieflussdiagramm zeigt dauerhaft unter und über 10 kW PV-Ertrag die richtigen Werte = Datenbasis ok

[image]

evcc zeigt über ca. 10 kW PV-Ertrag falsche Werte, PV-Ertrag mit 10,9 kW ist korrekt, angezeigter Netzbezug von 9,2 kW ist nicht vorhanden, Einspeiseleistung wird nicht angezeigt

[image]

Energieflussdiagramm zeigt dauerhaft unter und über 10 kW PV-Ertrag die richtigen Werte = Datenbasis ok

Die Screenshots wurden zeitlich hintereinander erstellt und sollen nur die Problemstellung verdeutlichen. Werte zwischen Scrennshots und Protokoll können abweichen. 

Wo kann das Problem liegen?

### Steps to reproduce

1. über ca. 10 kW PV-Ertrag = falsche Werte und Anzeige
2. unter ca. 10 kW PV-Ertrag = richtige Werte und Anzeige
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.xxx.xxx
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

[... 276 lines cut ...]
[lp-1 ] DEBUG 2025/08/01 16:34:18 charge currents: [0 0 0]A
[site ] DEBUG 2025/08/01 16:34:19 battery 1 power: 19W
[site ] DEBUG 2025/08/01 16:34:20 grid power: -9222W
[site ] DEBUG 2025/08/01 16:34:21 battery 1 soc: 100%
[site ] DEBUG 2025/08/01 16:34:23 pv 1 power: 10320W
[site ] DEBUG 2025/08/01 16:34:23 site power: -9103W
[lp-1 ] DEBUG 2025/08/01 16:34:23 charge total import: 3003.903kWh
[lp-1 ] DEBUG 2025/08/01 16:34:23 charger status: B
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

nightly.20250801-dc7d1a7

andig, 2025-08-02:
Bitte

    evcc meter --log trace,db:error

für die "richtigen" und falschen Werte.

andig, 2025-08-02:
Closing until logs available.

mrlehm, 2025-08-03:
Hallo andig,
vielen Dank für die Unterstützung. Leider scheint seit gestern bei uns keine Sonne mehr. Somit kommen wir nicht an die 10 kW Ertragsgrenze, um die gewünschten Log´s senden zu können. Sobald wieder ausreichend Sonne scheint, sende ich die Log´s. Die folgenden Screenshots zeigen die gewünschten Log-Einstellungen - sind das die Richtigen? Aktuell zeigen sie teilweise keine Inhalte! 
Ich will nur sicher gehen, dass ich Ihre Anfrage richtig verstanden habe und die richtigen Log´s sende, wenn die Sonne wieder scheint und das Problem wieder auftritt.

[image]

[image]

[image]

Mit sonnigen Grüßen

andig, 2025-08-03:
Je nachdem was Du mit "PV Ertrag" meinst musst du bitte den richtigen Zähler auswerten. Batterie ist sicher irrelevant. Bitte Logs als Text, **KEINE** Screenshots.

mrlehm, 2025-08-03:
Dann schlage ich vor die Protokolle über alle 10 Bereiche über DEBAG, TRACE und ERROR in durchsuchbarer Textform zu senden. Ich kann für mich nämlich nicht abschätzen, inwieweit die Einträge von proxy und site z.B. relevant sind. LG

mrlehm, 2025-08-03:
Hallo andig, 
die Sonne hat sich doch noch blicken lassen. Hier die Protokolleinträge mit folgenden Einstellungen (unterhalb durchsuchbar und lesbar). Die Screenshots dienen nur der Verdeutlichung der Auswahlkriterien. Damit kannst du sehen, dass in meter - Bereichen keine Werte zu finden sind:

[image]
DEBAG über alle Bereiche (Bereiche mit "grid power" und "site power" = (-) Vorzeichen = Werte und Anzeige richtig, Bereiche mit "grid power" und "site power" = (+) Vorzeichen = Werte und Anzeige falsch) -> "pv 1 power" unter- bzw. oberhalb ca. 10 kW.

```
[lp-1 ] DEBUG 2025/08/03 11:54:48 charge power: 0W
[lp-1 ] DEBUG 2025/08/03 11:54:48 charge currents: [0 0 0]A
[site ] DEBUG 2025/08/03 11:54:48 battery 1 power: 22W
[site ] DEBUG 2025/08/03 11:54:50 battery 1 soc: 99%
**[site ] DEBUG 2025/08/03 11:54:52 grid power: -8150W
[site ] DEBUG 2025/08/03 11:54:53 pv 1 power: 8500W
[site ] DEBUG 2025/08/03 11:54:53 site power: -8028W**
[lp-1 ] DEBUG 2025/08/03 11:54:53 charge total import: 3003.903kWh
[lp-1 ] DEBUG 2025/08/03 11:54:53 charger status: B
[site ] DEBUG 2025/08/03 11:55:18 ----
[lp-1 ] DEBUG 2025/08/03 11:55:18 charge power: 0W
[lp-1 ] DEBUG 2025/08/03 11:55:18 charge currents: [0 0 0]A
[site ] DEBUG 2025/08/03 11:55:18 battery 1 power: 22W
[site ] DEBUG 2025/08/03 11:55:19 grid power: 8298W
**[site ] DEBUG 2025/08/03 11:55:19 battery 1 soc: 99%
[site ] DEBUG 2025/08/03 11:55:22 pv 1 power: 11288W
[site ] DEBUG 2025/08/03 11:55:22 site power: 8420W**
[lp-1 ] DEBUG 2025/08/03 11:55:22 charge total import: 3003.903kWh
[lp-1 ] DEBUG 2025/08/03 11:55:22 charger status: B
[site ] DEBUG 2025/08/03 11:55:48 ----
[lp-1 ] DEBUG 2025/08/03 11:55:48 charge power: 0W
[lp-1 ] DEBUG 2025/08/03 11:55:48 charge currents: [0 0 0]A
[site ] DEBUG 2025/08/03 11:55:48 battery 1 power: 0W
[site ] DEBUG 2025/08/03 11:55:49 grid power: 6375W
[site ] DEBUG 2025/08/03 11:55:49 battery 1 soc: 99%
[site ] DEBUG 2025/08/03 11:55:52 pv 1 power: 13504W
[site ] DEBUG 2025/08/03 11:55:52 site power: 6475W
[lp-1 ] DEBUG 2025/08/03 11:55:52 charge total import: 3003.903kWh
[lp-1 ] DEBUG 2025/08/03 11:55:52 charger status: B
[site ] DEBUG 2025/08/03 11:56:18 ----
[lp-1 ] DEBUG 2025/08/03 11:56:18 charge power: 0W
[lp-1 ] DEBUG 2025/08/03 11:56:18 charge currents: [0 0 0]A
[site ] DEBUG 2025/08/03 11:56:18 battery 1 power: 26W
[site ] DEBUG 2025/08/03 11:56:19 grid power: 8977W
[site ] DEBUG 2025/08/03 11:56:19 battery 1 soc: 99%
[site ] DEBUG 2025/08/03 11:56:22 pv 1 power: 10752W
[site ] DEBUG 2025/08/03 11:56:22 site power: 9103W
[lp-1 ] DEBUG 2025/08/03 11:56:22 charge total import: 3003.903kWh
[lp-1 ] DEBUG 2025/08/03 11:56:22 charger status: B
[site ] DEBUG 2025/08/03 11:56:48 ----
[lp-1 ] DEBUG 2025/08/03 11:56:48 charge power: 0W
[lp-1 ] DEBUG 2025/08/03 11:56:48 charge currents: [0 0 0]A
[site ] DEBUG 2025/08/03 11:56:48 battery 1 power: 26W
[site ] DEBUG 2025/08/03 11:56:49 battery 1 soc: 99%
[site ] DEBUG 2025/08/03 11:56:51 grid power: 7730W
[site ] DEBUG 2025/08/03 11:56:52 pv 1 power: 12323W
[site ] DEBUG 2025/08/03 11:56:52 site power: 7856W
[lp-1 ] DEBUG 2025/08/03 11:56:52 charge total import: 3003.903kWh
[lp-1 ] DEBUG 2025/08/03 11:56:52 charger status: B
[site ] DEBUG 2025/08/03 11:57:18 ----
```

[image]
**DEBAG über meter-Bereiche**

[image]
**ERROR über alle Bereiche**

[image]
**TRACE über alle Bereiche**

```
[keba ] TRACE 2025/08/03 11:54:48 modbus: send f5 22 00 00 00 06 ff 03 03 fc 00 02
[... 1110 lines cut ...]
[proxy-5201] TRACE 2025/08/03 11:57:18 read holding: 00 00 00 00
[keba ] TRACE 2025/08/03 11:57:18 modbus: recv f5 5c 00 00 00 07 ff 03 04 00 00 00 00
[lp-1 ] DEBUG 2025/08/03 11:57:18 charge currents: [0 0 0]A
[my_grid] TRACE 2025/08/03 11:57:18 modbus: send 0b ae 00 00 00 06 01 03 97 a1 00 01
[proxy-5200] TRACE 2025/08/03 11:57:18 read holdings: id 1 addr 38817 qty 1
[proxy-5200] TRACE 2025/08/03 11:57:18 read holding: 92 86
[my_grid] TRACE 2025/08/03 11:57:18 modbus: recv 0b ae 00 00 00 05 01 03 02 92 86
[my_grid] TRACE 2025/08/03 11:57:18 modbus: send 0b af 00 00 00 06 01 03 97 a3 00 01
[proxy-5200] TRACE 2025/08/03 11:57:18 read holdings: id 1 addr 38819 qty 1
[proxy-5200] TRACE 2025/08/03 11:57:18 read holding: 90 e2
[my_grid] TRACE 2025/08/03 11:57:18 modbus: recv 0b af 00 00 00 05 01 03 02 90 e2
[my_battery] TRACE 2025/08/03 11:57:18 modbus: send 0b b0 00 00 00 06 01 03 99 46 00 01
[proxy-5200] TRACE 2025/08/03 11:57:18 read holdings: id 1 addr 39238 qty 1
[proxy-5200] TRACE 2025/08/03 11:57:18 read holding: 00 1b
[my_battery] TRACE 2025/08/03 11:57:18 modbus: recv 0b b0 00 00 00 05 01 03 02 00 1b
[site ] DEBUG 2025/08/03 11:57:18 battery 1 power: 27W
[my_pv ] TRACE 2025/08/03 11:57:18 modbus: send 0b b1 00 00 00 06 01 03 99 70 00 01
[proxy-5200] TRACE 2025/08/03 11:57:18 read holdings: id 1 addr 39280 qty 1
[proxy-5200] TRACE 2025/08/03 11:57:18 read holding: 0e 8d
[my_pv ] TRACE 2025/08/03 11:57:18 modbus: recv 0b b1 00 00 00 05 01 03 02 0e 8d
[my_grid] TRACE 2025/08/03 11:57:18 modbus: send 0b b2 00 00 00 06 01 03 97 a5 00 01
[proxy-5200] TRACE 2025/08/03 11:57:18 read holdings: id 1 addr 38821 qty 1
```

Mit sonnigen Grüßen

andig, 2025-08-03:
Nichts davon braucht es und es hilft auch nicht: https://github.com/evcc-io/evcc/issues/22668#issuecomment-3146399475

andig, 2025-08-03:
Wo? Hab ich die übersehen? Wo ist die Ausgabe?

mrlehm, 2025-08-03:
Ich probiere nochmals das, was ich in den Kommentaren mit Screenshots zum Ausdruck bringen wollte, in Worte zu fassen.
Ich habe unter evcc (nightly) im Bereich Logs "DEBUG" über "Alle Bereiche" nur Einträge von [lp-1] und [site]. Für die meter-Einträge "my grid", "my pv" und "my battery" habe ich unter allen Protokollarten außer "TRACE" keine Einträge zu stehen. Unter der Protokollart "ERROR" habe ich für "db" ebenfalls keine Einträge. Falls ich unter "evcc meter --log trace,db:error" etwas völlig falsches verstanden habe, dann berichtige mich bitte und schreibe mir bitte genau und etwas ausführlicher, was du genau für deine Analyse benötigst und wo ich es finden kann. LG

andig, 2025-08-07:
Wir lesen 3 Werte vorzeichenbehaftet aus, also:

```
5ea6 // 24230
512c // 20780
5d48 // 23880

b1d0 // -20016 45520
a596 // -23146 42390
afc8 // -20536 45000
```

Das Vorzeichen **BRAUCHT** es da Netz vorzeichenbehaftet sein muss.

mrlehm, 2025-08-24:
Hallo premultiply, ich kann deine Verwirrung zur Fragestellung gut nachvollziehen. Die Fragestellung / das Problem / der Fehler wird erst deutlich, wenn man sich die Screenshots siehe Anfang der Fragestellung mit ansieht. Ich versuche es anhand von Log-Ausgaben und Screenshots nochmals chronologisch aufzubauen. Wobei die enthaltenen Werte in den Logs und den Screenshots nie gleich sind, da sie nur zeitversetzt von mir ausgegeben werden können und die Werte in der Zeit wegen unterschiedlicher Sonneneinstrahlung schwanken. 

Der folgende Fall schildert den **korrekten** Ablauf. Der PV-Ertrag ist kleiner 10 KW und evcc zeigt alle Werte korrekt in der Übersicht an. Erzeugung = 7,4 kW, Verbrauch = 0,4 kW und Einspeisung = 7,1 kW (mit Rundungsfehlern passt in Summe).

[image]

Ein passendes Log dazu gibt den Wert **Power: -6889W** korrekt mit negativem Vorzeichen an (da zeitverzögert ausgelesen "Power: -6889W" = "Einspeisung 7,1 kW").

my_grid
-------
[my_grid] TRACE 2025/08/07 11:45:56 modbus: send 00 03 00 00 00 06 01 03 97 a1 00 01
[my_grid] TRACE 2025/08/07 11:45:56 modbus: recv 00 03 00 00 00 05 01 03 02 5e a6
[my_grid] TRACE 2025/08/07 11:45:56 modbus: send 00 04 00 00 00 06 01 03 97 a3 00 01
[my_grid] TRACE 2025/08/07 11:45:57 modbus: recv 00 04 00 00 00 05 01 03 02 51 2c
[my_grid] TRACE 2025/08/07 11:45:57 modbus: send 00 05 00 00 00 06 01 03 97 a5 00 01
[my_grid] TRACE 2025/08/07 11:45:57 modbus: recv 00 05 00 00 00 05 01 03 02 5d 48
[my_grid] TRACE 2025/08/07 11:45:57 modbus: send 00 06 00 00 00 06 01 03 9a c2 00 01
[my_grid] TRACE 2025/08/07 11:45:58 modbus: recv 00 06 00 00 00 05 01 03 02 1f cc
**Power:  -6889W**
Energy: 81.4kWh

Der nun folgende Fall schildert den **fehlerhaften** Ablauf. Der PV-Ertrag ist **größer** 10 KW und evcc zeigt Werte fehlerhaft in der Übersicht an. Erzeugung = 10,9 kW richtig, Verbrauch = 20,1 kW falsch, Einspeisung = 0,0 kW falsch und Netzbezug = 9,2 kW falsch. In der Realität gibt es nur einen geringen Verbrauch und der Rest ist Einspeisung. Netzbezug gibt es in der Realität zu dieser Zeit überhaupt nicht. Das ist auch an den oben eingestellten Screenshots der Energieflussdiagramme gut sichtbar. Da die Energieflussdiagramme stimmen, muss die Datengrundlage vom Wechselrichter (meter) in Ordnung sein. Dieser kuriose Fehler tritt reproduzierbar und permanent bei einem Sonnenertrag = Erzeugung größer ca. 10 kW auf.

[image]

Ein passendes Log dazu gibt den Wert **Power: 6370W** fehlerhaft mit positiven Vorzeichen und falschem Wert an. Eigentlich müsste hier ein **Power: um -9950W** = Einspeisung stehen.

my_grid
-------
[my_grid] TRACE 2025/08/07 12:34:15 modbus: send 00 03 00 00 00 06 01 03 97 a1 00 01
[my_grid] TRACE 2025/08/07 12:34:16 modbus: recv 00 03 00 00 00 05 01 03 02 b1 d0
[my_grid] TRACE 2025/08/07 12:34:16 modbus: send 00 04 00 00 00 06 01 03 97 a3 00 01
[my_grid] TRACE 2025/08/07 12:34:16 modbus: recv 00 04 00 00 00 05 01 03 02 a5 96
[my_grid] TRACE 2025/08/07 12:34:16 modbus: send 00 05 00 00 00 06 01 03 97 a5 00 01
[my_grid] TRACE 2025/08/07 12:34:17 modbus: recv 00 05 00 00 00 05 01 03 02 af c8
[my_grid] TRACE 2025/08/07 12:34:17 modbus: send 00 06 00 00 00 06 01 03 9a c2 00 01
[my_grid] TRACE 2025/08/07 12:34:17 modbus: recv 00 06 00 00 00 05 01 03 02 1f cc
**Power:  6370W**
Energy: 81.4kWh

Zusammenfassung: Bis zu einem Wert der Erzeugung kleiner 10 kW zeigt evcc alle Werte in der GUI korrekt an und der **Power-Wert** ist korrekt mit negativen Vorzeichen wegen der Einspeisung. Über ca. 10 kW Erzeugung wird  der **Power-Wert** fehlerhaft und mit falschem Vorzeichen (+) ausgegeben. Das führt zu komplett fehlerhaften Anzeigewerte in der evcc - GUI. Es werden Netzbezüge angezeigt, die es in der Realität nicht gibt und Einspeisewerte, die es in der Realität im hohen kW-Bereichen gibt, werden fehlerhaft mit 0,0 kW angezeigt. Die Ausgabewerte müssen stimmen, da alle anderen Produkte wie Dashboards und Energieflussdiagramme die Ausgabewerte richtig interpretieren. Es würde mich freuen, wenn meine Fragestellung / mein Problem verstanden wurde und eine Lösung gefunden wird.

#andig - welchen Hersteller meinst du? Den Hersteller des Produktes evcc? Da alle anderen Produkte die Ausgabewerte korrekt interpretieren können, gehen ich mal davon aus, dass der Hersteller Fox ESS der PV-Anlage raus ist. Was meinst du?
Danke für eure Unterstützung und LG

andig, 2025-08-24:
> der Hersteller Fox ESS

Genau den. Doku wäre super.

> Da alle anderen Produkte die Ausgabewerte korrekt interpretieren können

Welche? Wo kann man sich das im Quelltext anschauen?

premultiply, 2025-08-24:
Hier nur mal zum Vergleich

HA:
https://github.com/StealthChesnut/HA-FoxESS-Modbus/blob/693669327df58de0169157edf45b7f1c812e7602/custom_components/HA-FoxESS-Modbus/modbus-H3-LAN.yaml#L261-L296

evcc:
https://github.com/evcc-io/evcc/blob/ab2f706b4605d14f8a982ec6a61c7535d1725c40/templates/definition/meter/fox-ess-h3.yaml#L26-L49

Ich sehe keinen Unterschied und erwarte daher auch keine andere Interpretation der Daten.

premultiply, 2025-08-25:
https://github.com/nathanmarlor/foxess_modbus/blob/main/custom_components/foxess_modbus/entities/entity_descriptions.py