evcc-io/evcc #22674: Webasto Next Verbindungsfehler

JensB73, 2025-08-01:
### Describe the bug

Moin,

evcc kann keine Verbindung zur Webasto-NEXT herstellen.

[main ] FATAL 2025/08/01 23:46:58 charger [db:4] cannot create charger 'db:4': cannot create charger type 'template': cannot create charger type 'webasto-next': heartbeat: dial tcp 192.168.1.115:502: i/o timeout

Ich habe an der Konfiguration seit Wochen nichts geändert.
Weder über die Experimentelle Konfiguration noch über die evcc.yaml funktioniert aktuell die Verbindung.

Aktuelle Version: 0.206.1 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:
  - name: my_charger
    type: template
    template: webasto-next
    host: 192.168.1.115 # IP address or hostname
    port: 502 # Port, optional
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/08/01 23:46:55 evcc 0.206.1
[main  ] INFO 2025/08/01 23:46:55 using config file: /config/evcc.yaml
[db    ] INFO 2025/08/01 23:46:55 using sqlite database: /data/evcc.db
[main  ] INFO 2025/08/01 23:46:55 UI listening at :7070
[main  ] FATAL 2025/08/01 23:46:58 charger [db:4] cannot create charger 'db:4': cannot create charger type 'template': cannot create charger type 'webasto-next': heartbeat: dial tcp 192.168.1.115:502: i/o timeout
[main  ] FATAL 2025/08/01 23:46:58 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.206.1

mfuchs1984, 2025-08-02:
Wallbox schonmal neu gestartet?

JensB73, 2025-08-02:
> Wallbox schonmal neu gestartet?

Ja, das war das erste. Die Konfigurationsseite war vorher und nachher über das Netzwerk erreichbar und läd auch mit den fallback-Werten. Daher würde ich die Wallbox als Auslöser für das Problem ausschließen.

StefanSchoof, 2025-08-02:
Probiere mal ein Downgrade und prüfe, ob es noch mit der alten evcc Version klappt.

andig, 2025-08-02:
> dial tcp 192.168.1.115:502: i/o timeout

Da kann evcc auch nix machen

closed 2025-08-02 by andig