evcc-io/evcc #22680: Falsche Werte vom Kostal Plenticore

KHerzberg, 2025-08-02:
### Describe the bug

Hallo,
vom Kostal PLENTICORE plus 7.0 der ersten Generation werden die falschen Leistungswerte für die Einspeisung verwendet. Zwischen der Eingangsleistung eines PV-Generators und der Ausgangsleistung des Inverters besteht in der Regel eine kleine Differenz. Diese bewegt sich zwar in der Regel in einem Bereich von 40 bis 100 Watt, ist jedoch da. In meinem Beispiel sind es 43 Watt. Dies ist mit insbesondere auch dann aufgefallen wenn die Inverterleistung bereits bei 0 W angekommen war, aufgrund von Mond, Sternen und Dämmerung jedoch noch ein paar Watt vom PV-Generator kamen.
Der nachfolgende Screenshot wurde so aufgenommen, dass evcc und der Wechselrichter ihre Anzeigen im gleichen Moment aktualisierten.

[image]

### Steps to reproduce

1. Integriere einen PLENTICORE plus 7.0 der ersten Generation über das Template.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 1a1e15aa1c6abb62a338ff47e....................

interval: 10s # control cycle interval

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.xxx.xxx  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.xxx.xxx  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery2
- name: grid3
  type: custom
  power:
    source: http
    uri: http://localhost:8081
    jq: .power
  currents:
    - source: http
      uri: http://localhost:8081
      jq: .current_l1
    - source: http
      uri: http://localhost:8081
      jq: .current_l2
    - source: http
      uri: http://localhost:8081
      jq: .current_l3
  voltages:
    - source: http
      uri: http://localhost:8081
      jq: .voltage_l1
    - source: http
      uri: http://localhost:8081
      jq: .voltage_l2
    - source: http
      uri: http://localhost:8081
      jq: .voltage_l3
#- type: template
#  template: tibber-pulse 
#  usage: grid  
#  token: <Geheim> 
#  homeid: <Geheim>  
#  name: grid3

tariffs:
  solar:
    - type: template
      template: open-meteo
      lat: <Geheim> # Breitengrad
      lon: <Geheim> # Längengrad
      dec: <Geheim> # Neigung, 0 = horizontal, 90 = vertikal
      kwp: 7.0 # Maximale Generatorleistung (kWp)
      az:  <Geheim> # Azimut, -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden
  co2:
    type: template
    template: green-grid-compass
    apiKey: <Geheim> # API Key, Erstelle eine App in https://api-portal.ec>
    zone: DE_LU # Zonencode, [BE, DE_LU]

chargers:
- type: template
  template: elliconnect 
  ski: <Geheim>  
  ip: 192.168.xxx.xxx  
  name: wallbox5

vehicles:
- type: template
  template: vw 
  user: <Geheim>  
  password: <Geheim>  
  vin: <Geheim>  
  title: VW  
  capacity: <Geheim>  
  timeout: 10s  
  name: ev4

loadpoints:
- title: Garage links
  charger: wallbox5
  mode: pv

site:
  title: My home
  meters:
    grid: grid3
    pv:
    - pv1
    battery:
    - battery2

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          <Geheim>
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          <Geheim>
          -----END CERTIFICATE-----
```

### Log details

```text
Datenfehler, Fehler steht nicht im Log
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.206.1

premultiply, 2025-08-02:
Bei Hybrid-WR nicht anders möglich, da nur DC-Eingangsleistungen verfügbar.
In der Differenz ist die Verlustleistung und Eigenverbrauch des Gerätes zu sehen.

closed 2025-08-02 by premultiply

KHerzberg, 2025-08-03:
Dafür finde ich es aber erstaunlich, dass ich genau diese Werte per Modbus auslesen kann:
```
154 Current Phase 1 5.88
156 Active power Phase 1 1400.53
158 Voltage Phase 1 239.24
160 Current Phase 2 5.97
162 Active power Phase 2 1431.42
164 Voltage Phase 2 240.75
166 Current Phase 3 5.95
168 Active power Phase 3 1408.64
170 Voltage Phase 3 238.04
172 Total AC active power 4488.0
174 Total AC reactive power 570.28
178 Total AC apparent power 4278.77
258 Current DC1 6.34
260 Power DC1 2101.92
266 Voltage DC1 325.59
268 Current DC2 6.12
270 Power DC2 2043.92
276 Voltage DC2 326.07
278 Current DC3 -0.0
280 Power DC3 0.0
286 Voltage DC3 0.39

```

Berechnet werden können hingegen nur die folgenden Werte:
```
Doing some Calculations of the received information:
Left Side Raw Power Generation of Panles : 4145.8
BatteryCharge (-) / Discharge(+) is      : 0.0
Powerfromgrid (-) /To Grid (+) is        : 4488.0
Total current Home consumption is        : 0.0
```

Hier noch das [Skript zum Auslesen von Kilian Knoll (Python)](https://github.com/user-attachments/files/21565308/KostalCurrent.py.txt) und die [Interface-Beschreibung von Kostal (PDF)](https://github.com/user-attachments/files/21565336/BA_KOSTAL_Interface_MODBUS-TCP_SunSpec_with_Control.pdf)

premultiply, 2025-08-03:
Ja, dann mach das mal wenn die Batteriewerte nicht 0 sind...

Leistungsmessungen am Netzanschluss eines Hybrid-WR sind irrelevant, da sie einen undefinierbaren Mix von PV, Batterie und Eigenverbrauch darstellen.

Relevant sind nur die Summe der PV-Erzeugsleistungen, die Summe der Batterieleistungen und die Leistung am Netzübergang des Anwesens.