evcc-io/evcc #22690: unerlaubten Ladestart einfangen - Pause zwischen enable und disable notwendig

VolkerK62, 2025-08-02:
Ich untersuche gerade den unerlaubten Ladestart der DaheimLaden Wallbox.
Durch die letzte Änderung wird der jetzt durch evcc zwar erkannt, aber das "einfangen" funktioniert noch nicht.
Im Log habe ich gesehen, dass direkt hintereinander "enable (6A)" und "disable (0A)" geschickt werden.
Die "6A" scheinen nicht durchzukommen bzw. werden von den "0A" direkt wieder eingefangen.
Das Register steht aber bei 0A, deshalb reagiert die Box darauf nicht.
Ich hätte eigentlich erwartet, dass die Box enabled wird und dann der disable timer anfängt. War das nicht früher mal so?
Wenn ich in dem unerlaubten Zustand den Ladestrom von extern manipuliere (erst 6A, dann 0A) und dazwischen 1-2 Sek Pause lasse, funktioniert das Abschalten und die Box ist danach unter der Kontrolle von evcc.

```
Cvolker@raspberrypi:~ $ sudo evcc config -c /home/volker/Austausch/test/test.yaml
[main  ] INFO 2025/08/02 14:44:29 using config file: /home/volker/Austausch/test/test.yaml
[db    ] INFO 2025/08/02 14:44:29 using sqlite database: /home/volker/Austausch/test/evcc.db
charger
---
db:4 {Type:template Title: Icon: Product:DaheimLaden Smart/Touch Pro} map[host:localhost port:5022 template:daheimladen-pro]

meter
---
db:1 {Type:template Title:PVfix Icon: Product:Demozähler} map[maxacpower:0 power:5000 template:demo-meter usage:pv]

loadpoint
---
db:5 {Type: Title: Icon: Product:} map[charger:db:4 circuit: defaultMode:off maxCurrent:16 meter: mode:off phasesConfigured:1 soc:map[estimate:true poll:map[interval:3.6e+12 mode:charging]] thresholds:map[disable:map[delay:1.8e+11 threshold:0] enable:map[delay:6e+10 threshold:0]] title:dhl_test vehicle:]
```
```
[lp-1  ] INFO 2025/08/02 11:41:39 start charging ->
[lp-1  ] DEBUG 2025/08/02 11:41:39 wake-up timer: stop
[daheimladen-mb] TRACE 2025/08/02 11:41:39 modbus: send 03 f1 00 00 00 06 ff 03 00 5b 00 01
[proxy-5022] TRACE 2025/08/02 11:41:39 read holdings: id 255 addr 91 qty 1
[proxy-5022] TRACE 2025/08/02 11:41:39 read holding: 00 00
[daheimladen-mb] TRACE 2025/08/02 11:41:39 modbus: recv 03 f1 00 00 00 05 ff 03 02 00 00
[lp-1  ] WARN 2025/08/02 11:41:39 charger logic error: disabled but charging
[daheimladen-mb] TRACE 2025/08/02 11:41:39 modbus: send 03 f2 00 00 00 09 ff 10 00 5b 00 01 02 00 48
[proxy-5022] TRACE 2025/08/02 11:41:39 write holdings: id 255 addr 91 qty 1 val 0048
[proxy-5022] TRACE 2025/08/02 11:41:39 write multiple holding: 00 01
[daheimladen-mb] TRACE 2025/08/02 11:41:39 modbus: recv 03 f2 00 00 00 06 ff 10 00 5b 00 01
[lp-1  ] DEBUG 2025/08/02 11:41:39 charger enable
[daheimladen-mb] TRACE 2025/08/02 11:41:39 modbus: send 03 f3 00 00 00 09 ff 10 00 5b 00 01 02 00 00
[proxy-5022] TRACE 2025/08/02 11:41:39 write holdings: id 255 addr 91 qty 1 val 0000
[proxy-5022] TRACE 2025/08/02 11:41:39 write multiple holding: 00 01
[daheimladen-mb] TRACE 2025/08/02 11:41:39 modbus: recv 03 f3 00 00 00 06 ff 10 00 5b 00 01
[lp-1  ] DEBUG 2025/08/02 11:41:39 charger disable
[lp-1  ] DEBUG 2025/08/02 11:41:39 wake-up timer: stop
[daheimladen-mb] TRACE 2025/08/02 11:41:48 modbus: send 03 f4 00 00 00 06 ff 03 00 0c 00 02
[proxy-5022] TRACE 2025/08/02 11:41:48 read holdings: id 255 addr 12 qty 2
[proxy-5022] TRACE 2025/08/02 11:41:48 read holding: 00 00 2a 30
[daheimladen-mb] TRACE 2025/08/02 11:41:48 modbus: recv 03 f4 00 00 00 07 ff 03 04 00 00 2a 30
[lp-1  ] DEBUG 2025/08/02 11:41:48 charge power: 10800W
[daheimladen-mb] TRACE 2025/08/02 11:41:48 modbus: send 03 f5 00 00 00 06 ff 03 00 06 00 05
[proxy-5022] TRACE 2025/08/02 11:41:48 read holdings: id 255 addr 6 qty 5
[proxy-5022] TRACE 2025/08/02 11:41:48 read holding: 00 9c 00 00 00 9d 00 00 00 9a
[daheimladen-mb] TRACE 2025/08/02 11:41:48 modbus: recv 03 f5 00 00 00 0d ff 03 0a 00 9c 00 00 00 9d 00 00 00 9a
[lp-1  ] DEBUG 2025/08/02 11:41:48 charge currents: [15.6 15.7 15.4]A
[daheimladen-mb] TRACE 2025/08/02 11:41:48 modbus: send 03 f6 00 00 00 06 ff 03 00 6d 00 05
[proxy-5022] TRACE 2025/08/02 11:41:48 read holdings: id 255 addr 109 qty 5
[proxy-5022] TRACE 2025/08/02 11:41:48 read holding: 09 1c 00 00 09 1b 00 00 09 19
[daheimladen-mb] TRACE 2025/08/02 11:41:48 modbus: recv 03 f6 00 00 00 0d ff 03 0a 09 1c 00 00 09 1b 00 00 09 19
[lp-1  ] DEBUG 2025/08/02 11:41:48 charge voltages: [233 233 233]V
[lp-1  ] DEBUG 2025/08/02 11:41:48 detected active phases: 3p
[daheimladen-mb] TRACE 2025/08/02 11:41:48 modbus: send 03 f7 00 00 00 06 ff 03 00 1c 00 02
[proxy-5022] TRACE 2025/08/02 11:41:48 read holdings: id 255 addr 28 qty 2
[proxy-5022] TRACE 2025/08/02 11:41:48 read holding: 00 00 00 d5
[daheimladen-mb] TRACE 2025/08/02 11:41:48 modbus: recv 03 f7 00 00 00 07 ff 03 04 00 00 00 d5
[daheimladen-mb] TRACE 2025/08/02 11:41:48 modbus: send 03 f8 00 00 00 06 ff 03 00 1c 00 02
[proxy-5022] TRACE 2025/08/02 11:41:48 read holdings: id 255 addr 28 qty 2
[proxy-5022] TRACE 2025/08/02 11:41:48 read holding: 00 00 00 d5
[daheimladen-mb] TRACE 2025/08/02 11:41:48 modbus: recv 03 f8 00 00 00 07 ff 03 04 00 00 00 d5
[lp-1  ] DEBUG 2025/08/02 11:41:48 charge total import: 21.300kWh
[daheimladen-mb] TRACE 2025/08/02 11:41:48 modbus: send 03 f9 00 00 00 06 ff 03 00 00 00 01
[proxy-5022] TRACE 2025/08/02 11:41:48 read holdings: id 255 addr 0 qty 1
[proxy-5022] TRACE 2025/08/02 11:41:48 read holding: 00 04
[daheimladen-mb] TRACE 2025/08/02 11:41:48 modbus: recv 03 f9 00 00 00 05 ff 03 02 00 04
[daheimladen-mb] TRACE 2025/08/02 11:41:48 modbus: send 03 fa 00 00 00 06 ff 03 00 0c 00 02
[proxy-5022] TRACE 2025/08/02 11:41:48 read holdings: id 255 addr 12 qty 2
[proxy-5022] TRACE 2025/08/02 11:41:48 read holding: 00 00 2a 30
[daheimladen-mb] TRACE 2025/08/02 11:41:48 modbus: recv 03 fa 00 00 00 07 ff 03 04 00 00 2a 30
[lp-1  ] DEBUG 2025/08/02 11:41:48 charger status: C
[daheimladen-mb] TRACE 2025/08/02 11:41:48 modbus: send 03 fb 00 00 00 06 ff 03 00 5b 00 01
[proxy-5022] TRACE 2025/08/02 11:41:48 read holdings: id 255 addr 91 qty 1
[proxy-5022] TRACE 2025/08/02 11:41:48 read holding: 00 00
[daheimladen-mb] TRACE 2025/08/02 11:41:48 modbus: recv 03 fb 00 00 00 05 ff 03 02 00 00
[lp-1  ] WARN 2025/08/02 11:41:48 charger logic error: disabled but charging
```

andig, 2025-08-02:
Dann wäre die Lösung vmtl. das Delay in die Implementierung der Box einzubauen. Probiers bitte mal aus.

VolkerK62, 2025-08-02:
du meinst in die "daheimladen-mb.go"?
Ich vermute irgendwo hier?
https://github.com/evcc-io/evcc/blob/3911dfb303d064e72aa8cfd1aec85717b26356c0/charger/daheimladen-mb.go#L200

Sorry, da hab ich keine Ahnung, wie das gehen soll und selber einen PR testen (vor merge) kann ich auch nicht.

andig, 2025-08-02:
Genau da. Einfach

    time.Sleep(2*time.Second)

VolkerK62, 2025-08-02:
Also so?
```
// Enable implements the api.Charger interface
func (wb *DaheimLadenMB) Enable(enable bool) error {
	var current uint16
	if enable {
		current = wb.curr
                time.Sleep(2*time.Second)
	}
       
	return wb.setCurrent(current)
}

PR #22695 (2025-08-03): daheimladen: 2s enable/disable delay
fix #22690