evcc-io/evcc #22691: Solar Forecast differs between front page and forecast page

picki72, 2025-08-02:
### Describe the bug

I've observed a difference between the remaining solar production displayed on the front page of the UI in comparison to the value displayed on the forecast page and the associated graphices. 

[image]
[image]

### Steps to reproduce

This came up after replacing the direct solcast solar forecast by a json feed from Home Assistant. The feed should be OK as the details on the forecast page are consistent to the data provided via the HomeAssistant Sensor accessed by EVCC via a custom http access. 

### Configuration details

```yaml
related section from the tariffs UI interface: 

solar: # list of pv generation forecast (additive)
- type: custom
  forecast:
    source: http
    uri: http://ip:8123/api/states/sensor.evcc_solcast_forecast
    headers:
      - Authorization: Bearer 123
      - Content-Type: application/json
    insecure: true
    jq: ".attributes.forecast | tostring"
    timeout: 2s

The output of the sensor used looks like follows: 
{"entity_id":"sensor.evcc_solcast_forecast","state":"69.91","attributes":{"forecast":[{"start":"2025-08-01T22:00:00Z","end":"2025-08-01T23:00:00Z","value":0.0},{"start":"2025-08-01T23:00:00Z","end":"2025-08-02T00:00:00Z","value":0.0},{"start":"2025-08-02T00:00:00Z","end":"2025-08-02T01:00:00Z","value":0.0},{"start":"2025-08-02T01:00:00Z","end":"2025-08-02T02:00:00Z","value":0.0},{"start":"2025-08-02T02:00:00Z","end":"2025-08-02T03:00:00Z","value":0.0},{"start":"2025-08-02T03:00:00Z","end":"2025-08-02T04:00:00Z","value":0.006},{"start":"2025-08-02T04:00:00Z","end":"2025-08-02T05:00:00Z","value":0.284},{"start":"2025-08-02T05:00:00Z","end":"2025-08-02T06:00:00Z","value":0.966},{"start":"2025-08-02T06:00:00Z","end":"2025-08-02T07:00:00Z","value":1.81},{"start":"2025-08-02T07:00:00Z","end":"2025-08-02T08:00:00Z","value":1.975},{"start":"2025-08-02T08:00:00Z","end":"2025-08-02T09:00:00Z","value":4.607},{"start":"2025-08-02T09:00:00Z","end":"2025-08-02T10:00:00Z","value":3.891},{"start":"2025-08-02T10:00:00Z","end":"2025-08-02T11:00:00Z","value":4.272},{"start":"2025-08-02T11:00:00Z","end":"2025-08-02T12:00:00Z","value":4.87},{"start":"2025-08-02T12:00:00Z","end":"2025-08-02T13:00:00Z","value":5.629},{"start":"2025-08-02T13:00:00Z","end":"2025-08-02T14:00:00Z","value":5.063},{"start":"2025-08-02T14:00:00Z","end":"2025-08-02T15:00:00Z","value":3.938},{"start":"2025-08-02T15:00:00Z","end":"2025-08-02T16:00:00Z","value":2.836},{"start":"2025-08-02T16:00:00Z","end":"2025-08-02T17:00:00Z","value":1.942},{"start":"2025-08-02T17:00:00Z","end":"2025-08-02T18:00:00Z","value":0.872},{"start":"2025-08-02T18:00:00Z","end":"2025-08-02T19:00:00Z","value":0.141},{"start":"2025-08-02T19:00:00Z","end":"2025-08-02T20:00:00Z","value":0.0},{"start":"2025-08-02T20:00:00Z","end":"2025-08-02T21:00:00Z","value":0.0},{"start":"2025-08-02T21:00:00Z","end":"2025-08-02T22:00:00Z","value":0.0},{"start":"2025-08-02T22:00:00Z","end":"2025-08-02T23:00:00Z","value":0.0},{"start":"2025-08-02T23:00:00Z","end":"2025-08-03T00:00:00Z","value":0.0},{"start":"2025-08-03T00:00:00Z","end":"2025-08-03T01:00:00Z","value":0.0},{"start":"2025-08-03T01:00:00Z","end":"2025-08-03T02:00:00Z","value":0.0},{"start":"2025-08-03T02:00:00Z","end":"2025-08-03T03:00:00Z","value":0.0},{"start":"2025-08-03T03:00:00Z","end":"2025-08-03T04:00:00Z","value":0.003},{"start":"2025-08-03T04:00:00Z","end":"2025-08-03T05:00:00Z","value":0.105},{"start":"2025-08-03T05:00:00Z","end":"2025-08-03T06:00:00Z","value":0.48},{"start":"2025-08-03T06:00:00Z","end":"2025-08-03T07:00:00Z","value":1.154},{"start":"2025-08-03T07:00:00Z","end":"2025-08-03T08:00:00Z","value":1.516},{"start":"2025-08-03T08:00:00Z","end":"2025-08-03T09:00:00Z","value":1.65},{"start":"2025-08-03T09:00:00Z","end":"2025-08-03T10:00:00Z","value":2.238},{"start":"2025-08-03T10:00:00Z","end":"2025-08-03T11:00:00Z","value":2.8},{"start":"2025-08-03T11:00:00Z","end":"2025-08-03T12:00:00Z","value":2.965},{"start":"2025-08-03T12:00:00Z","end":"2025-08-03T13:00:00Z","value":3.213},{"start":"2025-08-03T13:00:00Z","end":"2025-08-03T14:00:00Z","value":3.322},{"start":"2025-08-03T14:00:00Z","end":"2025-08-03T15:00:00Z","value":2.729},{"start":"2025-08-03T15:00:00Z","end":"2025-08-03T16:00:00Z","value":2.274},{"start":"2025-08-03T16:00:00Z","end":"2025-08-03T17:00:00Z","value":1.651},{"start":"2025-08-03T17:00:00Z","end":"2025-08-03T18:00:00Z","value":0.616},{"start":"2025-08-03T18:00:00Z","end":"2025-08-03T19:00:00Z","value":0.099},{"start":"2025-08-03T19:00:00Z","end":"2025-08-03T20:00:00Z","value":0.0},{"start":"2025-08-03T20:00:00Z","end":"2025-08-03T21:00:00Z","value":0.0},{"start":"2025-08-03T21:00:00Z","end":"2025-08-03T22:00:00Z","value":0.0}],"icon":"mdi:weather-sunny","friendly_name":"evcc_solcast_forecast"},"last_changed":"2025-08-02T11:56:49.196050+00:00","last_reported":"2025-08-02T11:56:49.196210+00:00","last_updated":"2025-08-02T11:56:49.196050+00:00","context":{"id":"abc","parent_id":null,"user_id":null}}
```

### Log details

```text
no relevant log entries underpinning this
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.206.1

andig, 2025-08-02:
Was sagt /api/state in dem Moment?

andig, 2025-08-02:
> I've observed a difference between the remaining solar production displayed on the front page of the UI in comparison to the value displayed on the forecast page and the associated graphices.

What's the difference? Your screenshot doesn't show one.

realwuffi, 2025-08-02:
Just put in your Home Assitant IP:
http://<haip>:7070/api/state

SolarPower2024, 2025-08-02:
> On the frontpage it shows 0,0 kWh as forecasted production (right from „Erzeugung“) - when clicking on this link it leads to the forecast page where the forecast remaining for the day is 11 kWh as of the other screenshot above.
> 
> How can I find out /api/state when using EVCC as a HomeAssistant Addon? Happy to check this.

No, it shows 11 Wh, which get rounded down to 0,0 KWh

picki72, 2025-08-02:
> Was sagt /api/state in dem Moment?

Das ist nicht vom Zeitpunkt der Screenshots, aber so sieht das aus:

{"result":{"aux":[],"availableVersion":"0.207.0","battery":[{"title":"Solaredge Batterie","power":704,"capacity":4.6,"soc":58.88888931274414,"controllable":true}],"batteryCapacity":4.6,"batteryDischargeControl":true,"batteryEnergy":0,"batteryGridChargeActive":false,"batteryMode":"unknown","batteryPower":704,"batterySoc":58.889,"bufferSoc":0,"bufferStartSoc":0,"currency":"EUR","eebus":false,"ext":[],"forecast":{"solar":{"scale":1.0542603860163184,"today":{"energy":0.0010950738593793025,"complete":true},"tomorrow":{"energy":26.815,"complete":false},"dayAfterTomorrow":{"energy":0,"complete":false},"timeseries":[{"ts":"2025-08-02T00:00:00+02:00","val":0},{"ts":"2025-08-02T01:00:00+02:00","val":0},{"ts":"2025-08-02T02:00:00+02:00","val":0},{"ts":"2025-08-02T03:00:00+02:00","val":0},{"ts":"2025-08-02T04:00:00+02:00","val":0},{"ts":"2025-08-02T05:00:00+02:00","val":0.006},{"ts":"2025-08-02T06:00:00+02:00","val":0.284},{"ts":"2025-08-02T07:00:00+02:00","val":0.966},{"ts":"2025-08-02T08:00:00+02:00","val":1.81},{"ts":"2025-08-02T09:00:00+02:00","val":1.975},{"ts":"2025-08-02T10:00:00+02:00","val":4.607},{"ts":"2025-08-02T11:00:00+02:00","val":3.891},{"ts":"2025-08-02T12:00:00+02:00","val":4.272},{"ts":"2025-08-02T13:00:00+02:00","val":4.87},{"ts":"2025-08-02T14:00:00+02:00","val":5.629},{"ts":"2025-08-02T15:00:00+02:00","val":5.063},{"ts":"2025-08-02T16:00:00+02:00","val":3.938},{"ts":"2025-08-02T17:00:00+02:00","val":2.836},{"ts":"2025-08-02T18:00:00+02:00","val":1.942},{"ts":"2025-08-02T19:00:00+02:00","val":0.872},{"ts":"2025-08-02T20:00:00+02:00","val":0.141},{"ts":"2025-08-02T21:00:00+02:00","val":0},{"ts":"2025-08-02T22:00:00+02:00","val":0},{"ts":"2025-08-02T23:00:00+02:00","val":0},{"ts":"2025-08-03T00:00:00+02:00","val":0},{"ts":"2025-08-03T01:00:00+02:00","val":0},{"ts":"2025-08-03T02:00:00+02:00","val":0},{"ts":"2025-08-03T03:00:00+02:00","val":0},{"ts":"2025-08-03T04:00:00+02:00","val":0},{"ts":"2025-08-03T05:00:00+02:00","val":0.003},{"ts":"2025-08-03T06:00:00+02:00","val":0.105},{"ts":"2025-08-03T07:00:00+02:00","val":0.48},{"ts":"2025-08-03T08:00:00+02:00","val":1.154},{"ts":"2025-08-03T09:00:00+02:00","val":1.516},{"ts":"2025-08-03T10:00:00+02:00","val":1.65},{"ts":"2025-08-03T11:00:00+02:00","val":2.238},{"ts":"2025-08-03T12:00:00+02:00","val":2.8},{"ts":"2025-08-03T13:00:00+02:00","val":2.965},{"ts":"2025-08-03T14:00:00+02:00","val":3.213},{"ts":"2025-08-03T15:00:00+02:00","val":3.322},{"ts":"2025-08-03T16:00:00+02:00","val":2.729},{"ts":"2025-08-03T17:00:00+02:00","val":2.274},{"ts":"2025-08-03T18:00:00+02:00","val":1.651},{"ts":"2025-08-03T19:00:00+02:00","val":0.616},{"ts":"2025-08-03T20:00:00+02:00","val":0.099},{"ts":"2025-08-03T21:00:00+02:00","val":0},{"ts":"2025-08-03T22:00:00+02:00","val":0},{"ts":"2025-08-03T23:00:00+02:00","val":0}]}},"greenShareHome":0.993,"greenShareLoadpoints":0,"grid":{"power":5.008,"energy":8759.222,"powers":[-27.572,17.914,11.468],"currents":[-0.6007,0.862,1.0366]},"gridConfigured":true,"hems":{},"homePower":709.008,"influx":{"url":"","database":"","token":"","org":"","user":"","password":"","insecure":false},"interval":30,"loadpoints":[{"batteryBoost":false,"chargeCurrents":[0,0,0],"chargeDuration":0,"chargePower":0,"chargeTotalImport":1561.942,"chargeVoltages":[230.6399994,228.7799988,230.3300018],"chargedEnergy":0,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":false,"chargerIcon":null,"chargerPhases1p3p":true,"chargerSinglePhase":false,"charging":false,"connected":true,"connectedDuration":0,"disableDelay":180,"disableThreshold":0,"effectiveLimitSoc":80,"effectiveMaxCurrent":16,"effectiveMinCurrent":6,"effectivePlanId":0,"effectivePlanSoc":0,"effectivePlanTime":null,"effectivePriority":0,"enableDelay":60,"enableThreshold":0,"enabled":false,"limitEnergy":0,"limitSoc":80,"maxCurrent":16,"minCurrent":6,"mode":"pv","offeredCurrent":0,"phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":0,"planActive":false,"planEnergy":0,"planOverrun":0,"planPrecondition":0,"planProjectedEnd":null,"planProjectedStart":null,"planTime":null,"priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"smartCostActive":false,"smartCostLimit":null,"smartCostNextStart":null,"smartFeedInPriorityActive":false,"smartFeedInPriorityLimit":null,"smartFeedInPriorityNextStart":null,"title":"go-e Garage","vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleLimitSoc":80,"vehicleName":"db:6","vehicleOdometer":5929,"vehicleRange":467,"vehicleSoc":80,"vehicleTitle":"ID7","vehicleWelcomeActive":false}],"messaging":false,"modbusproxy":null,"mqtt":{"broker":"","user":"","password":"","clientID":"","insecure":false,"caCert":"","clientCert":"","clientKey":"","topic":"evcc"},"network":{"schema":"http","host":"evcc.local","port":7070},"prioritySoc":100,"pv":[{"title":"Solaredge Wechselrichter","power":-26.899999999999977}],"pvEnergy":0,"pvPower":0,"residualPower":100,"siteTitle":"","smartCostAvailable":false,"smartCostType":"pricestatic","smartFeedInPriorityAvailable":false,"sponsor":{"name":"picki72","expiresAt":"2066-03-05T06:00:00Z"},"startup":true,"statistics":{"30d":{"avgCo2":0,"avgPrice":0.08398459245898827,"chargedKWh":158.09000000000037,"solarPercentage":98.69096821974198},"365d":{"avgCo2":0,"avgPrice":0.08340414235986804,"ch [... cut ...] 5463172952},"thisYear":{"avgCo2":0,"avgPrice":0.08340414235986804,"chargedKWh":730.0740000000002,"solarPercentage":98.9535463172952},"total":{"avgCo2":0,"avgPrice":0.08340414235986804,"chargedKWh":730.0740000000002,"solarPercentage":98.9535463172952}},"tariffFeedIn":0.081,"tariffGrid":0.309,"tariffPriceHome":0.083,"tariffPriceLoadpoints":0.309,"tariffSolar":0.141,"telemetry":false,"vehicles":{"db:6":{"title":"ID7","icon":"car","minSoc":25,"limitSoc":80,"repeatingPlans":[]}},"version":"0.206.1"}}