evcc-io/evcc #22705: Hyundai/Kia Login broken

tobiasbayer, 2025-08-03:
### Describe the bug

Hyundai/Kia changed login specifics. Login not possible anymore.

Possible fix see 

https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api/pull/850
https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api/pull/848

### Steps to reproduce

1. Start evcc with a configured Hyundai vehicle
2. See log: "Cannot create vehicle type 'template': cannot create vehicle type 'hyundai': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)"

### Configuration details

```yaml
Not applicable
```

### Log details

```text
Not applicable
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

StefanSchoof, 2025-08-03:
See https://github.com/evcc-io/evcc/issues/22634

StefanSchoof, 2025-08-03:
> Warum wird denn überall die Konversation geblockt ?

Ich vermute, weil Leute immer noch mit "bei mir auch" antworten oder mit anderen Themen den Thread nutzen. Da wird jedes Mal eine Mail an alle Teilnehmer des Threads geschickt und es macht das Auffinden von relevanten Informationen für die an einem Fix arbeiten unnötig aufwendig.

Es gibt einen PR der das fixen soll  #22709. Wenn jemand helfen will das Problem zu fixen könnte jemand den mit dem Problem bei sich bauen und testen. Wenn man das nicht kann, ist gerade die Zeit abzuwarten.

steps56, 2025-08-03:
Danke für die Info und den Link.
Leider habe ich null Wissen wie und wo ich da was einfügen/finden/testen kann und muss abwarten, wann  diese Lösung (sieht super und nach echter Lösung aus) in ein Evcc Update Eunzug hält.

Großen Dank und Respekt an ALLE die an Evcc arbeiten und dieses Toptool möglich machen 🤩.

andig, 2025-08-03:
Bitte offene Issues anschauen bevor immer wieder die gleichen Duplikate eröffnet werden...

closed 2025-08-03 by andig

tobiasbayer, 2025-08-03:
Hatte ich übersehen, weil ich nur nach Hyundai gesucht hatte aber im anderen Issue nur Kia im Titel stand.