evcc-io/evcc #22718: Shelly Gen2 API not implemented correctly?

S3vro, 2025-08-03:
### Describe the bug

Hallo zusammen,

nach mehreren Stunden debugging ist mir jetzt aufgefallen, das die gen2.go die Shelly RPC API nicht so implementiert wie spezifiziert. Zumindest nach meinem Verständnis des Codes.
Nach Shelly RPC Protocol (https://shelly-api-docs.shelly.cloud/gen2/General/RPCProtocol) wird sowohl als request body das folgende erwartet:
`{
   "id": 1,
   "src":"user_1",
   "method":"Switch.GetConfig",
}`. Das ist korrekt implementiert und wird alles richtig mitgeschickt.
Als Antwort wird nach dem Protocol das folgende gegeben:
`
{
   "id": 2,
   "src": "shellypro4pm-f008d1d8b8b8",
   "dst": "user_1",
   "result": {
      "was_on": false
   }
}
`.
Für Shelly.ListMethods wäre also dann das Ergebnis: 
`
{
   "id": 2,
   "src": "shellypro4pm-f008d1d8b8b8",
   "dst": "user_1",
   "result": {
      "methods": [<array of shelly methods>]
   }
}
`. Die gen2 implementierung erwartet aber immer das die Anwort als root zurückkommt (also direkt methods:" und nicht mit dem spezifizierten RPC Rahmen.

Habe ich hier irgendetwas übersehen oder liegt hier tatsächlich ein Implementierungsunterschied vor?

### Steps to reproduce

/

### Configuration details

```yaml
/
```

### Log details

```text
/
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.207.0 (241f3362)

andig, 2025-08-03:
Gibts denn jenseits der Theorie ein Problem? Welches?

/cc @thierolm

closed 2025-08-03 by andig

macrl2000, 2025-08-14:
Also mein "Problem" war, daß ich keinen shelly hatte und daher uni-meter zum testen genommen habe, und uni-meter implementiert die shelly spec 1:1 korrekt. das hat dann aber nicht zu den anfragen von evcc gepasst. (insbesondere der unterschied von POST zu GET bei der struktur der antworten). Ich hab jetzt einen shelly gekauft und siehe da, die antworten sind halt einfach nicht so wie in der spec :). sondern es kommt sowohl bei POST wie auch bei GET dieses {"methods":["EMData.Get.... Von daher - ja war alles akademisch. Kann man zumachen und beerdigen das Thema hier :-)