evcc-io/evcc #22723: QCells unregelmäßig Fehler durch zu schnelle Anfragen und niedrige Latenz

Caerum5, 2025-08-03:
### Describe the bug

Hallo,
ich habe einen kleinen Bug, dass immer mal wieder teilweise die Anfragen zum Wechselrichter für grid, pv und battery einen Fehler werfen. Aber das passiert nicht konstant.

Ich habe von QCells einen LAN-Adapter für den Wechselrichter. Vorher hatte ich einen WLAN-Adapter, wo das nicht passierte. Es gibt auch kein anderes Gerät, was gleichzeitig eine Anfrage sendet.

Nach etwas Testen habe ich herausgefunden, dass die Latenz vom evcc-Container zum LAN-Adapter bei ungefähr 0,1 bis 0,2 ms liegt (bei WLAN war dies bei ca. 200-300 ms). Ich vermute, dass die Anfragen von evcc aufgrund der Latenz manchmal zu schnell hintereinander geschickt werden, bevor der Adapter wieder frei schaltet, weshalb dann teilweise diese Fehler passieren. 
Also ich denke, wenn es eine kleine Verzögerung gäbe, sollte das nicht passieren. 
Vielleicht könnte auch so ein kleiner Delay über die Config einstellbar sein?

Dazu kommt: wenn die erste Anfrage einen Timeout als Fehler bekommt, dann werden die anschließenden Anfragen gar nicht mehr versucht und als Connection refused markiert.

Hier habe ich verschiedene Logs, die ich zur Übersicht etwas gekürzt habe:

#### Fehler bei grid energy, verursachte dann auch bei pv energy und battery soc (Latenz lag bei 0,13 ms, als der Fehler passierte)

```
[site ] DEBUG 2025/08/03 15:58:28 ----
[amperfied] TRACE 2025/08/03 15:58:28 modbus: send 01 65 00 00 00 06 fe 04 00 0e 00 01
[amperfied] TRACE 2025/08/03 15:58:28 modbus: send 01 6f 00 00 00 06 ff 04 00 0e 00 01
[amperfied] TRACE 2025/08/03 15:58:28 modbus: recv 01 65 00 00 00 05 fe 04 02 00 03
[lp-2 ] DEBUG 2025/08/03 15:58:28 charge power: 3W
[amperfied] TRACE 2025/08/03 15:58:28 modbus: send 01 66 00 00 00 06 fe 04 00 06 00 03
[amperfied] TRACE 2025/08/03 15:58:28 modbus: recv 01 6f 00 00 00 05 ff 04 02 00 00
[lp-1 ] DEBUG 2025/08/03 15:58:28 charge power: 0W
[amperfied] TRACE 2025/08/03 15:58:28 modbus: send 01 70 00 00 00 06 ff 04 00 06 00 03
[amperfied] TRACE 2025/08/03 15:58:28 modbus: recv 01 66 00 00 00 09 fe 04 06 00 00 00 00 00 00
[lp-2 ] DEBUG 2025/08/03 15:58:28 charge currents: [0 0 0]A
[amperfied] TRACE 2025/08/03 15:58:28 modbus: recv 01 70 00 00 00 09 ff 04 06 00 00 00 00 00 00
[lp-1 ] DEBUG 2025/08/03 15:58:28 charge currents: [0 0 0]A
[grid1 ] TRACE 2025/08/03 15:58:28 modbus: send 01 27 00 00 00 06 01 04 00 46 00 02
[grid1 ] TRACE 2025/08/03 15:58:28 modbus: recv 01 27 00 00 00 07 01 04 04 00 00 00 00
[site ] DEBUG 2025/08/03 15:58:28 grid power: -0W
[pv2 ] TRACE 2025/08/03 15:58:28 modbus: send 01 28 00 00 00 06 01 04 00 0a 00 01
[pv2 ] TRACE 2025/08/03 15:58:29 modbus: recv 01 28 00 00 00 05 01 04 02 02 1b
[pv2 ] TRACE 2025/08/03 15:58:29 modbus: send 01 29 00 00 00 06 01 04 00 0b 00 01
[pv2 ] TRACE 2025/08/03 15:58:29 modbus: recv 01 29 00 00 00 05 01 04 02 02 8f
[site ] DEBUG 2025/08/03 15:58:29 pv 1 power: 1194W
[battery3] TRACE 2025/08/03 15:58:29 modbus: send 01 2a 00 00 00 06 01 04 00 16 00 01
[battery3] TRACE 2025/08/03 15:58:30 modbus: recv 01 2a 00 00 00 05 01 04 02 01 09
[site ] DEBUG 2025/08/03 15:58:30 battery 1 power: -265W
[grid1 ] TRACE 2025/08/03 15:58:30 modbus: send 01 2b 00 00 00 06 01 04 00 4a 00 02
[grid1 ] TRACE 2025/08/03 15:58:33 modbus: close connection: read tcp 10.1.10.9:39022->10.1.10.200:502: i/o timeout
[site ] ERROR 2025/08/03 15:58:33 grid energy: read failed: read tcp 10.1.10.9:39022->10.1.10.200:502: i/o timeout
[site ] ERROR 2025/08/03 15:58:33 pv 1 energy: read failed: dial tcp 10.1.10.200:502: connect: connection refused
[site ] ERROR 2025/08/03 15:58:33 battery 1 soc: read failed: dial tcp 10.1.10.200:502: connect: connection refused
[site ] DEBUG 2025/08/03 15:58:33 battery has priority at soc 0% (< 75%)
[site ] DEBUG 2025/08/03 15:58:33 site power: 100W
[amperfied] TRACE 2025/08/03 15:58:33 modbus: send 01 67 00 00 00 06 fe 04 00 0a 00 03
[amperfied] TRACE 2025/08/03 15:58:33 modbus: recv 01 67 00 00 00 09 fe 04 06 00 ea 00 eb 00 eb
[lp-2 ] DEBUG 2025/08/03 15:58:33 charge voltages: [234 235 235]V
[lp-2 ] DEBUG 2025/08/03 15:58:33 detected connected phases: 3p

```

#### Hier direkt der nächste Zyklus 30 s später als Vergleich, ohne Fehler (Latenz bei 0,22 ms als die Anfragen für grid energy usw. waren)
```
[site ] DEBUG 2025/08/03 15:58:58 ----
[amperfied] TRACE 2025/08/03 15:58:58 modbus: send 01 70 00 00 00 06 fe 04 00 0e 00 01
[amperfied] TRACE 2025/08/03 15:58:58 modbus: send 01 75 00 00 00 06 ff 04 00 0e 00 01
[amperfied] TRACE 2025/08/03 15:58:58 modbus: recv 01 70 00 00 00 05 fe 04 02 00 02
[lp-2 ] DEBUG 2025/08/03 15:58:58 charge power: 2W
[amperfied] TRACE 2025/08/03 15:58:58 modbus: send 01 71 00 00 00 06 fe 04 00 06 00 03
[amperfied] TRACE 2025/08/03 15:58:58 modbus: recv 01 75 00 00 00 05 ff 04 02 00 00
[lp-1 ] DEBUG 2025/08/03 15:58:58 charge power: 0W
[amperfied] TRACE 2025/08/03 15:58:58 modbus: send 01 76 00 00 00 06 ff 04 00 06 00 03
[amperfied] TRACE 2025/08/03 15:58:58 modbus: recv 01 71 00 00 00 09 fe 04 06 00 00 00 00 00 00
[lp-2 ] DEBUG 2025/08/03 15:58:58 charge currents: [0 0 0]A
[amperfied] TRACE 2025/08/03 15:58:58 modbus: recv 01 76 00 00 00 09 ff 04 06 00 00 00 00 00 00
[... 106 lines cut ...]
  mppt3: false  
  name: battery3
```

### Log details

```text
siehe oben
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.0

Caerum5, 2025-08-03:
Ich hab es mal für dieses QCells bzw. Solax Template getestet:
meters:
- type: template
  template: solax 
  id: 1  
  host: 10.1.10.200 
  port: 502  
  usage: grid  
  modbus: tcpip  
  mppt3: false
  name: grid1
  delay: 100ms  

Aber das klappt auf jeden Fall so nicht: 
```
[main ] FATAL 2025/08/03 18:06:57 meter [grid1] cannot create meter 'grid1': cannot create meter type 'template': invalid key: delay
```

Gibt es irgendwo ein Beispiel für custom, an was ich mich halten kann?

andig, 2025-08-05:
@Caerum5 wenn Du evcc mit

    EVCC_TEMPLATE_RENDER=solax evcc

startest dann zeigt Dir evcc auf der Konsole, wie es das Template in ein "custom" Gerät verwandelt hat, also die Version in der schon alle Templatebefehle durch finale Konfiguration ersetzt sind. Das könntest Du dann als Config verwenden und weiter anpassen.

Caerum5, 2025-08-05:
@andig danke! Das hat geklappt mit dem Generieren der Custom Config. Ich habe das leicht angepasst und je Komponente einen Delay von 500 ms eingebaut. Seitdem bekomme ich keine Fehlermeldungen mehr 👍🏻 

Hat es eigentlich einen Grund, weshalb bei den Templates keine Delay Option eingebaut ist? 
Ansonsten denke ich, können wir das Issue hier schließen.

@steps56 danke dir auch für den Hinweis mit Waveshare. Ich werde mir das bei Gelegenheit mal anschauen und vielleicht auch zukünftig ausprobieren. 
Der LAN-Dongle kann durch ein Firmware Update mittlerweile zwar Modbus, aber ja wie du sagtest, ist dies vielleicht nicht so sauber, auch wenn es dank der Delays bei mir jetzt geht.

andig, 2025-08-05:
@premultiply wie wäre es denn, wenn wir das Delay mit ins Modbus Template aufnehmen? Man braucht's ja eigentlich nicht, aber falls doch ist es enorm hilfreich?

tolot27, 2025-10-05:
Ich habe in #23752 ähnliche Probleme und dachte, es lässt sich mit dem `timeout`-Parameter lösen. Nun lese ich zum ersten Mal vom `delay`-Parameter, welcher jedoch nicht unter https://docs.evcc.io/docs/reference/modbus dokumentiert ist. Was ist der Unterschied von beiden?

steps56, 2025-10-05:
Okay und danke für den Link.
Hört sich nach längst vor Wochen implementiert an... nightly mindestens, oder lese ich das falsch 🤔 ?
Mit nightly könnte ich testen, da mit Final nebeneinander installiert (aber logo immer nur eine Variante aktiv).
Ist bei mir aber echtes Luxusdings und nur meinem technischen Interesse geschuldet 😇.