evcc-io/evcc #22734: homeassistant vehicles is not working

tuxtof, 2025-08-04:
### Describe the bug

i try to use home assistant vehicule and it failed with

```
[main  ] ERROR 2025/08/04 07:52:45 creating vehicle my_car failed: cannot create vehicle type 'template': empty instance type- check for missing usage
```

### Steps to reproduce

1. use following config

```
vehicles:
  - name: my_car
    type: template
    template: homeassistant
    uri: http://ha.geo6.net:8123
    token: xxxxxxxxxxxxxxxxxxxx
    soc: sensor.state_of_charge
```

2. restart evcc

### Configuration details

```yaml
./evcc dump --cfg
[main  ] INFO 2025/08/04 08:01:21 using config file: /Users/cris/temp/evcc/evcc.yaml
[db    ] INFO 2025/08/04 08:01:21 using sqlite database: /Users/cris/.evcc/evcc.db
[main  ] ERROR 2025/08/04 08:01:22 creating vehicle my_car failed: cannot create vehicle type 'template': empty instance type- check for missing usage
[tariff] DEBUG 2025/08/04 08:01:22 using cache: electricitymaps-free-9be99cc67300850ccd95aa0a2178529b9458b5d4879783bde0ecbebc1d1f469a (start: 2025-08-04 07:00:00 +0200 CEST, end: 2025-08-06 07:00:00 +0200 CEST)

Version: `0.207.0 (43aac75f)`
```

### Log details

```text
[main  ] ERROR 2025/08/04 07:52:45 creating vehicle my_car failed: cannot create vehicle type 'template': empty instance type- check for missing usage
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.207.0 (43aac75f)

andig, 2025-08-04:
This should be fixed in master. /cc @thecem

tuxtof, 2025-08-04:
oops i forgot to sync my fork against upstream

now this is fixed with evcc version 0.207.1 (fa718be3)

let's test deeply

thanks

thecem, 2025-08-04:
If you have suggestions or problems pls use the discussion https://github.com/evcc-io/evcc/discussions/22676