evcc-io/evcc #22736: Ladedauer bei Wärmepumpe bleibt wieder stehen

LeonSol, 2025-08-04:
### Describe the bug

Eigentlich war dieses Problem #22221 mit 0.206.0 gelöst - nun, nach update auf 0.207.0 ist es wieder da:

alles wird auf null gesetzt, nur die Ladedauer nicht - gestern der Ladevorgang, heute Dauer immer noch da!

[image]

Wie kann die Änderung von einer Version auf die nächste wieder verloren gehen?

Sonnige Grüße

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Nicht nötig
```

### Log details

```text
Nicht nötig
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.0

LeonSol, 2025-08-07:
Könnte bitte jemand die Code-Zeilen die dieses Verhalten behoben haben erneut in das nächste Release packen?

Vielen Dank

Stony111, 2025-08-08:
I have also noticed the same problem in connection with a heating rod:

[image]

Loading energy is resetted but loading time not!

PS: I have already installed version 0.207.1 !

LeonSol, 2025-08-13:
@andig kannst du bitte die Lösung von #22396 wieder in das nächste release aufnehmen?

andig, 2025-08-13:
Closed in 77bceff55735358b8ec24992998884800dcea837. Scheint als wäre das niemals behoben worden, nur hat es mangels Test bis heute auch niemand bemerkt/gestört...

closed 2025-08-13 by andig