evcc-io/evcc #22742: Fehlerhafte/Unklare Anzeige PV-Vorhersage unter Tarife.

OberfragGER, 2025-08-04:
### Describe the bug

Unter Konfiguration -> Netzanschluss -> Tarife habe ich u.a. 4 verschiedene Ausrichtungen konfiguriert.

[image]

Hier steht 2,2 kW. Den Wert kann ich mir nicht herleiten. Ich vermute dass hier ggf. nur 1ne Seite berücksichtigt wird. Außerdem ist die Angabe "kW" ja auch eher eine aktuelle Ertragsanzeige und nicht die Vorhersage für den restlichen Tag. Dann müsste das ja "kWh" sein.

[image]

Unter der Gesamtübersicht siht mann auch den Forecast  ~29,2 kWh.

In Summe kann ich den Wert nicht nachvollziehen oder irgendwie ableiten.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Version  0.207.0

solar:
  - type: template
    template: forecast-solar
    lat: x # Breitengrad
    lon: y # Längengrad
    dec: 22 # Neigung, 0 = horizontal, 90 = vertikal
    kwp: 0.86 # Maximale Generatorleistung (kWp)
    az: -8 # Azimut, -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden
  - type: template
    template: forecast-solar
    lat: x # Breitengrad
    lon: y # Längengrad
    dec: 22 # Neigung, 0 = horizontal, 90 = vertikal
    kwp: 3.01 # Maximale Generatorleistung (kWp)
    az: 172 # Azimut, -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden
  - type: template
    template: forecast-solar
    lat: x # Breitengrad
    lon: y # Längengrad
    dec: 22 # Neigung, 0 = horizontal, 90 = vertikal
    kwp: 5.16 # Maximale Generatorleistung (kWp)
    az: -98 # Azimut, -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden
  - type: template
    template: forecast-solar
    lat: x # Breitengrad
    lon: y # Längengrad
    dec: 22 # Neigung, 0 = horizontal, 90 = vertikal
    kwp: 3.44 # Maximale Generatorleistung (kWp)
    az: 82 # Azimut, -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden
```

### Log details

```text
...
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-08-04:
> Hier steht 2,2 kW. Den Wert kann ich mir nicht herleiten. Ich vermute dass hier ggf. nur 1ne Seite berücksichtigt wird. Außerdem ist die Angabe "kW" ja auch eher eine aktuelle Ertragsanzeige und nicht die Vorhersage für den restlichen Tag.

@naltatis korrekterweise müsste man hier interpolieren oder den Zeitpunkt mit angeben.

> Unter der Gesamtübersicht siht mann auch den Forecast ~29,2 kWh.
> In Summe kann ich den Wert nicht nachvollziehen oder irgendwie ableiten.

@OberfragGER was sagt zu diesem Zeitpunkt /api/state?

andig, 2025-08-04:
> Der angezeigte Wert ist die für die aktuelle Stunde erwartete Produktion. 

Es gibt zwei Werte der Leisungsvorhersage: zu Beginn und zu Ende der Stunde. Ich vermute hier ist der Wert aus Beginn der Stunde gemeint? 

> Da wir auf Stundenslots sind entspricht das auch der aktuellen erwarteten Leistung (kW).

Nein, tut es nicht- das wäre de Durchschnitt aus beiden Werten.

andig, 2025-08-04:
> @andig Wir zeigen hier aktuell direkt den gepublishten tariffSolar aus dem Tarif an. Das ist, wie bei den Preistarifen der Wert vom Stundenbeginn. 

By Preisen ist das der Wert für die ganze Stunde. Bei PV ist es die Leistung zum Zeitpunkt.

> Der PV-Vorhersagewert ist nicht sehr aussagekräftig, aber ich würd ich erst einmal so lassen weil plausibel und nicht kritisch.

Da kann ich mit leben. Oder einfach raus nehmen weil an der Stelle physikalisch falsch.

OberfragGER, 2025-08-04:
Danke für eure ausführlichen Antworten.
So wie @naltatis sagte. "Der PV-Vorhersagewert ist nicht sehr aussagekräftig, aber ich würd ich erst einmal so lassen weil plausibel und nicht kritisch."

Verstanden. Sollte das irgendwie wichtig sein habt ihr das sicher schon ins Backlog aufgenommen. Um die Issues nicht weiter vollzumüllen, mache ich mal zu.

closed 2025-08-04 by OberfragGER