evcc-io/evcc #22743: Problem mit Tesla Powerwall 2 beim einstellen der Backup Reserve / Netzladen auf 100 %

Dragonkit, 2025-08-04:
### Describe the bug

Im EVCC gibt es den Fehler wenn ich Netzladen einstelle auf 100%.

 ''' battery mode: battery reserve failed: backup reserve updated '''

Das Problem ist wohl nach dem Update von Tesla auf Version 25.18.4

Es wird auch in verschiedenen anderen Foren schon geschimpft darüber.

z.B.
https://teslamotorsclub.com/tmc/threads/powerwall-charge-throttled-at-80-with-25-18-4-setting-pw-reserve-between-81-99-restricted-by-tesla-in-some-modes.347099/

Man kann wohl keinen Wert mehr zwischen 81-99 % angeben Sondern muss bei max 80 % bzw. genau 100 % sein.
Ich glaube mich zu errinnern das hier mal was angepasst wurde im EVCC weil er Probleme hatte den Wert auf 100 % zu setzen wenn der Akku zu 100 % geladen war und es deshalb ein +1 gab oder so.
Denke das ist nun das Problem mit den "genau" auf 100 % setzen. 
Vielleicht könnt ihr das irgendwie fixen.

Er setzt es dummerweise danach auch noch mehr auf den normalen Wert zurück. In meinem Fall 20 %. Das muss ich dann per Hand über die Tesla App machen. Im Sommer noch nicht so schlimm aber im Winter wenn man Nachts laden will eher doof.

Danke.

### Steps to reproduce

1.
2.
3.Im EVCC gibt es den Fehler wenn ich Netzladen einstelle auf 100%.

 ''' battery mode: battery reserve failed: backup reserve updated '''

Das Problem ist wohl nach dem Update von Tesla auf Version 25.18.4

Es wird auch in verschiedenen anderen Foren schon geschimpft darüber.

z.B.
https://teslamotorsclub.com/tmc/threads/powerwall-charge-throttled-at-80-with-25-18-4-setting-pw-reserve-between-81-99-restricted-by-tesla-in-some-modes.347099/

Man kann wohl keinen Wert mehr zwischen 81-99 % angeben Sondern muss bei max 80 % bzw. genau 100 % sein.
Ich glaube mich zu errinnern das hier mal was angepasst wurde im EVCC weil er Probleme hatte den Wert auf 100 % zu setzen wenn der Akku zu 100 % geladen war und es deshalb ein +1 gab oder so.
Denke das ist nun das Problem mit den "genau" auf 100 % setzen. 
Vielleicht könnt ihr das irgendwie fixen.

Er setzt es dummerweise danach auch noch mehr auf den normalen Wert zurück. In meinem Fall 20 %. Das muss ich dann per Hand über die Tesla App machen. Im Sommer noch nicht so schlimm aber im Winter wenn man Nachts laden will eher doof.

Danke.
...

### Configuration details

```yaml
meters:
  - name: my_grid
    type: template
    template: tesla-powerwall
    usage: grid
    host: ***
    password: ***
    refreshToken: *** 
  - name: my_pv
    type: template
    template: tesla-powerwall
    usage: pv
    host: *** 
    password: *** 
    refreshToken: ***  
  - name: my_battery
    type: template
    template: tesla-powerwall
    usage: battery
    host: *** 
    password: *** 
    refreshToken: *** 
    minSoc: 20
    maxSoc: 100
```

### Log details

```text
ist aktuell leider überschrieben, ich kann das aber gerne nochmal umstellen und das Log dann nachreichen.
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

207.0

Dragonkit, 2025-08-05:
> aber es wäre schön, wenn man in evcc komplett abschalten könnte, dass es die Notstromreserve der Powerwall verstellt.

Warum hast Du das dann aktiviert im EVCC ? Wenn Du es nicht aktivierst stellt es auch nix um :-)

bill340, 2025-08-05:
Das möchte ich ja eben auch.
Aber wo in evcc stelle ich denn die Notstromreserve ein?

andig, 2025-08-05:
https://github.com/evcc-io/evcc/issues/22743#issuecomment-3150242302

GrimmiMeloni, 2025-08-07:
Ich sehe das hier auch lokal bei mir. Ich schnapp mir das und passe es an.

Die Batteriesperre für die PW2 ist minSoc basiert, und wenn es jetzt einen "toten" Bereich 81% - 99% gibt müssen wir entscheiden was wir bei einem aktuellen PW soc in diesem Bereich wollen:

a) den minSoc für die Batteriesperre auf maximal 80% setzen - führt dann zu Entladung bis runter auf 80%.
b) den Wert rauf auf 100% setzen - führt zu Netzladung (falls für die PW aktiviert). 

Mir erscheint a) sinnvoller.