evcc-io/evcc #22770: Load Manager per phase

pczolee, 2025-08-04:
Dear EVCC team.

Thanks for this great work. I have an idea. In my house I have: 3phase. Every phase have an 32A main circuit braker. My house have no symmetric usage in 3phase so it can happen that the usage are: L1: 10A L2: 6A L3: 20A
So in that case I want evcc to enable only max 12A/phase to charge.

Thanks

pczolee, 2025-08-05:
> current is maximum per phase, not the sum. So if your breaker is 32A, use `maxCurrent: 32`

Ok, I see! But what to do if my breakers are not same, for example: L1: 32A L2: 25A L3: 20A

mfuchs1984, 2025-08-05:
This would be a rather strange setup.

premultiply, 2025-08-05:
I have never seen cables where the phase conductors have different diameters yet.

pczolee, 2025-08-05:
> You wrote this, so this seems to be not a real issue for you
> 
> > Every phase have an 32A main circuit braker.

Yes for me maybe it will work with this setup (I need to test), but I have some friend who have that strange setup...

mfuchs1984, 2025-08-05:
With the current simple load management implementation in evcc, such special setups cannot be covered.

SolarPower2024, 2025-08-05:
Well, in the end it can be covered: you need to define the weakest phase for the loadmanagement.

Anyway the charger will charge the 3 phases synchron.

mfuchs1984, 2025-08-05:
Sure, this way, evcc makes sure you never exceed the current the weakest phase can provide.

mfuchs1984, 2025-08-05:
True, when the charger is just connected to one phase, the limit of this phase can be used as the current limit in evcc.

pczolee, 2025-08-05:
Thanks everybody for helping! Yes there are no to many setup like 20/6/6 , but sometimes can happen. I'll do what you mentioned, so I'll see how working if I use the lowest phase value!

MartijnvanGeffen13, 2025-08-08:
ON 20/6/6 
Its is NOT safe to do 3P charging and you should switch / stick to single Phase charging, 

As the minimum for 3P = 6 A per phase , You would blow out our main breaker at the moment you switch a electric device on on 1 of the 2 Phases. 

If any phase is lower than 16A its debatable if its safe , atleast make sure the LB is working and enough overhead is set in the maximum so you dont blow your mains.  

Finally chargers cant loadbalance between phases, they will always draw same current from all 3 Phases. hence the weakerst phase is allways leading in the LB.   if you have 25A/25A/25A    and the house is using 15A on L1 and nothing on others, you only have 3x10A -overhead (atleast 2A but more is advised) available roughly 5.5 Kwh charging.